hildogjr/KiCost #375: KitSpace is unreachable

romain145, 2019-11-28:
Using KiCost v1.1.3 with a KiCAD 5.1.4 xml BOM.

```
kicost -w -i esc10080.xml --debug 8
[...]
# Getting part data from KitSpace...
Starting new HTTPS connection (1): dev-partinfo.kitspace.org:443
https://dev-partinfo.kitspace.org:443 "POST /graphql HTTP/1.1" 504 192
Progress:   0%|                                                                                                                                                                          | 0/42 [01:00<?, ?part/s]Traceback (most recent call last):
  File "/home/romain/.local/bin/kicost", line 11, in <module>
    sys.exit(main())
  File "/home/romain/.local/lib/python3.6/site-packages/kicost/__main__.py", line 319, in main
    dist_list=dist_list, currency=args.currency)
  File "/home/romain/.local/lib/python3.6/site-packages/kicost/kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "/home/romain/.local/lib/python3.6/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 324, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "/home/romain/.local/lib/python3.6/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 183, in get_part_info
    results = api_partinfo_kitspace.query(query)
  File "/home/romain/.local/lib/python3.6/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 124, in query
    raise Exception('One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.')
Exception: One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.
Progress:   0%|                                                                                                                                                                          | 0/42 [01:01<?, ?part/s]
```

hildogjr, 2019-11-29:
We use Partinfo API developed by @kasbah, I don't if there is some problem with the service.

kasbah, 2019-11-29:
Hey, we hit the Octopart limit I set. But I haven't yet used the ones for my work so have increased the limit for your use. It should work again now.

hildogjr, 2019-11-29:
Yes, I tested and is fine now.
Just remembering, @romain145. The Partinfo API service is kept by a [Open Collective Funding](https://opencollective.com/kitspace). If you want to play a large number requests, KiCost still supporting your on Octopart KEY.

closed 2019-11-29 by hildogjr