hildogjr/KiCost #376: DigiKey part numbers target Digi-Reel

romain145, 2019-12-04:
The DigiKey part numbers returned are a mix between cut-tape and Digi-Reel.
The difference is that Digi-Reel adds cost (typically $7.00) to each part number.
Is there a way to specify a preference to KiCost or to the Partinfo/Octopart API?

hildogjr, 2019-12-04:
Give us a file example to help track the issue and provide the KiCost version `kicost --version` /  `kicost --info`.

romain145, 2019-12-04:
```
Z:\git\github\KiCost>kicost --version
KiCost v.1.1.3
```

See attached [esc.zip](https://github.com/xesscorp/KiCost/files/3922083/esc.zip)

DigiKey references containing "HDKR" are Digi-Reel.
DigiKey references containing "HCT" are Cut-Tape.

hildogjr, 2019-12-06:
Could you be more specific?
Which part are related to the problem? (Are all parts?) What the problem cause in the spreadsheet? (I image that change the price tiers in the balloon and the price function).
I have little time between my jobs to catch the issues, specific descriptions could help.

romain145, 2019-12-08:
Sure!
The part numbers are a mix of Cut-Tape and Digi-Reel as shown below:

[image]

When searching for the part on Octopart, it returns both options. not sure how Partinfo picks one and return it to KiCost?

[image]

fri-sch, 2019-12-22:
I can confirm this issue and I also have a fix: #384

hildogjr, 2019-12-22:
@romain145, I just tested the fix provided by @fri-sch, appear fine and without any collateral. Could you confirm that fix the reported case?

fri-sch, 2019-12-27:
> I just concerned because it check by Stock numbers that ends with "DKR-ND", I am checking some user case to be sure that covers all Digikey Reels (it appear be the only information that tha API give us and differentiates the parts).

You're concerns are justified. I just found out, that the 'DKR-ND' ending doesn't cover all DigiKey reels. It seems like only capacitors, resistors, etc. are named like this. Other components, like ICs have '-6-ND' for the reels and '-1-ND' for the cut-tape. I don't know if there are other schemes for other kinds of parts.

Here's another fix: #385

hildogjr, 2019-12-28:
Thanks by check @fri-sch, I merged your solution, I just will improve to add this configuration at the Digikey profile of KiCost before the official release.
I also had add your name to official contribution history https://github.com/xesscorp/KiCost/blob/master/kicost/AUTHORS.rst (this is displayed at the GUI interface and help page).

hildogjr, 2019-12-31:
There still some inconsistencies about the link and stock code.

commit 05ba83a7 (2019-12-22): Merge pull request #384 from fri-sch/partinfo_packaging_selection

Improve the selection for different packaging offers for partinfo Fix #376
commit b3423fbc (2019-12-31): Fix #376 and release
closed 2019-12-31 by hildogjr