hildogjr/KiCost #380: Tests are not run on CI

johnthagen, 2019-12-08:
#### Feature / Enhancement request

Currently the Travis CI build runs:

https://github.com/xesscorp/KiCost/blob/fca4ac69931179ecd93fdea6135b56495faff26b/.travis.yml#L16

But the only Python test file is a blank test:

https://github.com/xesscorp/KiCost/blob/fca4ac69931179ecd93fdea6135b56495faff26b/tests/test_kicost.py#L21-L22

But there are test scripts that perhaps could be run on CI:

- https://github.com/xesscorp/KiCost/blob/master/tests/test.sh
- https://github.com/xesscorp/KiCost/blob/master/tests/test_small.sh

Or the `unittest` tests could be enhanced.

hildogjr, 2019-12-09:
Although I am the main active programmer at the project, I have no experience to in configure CI. Tips will be very useful, @johnthagen .

johnthagen, 2019-12-09:
It may be as simple as running the `.sh` scripts after the environment is setup here:

https://github.com/xesscorp/KiCost/blob/df700a0def22819f3accb41279e7ed122a2a2ce7/.travis.yml#L18

I'll try to put together a PR to see if I can get it to work.

closed 2019-12-10 by hildogjr

hildogjr, 2019-12-20:
@johnthagen , thanks by fix this.
Is it possible to configure this repository to automatically publish KiCost::master at https://pypi.org/project/kicost/? Could you help me to set this up?

johnthagen, 2019-12-21:
I don't know to how to auto publish via the GitHub, but this is the guide for how for can upload a new version yourself: https://packaging.python.org/tutorials/packaging-projects/#generating-distribution-archives

PR #381 (2019-12-10): Run test scripts on CI
Closes #380