hildogjr/KiCost #387: Enabling TME in scraping

maciekKuba, 2020-02-10:
#### Issue / Problem report
1. KiCost version 1.1.3. Run on Windows 10.
2. No data in output file from TME distributor.
3. kicost -i tests\RPi-Test.xml
4. Tested with files from repository's test directory.

I have fix for it, however have problems with testing for py26, py33 and py34. I did original trials with python3.8 and with my fix the lacking previously data from TME appeared successfully. In case of running tox and having installed all required versions from py26 to py34 I get errors related to incorrect version of lxml. I also have problem with making virtualenv for py26,py33,py34.

Log example from running tox -e py34:
Collecting lxml>=3.7.2 (from kicost==1.1.3)
  Using cached https://files.pythonhosted.org/packages/e1/f5/5eb3b491958dcfdcfa5daae3c655ab59276bc216ca015e44743c9c220e9e/lxml-4.4.0.tar.gz
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: This lxml version requires Python 2.7, 3.5 or later.
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in C:\Users\hp640\AppData\Local\Temp\pip-install-t9ww392x\lxml\

I would like to apply personally, but need help with testing.

Thank you

hildogjr, 2020-02-10:
I made a first fix (https://github.com/xesscorp/KiCost/commit/3ee85666eab93be7557f431645a495e1823f602c. Please, check i. Basically:

- `distributors/distributors.py` have all the information about the web distributors;
- `distributors_modules_dict['distributors']` and `distributors_modules_dict['dist_translation']` inside `distributors/api_partinfo_kitspace.py` and `distributors/api_octopart.py` enable the distributor in the current API.

KiCost doesn't do scrap anymore, it was a cumbersome to maintenance, nowdays it use the [Kitspace Open Collective](https://opencollective.com/kitspace) to run the service. Other APIs can be added by just create new `distributors/api_*.py` and new distributors can be included editing `distributors/distributors.py`.
The is also a plan do create a direct distributors API access by using `distributors/dist_*.py` files.

closed 2020-02-10 by hildogjr

maciekKuba, 2020-02-13:
I had the same fix. Works good now. How about running tests for python 26/33/34. Do you have same issue?

hildogjr, 2020-02-14:
I didn't test all the environment but by current machine use Py3.4.