hildogjr/KiCost #388: KiCost it! button not working and there's no output

MohamedElwazzan, 2020-02-23:
Kicost v.1.1.3
GUI
I've installed Kicost two months ago and it was working well
today I decide to generate spreadsheet  but unfortunately without any reason it just not responding to the "KiCost it!" button.
I've tried to use an old xml file that worked before but still the same problem 
Also I tried to reinstall it and figure out if there's any error log or something but I have no idea.

Best Regards

hildogjr, 2020-02-24:
thanks by reporting, I have tested it now and I didn't get this message. it could had be some temporary unavailable of the [Kispace server](https://opencollective.com/kitspace).

Please, check it again today.

MohamedElwazzan, 2020-02-25:
Thanks for your reply.
I've checked and it works well now 
yes maybe as you said it was just a temporary problem with Kispace server and now it works again 
Regards,

closed 2020-02-25 by MohamedElwazzan

hildogjr, 2020-02-25:
There is a plan to add multiple sources to KiCost (directly Digikey, Mouse through them API) but it still in a plan.