hildogjr/KiCost #389: Ignore Value when grouping

metropt, 2020-03-06:
How can I group the following parts by footprint and manf# per example?

I tried to use the --group_fields for value but it stopped with an error.

[image]

Thank you

hildogjr, 2020-03-06:
I think it not possible, did you get some warning message?
By safety I believe the is not possible to merge different value parts, because they are essentially different component electrical parts.
I have the same use of your and I use the `Desc` field at KiCost (or other EDA) to tell the button function (`kicost foo --group_fields desc`). The values of all them stay as "micro switch" or something like that.

metropt, 2020-03-06:
I'll do that. Will start Desc for this kind of information.

closed 2020-03-06 by metropt

hildogjr, 2020-03-06:
"desc" and "description" are deal as alias by KiCost.
You can also run `kicost --setup` (update your version to the last one available here) to create all the configurations and default fields in KiCad.