hildogjr/KiCost #391: Subpart documentation vs implementation

metropt, 2020-03-31:
Hi, the documentation of kicost define a subpart as two part-numbers separate by a comma however it is implemented to accept a `;` or a `,`

https://github.com/xesscorp/KiCost/blob/b24947d7b755e4a2b21b78f4c80ab65914c6b778/kicost/edas/tools.py#L39

Lately I have selected some parts that has a `,` on the part-number.

How do you think it should be handled?

hildogjr, 2020-03-31:
Both are possible to split the subparts.
The only think that you have to keep in mind is that "manf#" code of parts that have "," or ";" should have it preceded by escape argument "\". Example: "REGULATOR-CODE,3V3" should be typed as "REGULATOR-CODE\,3V3" in your BOM (KiCad or CSV file).

I think the docs may need a user revision (for us programmers it is already explained)

hildogjr, 2020-03-31:
If you are a new user, first run `kicost --setup` after install. It should be automatic but #372. It not mandatory but will improve the KiCad integration.

closed 2020-03-31 by hildogjr

metropt, 2020-03-31:
You probably mean "REGULATOR-CODE\,3V3"

hildogjr, 2020-03-31:
Yes, sorry. I had have to type \\\\, the first to scape the GitHub interpreter.