hildogjr/KiCost #393: Cannot paste BOM on Farnell

romain145, 2020-05-17:
The generated list of parts to purchase is not in a format accepted by Farnell Quick-Buy (also called Quick-Paste) tool.

The list of parts generated in the spreadsheet doe snot have the comma or tab separators:

[image]

[image]

Using Kicost v1.1.4.

hildogjr, 2020-05-17:
I don't have a Farnell account. Please, provide me the correct format/sequence to I make the internal chances.

romain145, 2020-05-17:
The format is here: 
https://export.farnell.com/quick-order?isQuickPaste=true

Example:
'4340504, 10, Need for project X', CUST1234
'4333974, 100, Order by Bob'

hildogjr, 2020-05-17:
So: stock code, quantity, description, user fields?

Are the user fields (optional) really used/stamped at the delivered package?

romain145, 2020-05-17:
Yes they are and it's really handy :)
The parts reference get printed on the bag:

[image]

hildogjr, 2020-05-17:
This will need some lines of code. Could you give more information about específica this: maximum fields allowed (I suppose that is printed in a by-line format), maximum length allowed (characters allowed at the line), is the some not allowed characters,...

romain145, 2020-05-17:
Maximum characters allowed for line comment and part number seems to be 30 (including commas).
Usually I leave the line comment empty by putting a double comma like this:
`2211140, 10,, CUST1234,CUST1234,CUST1234,CUST1234,CUST1234,`

Note that if the part number is longer than 30 characters, it gets truncated down to 30 like this:
[image]
However if the field is longer than 65 the line is ignored.

Special characters are supported, except for ' and " which are parsed out. Hope this helps :)

hildogjr, 2020-05-18:
Fix the order, missing the implementation of the user custoom fields.
@romain145 , please check it.
It will be helpful if you could also give some try on other distributors / link the link for direct importation (for example, Digikey is https://www.digikey.com/ordering/shoppingcart)

hildogjr, 2020-05-18:
Next step will be include the 'desc' (description) and others custom fields, all displayed at the spradsheet.
Could you provide me a XML example and the intended table list?

hildogjr, 2020-05-19:
Now, for default Farnell and Newark add all the user fields as your example for custom fields.

You can edit for just one specific in the `distributors_info.py` file.

closed 2020-05-19 by hildogjr

hildogjr, 2020-05-21:
The feature request is duplicated from #394

commit fc166406 (2020-05-18): Fix #393
commit 7724cee2 (2020-05-19): Fix #393