hildogjr/KiCost #394: Multipart component - pricing, datasheets, otherfields

mdeweerd, 2020-05-19:
I am using "KiCost v.1.1.2"

For a Fuse holder, I defined the holder, the fuse and the clip cover.

[image]

In the Excel file, I have three lines (ok), with the same pricing (nok), and the other fields are not "split" across the excel lines (desc, S1MN,S2PN,S1PN).

[image]

I also tried to add an empty price (kicost:we:pricing=1:EUR0.07;;1:EUR0.03) but that resulted in a fatal error.

Here is the test case:
[multipart.zip](https://github.com/xesscorp/KiCost/files/4652208/multipart.zip)

I did not test the most recent version as I do not want to break my setup at this moment.

hildogjr, 2020-05-19:
Personal distributors (for now) doesn't allow multiparts.

mdeweerd, 2023-04-27:
Stale, closing (limiting my issue backlog).

closed 2023-04-27 by mdeweerd