hildogjr/KiCost #395: Kitspace component information fetching fails

mdeweerd, 2020-05-20:
I've not been able to determine exactly what makes the following component entry fail:
[image]

It worked reasonably until I added the mouser reference and change the fuse reference.
But removing the mouser reference did no help and changing the fuse reference to one without spaces did not help either.

I upgraded to Kicost 1.1.4 

Test case & log:
[TEST.zip](https://github.com/xesscorp/KiCost/files/4656698/TEST.zip)

hildogjr, 2020-05-20:
Appear that we are experiencing some issue on Kitspace services. I am summon @kasbah to be aware.

At the long plan, would be nice have help of other developers to expand the APIs on KiCost, adding the limited and maybe payed ones.

mdeweerd, 2020-05-21:
@hildogjr Regarding the contribution of other developers. 

Here is some feedback regarding my experience - hopefully it can help to find ways to enable contributions.

- I worked on the code on my end in the past and by the time it was ready to share, the KiCost code structure itself changed -> efforts lost.  I've done something like that a second time, and another incompatibility occured.
- I fixed IMHO an issue in november 2019 and made a pull réquest.  It is still pending (BTW, if it is not going to be integrated in any way, I guess I should close it).
- I worked on the proposal of a logo as was requested, but that is also pending.
- I made suggestions with regards to the spreadsheet organisation and formulas, providing at least one spreadsheet demonstrating the proposal..  I am not sure what happened with that.

In my experience that is what happens in most cases where I submit suggestions for changes, while on the contrary I accept changes opensource that I created and implicitally manage.

This is why in most cases I participate to open source by filing reports - hopefully with sufficient details and test case so that they are easily repeatable.

With regards to my own opensource software, I notice that other developers exclusively develop parts that they need, while evolving the software generally is in my case "a passion" of the opensource owner based on his own needs for functions, exploring and meeting non-technical user requests.

So I am ok with helping.  How is another question, and "what" should be mostly related to my needs.  IMHO it should also be possible to get paid by some distributors to integrate their API because if they are not part of the ecosystem, they have less customers.  We are not only helping ourselves, we are also helping them.

kasbah, 2020-05-21:
Haven't read this whole thread but regarding Kitspace Partinfo API:

1. It hit a limit again, we currently only have 5000 requests a month shared between KiCost, Horizon EDA and Kitspace use. I have added a 1000 this month for non-Kitspace use. 
2. Ocotpart are beta testing a "free for open source" GraphQL API, this may be the best way forward for KiCost. I haven't had the time to figure out the best way to integrate with it myself, but I could put you in touch with someone from Octopart to ask for direct access.

hildogjr, 2021-01-23:
@kasbah, did we reach the limits again?

kasbah, 2021-01-24:
Another problem with the server actually. (Might be running out of RAM.)

hildogjr, 2021-02-01:
@kasbah, the error returned. Are we (KiCost users) making to much attempts? Do you have this statistic?

kasbah, 2021-02-02:
It's up again, sorry about that. I need to find some time to find out what's going on exactly. I don't have any statistics unfortunately.

hildogjr, 2021-02-02:
Thanks again @kasbah... we still test and fixing, no problem ;-)

Just to check, I am getting the follow log asking for specific LCSC stock numbers using the "match query" of your API (line 78 of `api_partinfo_kitspace.py` file).

Query used:
```
{'query': 'query ($input: [MpnOrSku]!){ match(parts: $input) {mpn{manufacturer,part},datasheet,description,specs{key,value},offers(from:["Digikey","LCSC","Mouser"]){product_url,sku{vendor,part},description,moq,in_stock_quantity,prices{EUR,USD,GBP}}} }','variables': '{"input":[{"sku":{"vendor":"LCSC","part":"C107644"}},{"mpn":{"manufacturer":"","part":"TL074IDT"}},{"sku":{"vendor":"LCSC","part":"C38423"}},{"mpn":{"manufacturer":"","part":"REF3030AIDBZT"}},{"sku":{"vendor":"LCSC","part":"C15850"}},{"mpn":{"manufacturer":"","part":"C0805C106K8PACTU"}},{"sku":{"vendor":"LCSC","part":"C15850"}},{"mpn":{"manufacturer":"","part":"C0805C106Z8VACTU"}},{"sku":{"vendor":"LCSC","part":"C504992"}},{"mpn":{"manufacturer":"","part":"0022232061"}},{"mpn":{"manufacturer":"","part":"0022012067"}},{"mpn":{"manufacturer":"","part":"08-50-0114"}},{"sku":{"vendor":"Digikey","part":"398-1010-ND"}},{"mpn":{"manufacturer":"","part":"LA55-P"}},{"sku":{"vendor":"Digikey","part":"398-1019-ND"}},{"mpn":{"manufacturer":"","part":"lv25-P"}},{"sku":{"vendor":"LCSC","part":"C218472"}},{"mpn":{"manufacturer":"","part":"RMCF0805FT27K0"}},{"sku":{"vendor":"LCSC","part":"C67869"}},{"mpn":{"manufacturer":"","part":"RMCF0805FT2K20"}}]}'} 
```

Error:
```
No information found for parts 'R205,R605,R805,R1005,R1205,R1405,R1605,R1805' query `{'sku': {'vendor': 'LCSC', 'part': 'C218472'}}`
```

I am making some mistake with the "LCSC" vendor name? (Should it be lower case?)

closed 2021-03-15 by hildogjr