hildogjr/KiCost #397: Fails on "inf" quantity

mdeweerd, 2020-05-22:
Import fails on the following line (1.1.4, api_partinfo_kitspace.py, 293), with "inf" quantity:
```python
    (part_qty_increment < part.qty_increment[dist]) or
```
I added debug information to the code, so the line number changed.  The contents of some variables is shown first:
```
part.qty_increment[dist] = 9
dist_part_num = 'BVEI3027015'
part_qty_increment = inf
dist = 'tme'
part.qty_increment = {'local_template': None, 'arrow': None, 'digikey': None, 'farnell': None, 'mouser': None, 'newark': None, 'rs': None, 'tme': None, 'lcsc': None, 'tianli': None, 'pcb': None, 'cosmtec': None, 'we': None}
part.qty_increment[dist] = None
Traceback (most recent call last):
  File "/usr/bin/kicost", line 11, in <module>
    load_entry_point('kicost==1.1.4', 'console_scripts', 'kicost')()
  File "/usr/lib/python3.7/site-packages/kicost/__main__.py", line 311, in main
    dist_list=dist_list, currency=args.currency)
  File "/usr/lib/python3.7/site-packages/kicost/kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "/usr/lib/python3.7/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 366, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "/usr/lib/python3.7/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 306, in get_part_info
    (part_qty_increment < part.qty_increment[dist]) or \
TypeError: '<' not supported between instances of 'float' and 'NoneType'
```

Test case.

hildogjr, 2020-05-22:
I could not reproduce.
I tried your XML file and a CSV with 'BVEI3027015' (kicost -wi file.csv --eda csv). Also, I am getting error when extraction your "test.log" file.
Could you provide more info?

hildogjr, 2020-05-22:
Founded!

mdeweerd, 2020-05-22:
I tried to follow the virtualenv instructions (https://xesscorp.github.io/KiCost/docs/_build/singlehtml/index.html ) but that didn't quite work.

Then I tried 
```
pip install virtualenv
virtualenv kicost
cd kicost
. bin/activate
python setup.py develop
```

`which kicost` shows a local path.  `kicost --version` stil indicates 1.1.4.  running "binkicost" give:
```
pkg_resources.DistributionNotFound: The 'lxml>=3.7.2' distribution was not found and is required by kicost
(kicost)

```

I tried installing using `pip install lxml`.  Installing lxml failed, I need to investigate that further...

mdeweerd, 2020-05-23:
Ok I managed to set up the development environment.  I needed to install some system libraries (for development), including python-devel .  We also need to use `python -m venv DEVVIRTUALENV` now and source "activate" in "DEVVIRTUALENV/bin/activate" .

That being done, the test case still fails with the latest git version - new log included:
[TEST.zip](https://github.com/xesscorp/KiCost/files/4671746/TEST.zip)

closed 2020-05-24 by hildogjr