hildogjr/KiCost #410: Master branch issue

leoheck, 2020-06-08:
Trying to install the master version, on Ubuntu 2020
Last commit: a02201ce4ba71a82cb9bfb6d81e82500fe86838a

I am seeing this issue
[image]

Also, maybe I don't have all the dependencies. Where I can find a dependency list?

hildogjr, 2020-06-08:
It should be list at https://github.com/xesscorp/KiCost/blob/master/setup.py `requirements`.

Could you report the command used to run the installation?

leoheck, 2020-06-08:
Sure!
I was using `sudo make install`
Do you have any recommendations? 

I didn't find any installation procedure in the README.md. It would be nice to have some instructions there too. Also, if it is possible to add a list of distributors, on the README and/or update the screenshot showing this list will be nice for new people coming here for the first time.

I want to use the master version to be able to use the https://lcsc.com/en service

Also, I tested the stable version 1.1.4 but I was not able to make it work, from the GUI. 
Then when I saw the command to run it from the command line, it worked fine with this command
```
kicost -w -i board.xml --debug 8
```

hildogjr, 2020-06-08:
Use the default Python installer `pip install setup.exe` (I also recommend use `pip3` instead `pip` to install the Python3 version instead of the decrepit, and not present at Ubuntu 20.04, Python2).

You could also use `pip install kicost` (that direct download from https://pypi.org/project/kicost/) and replace the installation directory with the files here in Git to get the new features and fixes.

The LCSC distributor is present at the PyPI version, just missing some fix
https://github.com/xesscorp/KiCost/blob/master/kicost/HISTORY.rst

leoheck, 2020-06-08:
Ok, Using a simple pip install worked fine.

Is it possible to add this to the README? I would recommend something like this:

```
### Install stable version
pip3 install kicost

### Install from GitHub version
git clone https://github.com/xesscorp/KiCost.git
cd KiCost
pip3 install .
```

hildogjr, 2020-06-08:
As tip, after install (and had give KiCad also a first run), execute `kicost --setup` to make the KiCad-KiCost integration. It should be automatic but is not working yet.

Also install the wxPython (`sudo pip3 install wxpython`) to have the GUI (Window dialog) version of KiCad. If not installed will run just as CLI (command/terminal). This also should be automatically, it related with the same issue above.

hildogjr, 2020-06-08:
Please @leoheck, if you feel fine to have a look, we miss some user revision at the documents. You can propose the changes directly.

closed 2020-06-08 by hildogjr

leoheck, 2020-06-08:
Good to hear that. Thanks for the feedback.