hildogjr/KiCost #412: GUI is not working...

leoheck, 2020-06-08:
Using (testing) v1.1.5
Using on Ubuntu 20.04

From the GUI

[image]
[image]

Still seeing this issue
[image]

Debuggin this with the command line given here it is working.

Maybe the issue is with the flag `Open spreadsheet`
So, if I unmark this checkbox, the whole gui hangs. 

A 3rd test gave this...
BOM file: board.csv
Reconigzed EDAs: CSV file
Open spreadsheet unmarked 
```
[3]    76634 segmentation fault (core dumped)  kicost
```

hildogjr, 2020-06-08:
Attach the answer to `kicost --info` on terminal.

leoheck, 2020-06-08:
```
✦2 ➜ kicost --info
KiCost v.1.1.5.at Python 3.8.2.on Linux-5.4.0-34-generic-x86_64-with-glibc2.29(64bit).Graphical library: 4.0.7 gtk3 (phoenix) wxWidgets
3.0.4.
```

hildogjr, 2020-06-08:
Confirmed. Also some freeze at Ubuntu 16.04.

patrickelectric, 2020-10-15:
Same problem with current master branch:
hash: c1ea5d401c3714af4c6922f23360de7616a53016
KiCost v.1.1.5.at Python 3.8.6.on Linux-5.8.14-arch1-1-x86_64-with-glibc2.2.5(64bit).Graphical library: 4.0.7.post2 gtk3 (phoenix) wxWidgets 3.0.4.

AlkalineZN, 2020-12-04:
KiCost v.1.1.4.at Python 3.8.5.on Windows-10-10.0.19041-SP0(64bit).Graphical library: 4.0.7.post2 msw (phoenix)
wxWidgets 3.0.5.

GasDotto, 2020-12-07:
Hi, I have the same issue.

The kicost --info output is:
`KiCost v.1.1.4.at Python 3.7.9.on Windows-10-10.0.19041-SP0(64bit).Graphical
library: 4.1.0 msw (phoenix) wxWidgets 3.1.4.`

The output of the script is:

```
Progress:   0%|                                                                              | 0/213 [00:00<?, ?part/s]Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\connection.py", line 160, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\util\connection.py", line 84, in create_connection
    raise err
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\util\connection.py", line 74, in create_connection
    sock.connect(sa)
TimeoutError: [WinError 10060] Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\connectionpool.py", line 381, in _make_request
    self._validate_conn(conn)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\connectionpool.py", line 978, in _validate_conn
    conn.connect()
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\connection.py", line 309, in connect
    conn = self._new_conn()
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\connection.py", line 172, in _new_conn
    self, "Failed to establish a new connection: %s" % e
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x0000019E7F767CC8>: Failed to establish a new connection: [WinError 10060] Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\connectionpool.py", line 727, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\urllib3\util\retry.py", line 439, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='dev-partinfo.kitspace.org', port=443): Max retries exceeded with url: /graphql (Caused by NewConnectionError("<urllib3.connection.HTTPSConnection object at 0x0000019E7F767CC8>: Failed to establish a new connection: [WinError 10060] Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato"))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\kicost\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\programdata\anaconda3\envs\kicost\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\ProgramData\Anaconda3\envs\kicost\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\kicost\__main__.py", line 311, in main
    dist_list=dist_list, currency=args.currency)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\kicost\kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\kicost\distributors\api_partinfo_kitspace.py", line 353, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\kicost\distributors\api_partinfo_kitspace.py", line 197, in get_part_info
    results = api_partinfo_kitspace.query(query, distributors)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\kicost\distributors\api_partinfo_kitspace.py", line 124, in query
    response = requests.post(QUERY_URL, {'query': query_type, "variables": variables})
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\requests\api.py", line 119, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\requests\api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\requests\sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\requests\sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "c:\programdata\anaconda3\envs\kicost\lib\site-packages\requests\adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
[... cut ...] "<urllib3.connection.HTTPSConnection object at 0x0000019E7F767CC8>: Failed to establish a new connection: [WinError 10060] Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato"))
Progress:   0%|                                                                              | 0/213 [00:21<?, ?part/s]
```
It seems that dev-partinfo.kitspace.org is not responding, it is down or there is a bug in kicost?

sliedes, 2020-12-28:
Ok, I believe a few issues are conflated here (especially since #421 was marked as a duplicate). I think I have an idea what's going on.

1. The issue in https://github.com/xesscorp/KiCost/issues/421#issuecomment-691255711: wxPython has become more strict about accepting nonsensical combinations of sizer flags. From the release notes of wxPython 4.1.1:

> This should have been mentioned in the notes for the last release, but alas, it wandered away and got lost. wxWidgets is now validating the flags passed when adding items to a sizer, to ensure that they are the correct flags for the type of the sizer. If the given flags do not make sense, for example using horizontal alignment flags in a horizontal box sizer, then a wxAssertionError error is raised.

... which is exactly what happens. The error messages even tell what the problem is ("wxEXPAND flag will be overridden by alignment flags"). The solution is to either remove the wx.EXPAND flag or the alignment flags, depending on what the intent here is. Getting the GUI to start required a modification in kicost_gui.py of lines 367, 384 and 449. I'm not sending a pull request since I don't know what the GUI is supposed to look like and hence which of the conflicting flags should be removed.

2. The xcb/XInitThreads error. What is happening here is that GUI code is being run from a thread different than the one that created the window. This is not supported in wxPython. Specifically, it seems that formKiCost.run() is run from a different thread (started in button_run()), and it contains GUI code such as `self.m_gauge_process.SetValue(0)` (and lots of other GUI manipulation). This causes breakage. See eg. [this thread](https://discuss.wxpython.org/t/xcb-unknown-request-in-queue-while-dequeuing/26830/1). Hacking button_run() to just call self.run() instead of starting a thread fixes this for me.

set-soft, 2021-03-14:
Hi @hildogjr !

If you incorporate the flake8 patch first (which I strongly recommend) the #451 becomes:

```diff
diff --git a/kicost/kicost_gui.py b/kicost/kicost_gui.py
index 231e11f..bb2dcb9 100644
--- a/kicost/kicost_gui.py
+++ b/kicost/kicost_gui.py
@@ -41,7 +41,7 @@ import webbrowser  # To update informations.
 import sys
 import os
 import subprocess  # To access OS commands and run in the shell.
-import threading
+from threading import Thread
 import time  # To elapse time.
 import tempfile  # To create the temporary log file.
 from datetime import datetime  # To create the log name, when asked to save.
@@ -96,6 +96,18 @@ PAGE_POWERED_BY = 'https://kitspace.org/'
 kicostPath = os.path.dirname(os.path.abspath(__file__))  # Application dir.
 
 
+# ======================================================================
+class ChildThread(Thread):
+    """ Helper class to safetly call the run action """
+    def __init__(self, myframe):
+        """Init Worker Thread Class."""
+        Thread.__init__(self)
+        self.myframe = myframe
+
+    def run(self):
+        wx.CallAfter(self.myframe.run)
+
+
 # ======================================================================
 def open_file(filepath):
     '''@brief Open a file with the default application in different OSs.
@@ -714,10 +726,9 @@ class formKiCost(wx.Frame):
     def button_run(self, event):
         ''' @brief Call to run KiCost.'''
         event.Skip()
-        # self.run()
-        # wx.CallLater(10, self.run) # Necessary to not '(core dumped)' with wxPython.
-        t = threading.Thread(target=self.run)  # , args=[self])
-        t.start()
+        self.child = ChildThread(myframe=self)
+        self.child.daemon = True
+        self.child.start()
 
     # ----------------------------------------------------------------------
     # @anythread
```

PR #451 (2021-03-15): Fixes unsafe multithread mechanism in button_run
- Fixes #412
- On Linux it fixes the error:
```
[xcb] Unknown sequence number while processing queue
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
python3: ../../src/xcb_io.c:269: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.
```
- The solution is just a copy & paste from StackOverflow:
  https://stackoverflow.com/questions/32410359/threading-in-wxpython
- From what I interpret the main point here is the use of wx.CallAfter
commit 6be96473 (2021-03-15): Fixes unsafe multithread mechanism in button_run

- Fixes #412
- On Linux it fixes the error:
```
[xcb] Unknown sequence number while processing queue
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
python3: ../../src/xcb_io.c:269: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.
```
- The solution is just a copy & paste from StackOverflow:
  https://stackoverflow.com/questions/32410359/threading-in-wxpython
- From what I interpret the main point here is the use of wx.CallAfter
closed 2021-03-15 by hildogjr