hildogjr/KiCost #413: Generate .ods file instead of .xlsx from the command line

leoheck, 2020-06-08:
Using v1.1.5

There are any way to export .ods from the command line instead of .xlsx?
Looking there in the --help, I could not find anything like that

hildogjr, 2020-06-08:
No, it is just a LibreOffice conversion.

this is given by the line 819 in the GUI code (just there and there is issue related with this execution #361):

    os.system('{e} --headless --convert-to ods {i} --outdir {o}'.format(e=libreoffice_executable, si=spreadsheet_file, o=os.path.dirname(spreadsheet_file)))

closed 2020-06-08 by hildogjr

hildogjr, 2020-06-08:
Reference: https://www.ubuntubuzz.com/2016/08/libreoffice-command-line-convert-multiple-files-xlsx-to-ods.html

leoheck, 2020-06-08:
> Yes, but it is the user using it, not the Kicost developer.

The UI should be simple. But ok, if this is your decision.

hildogjr, 2020-06-08:
Just because I are without really active programmers/developers. We have to focus the main features to keep it stable, before add some more. cosmetics.
The road map planned some drastic changes at the distributors module to allow more than one API.