hildogjr/KiCost #414: Documentation image broken

leoheck, 2020-06-09:
https://xesscorp.github.io/KiCost/docs/_build/singlehtml/index.html

[image]

leoheck, 2020-06-09:
It is really easier to read if you avoid justifying text like in the last paragraph.
Also, the Markdown is not being used there and this `[Kitspace](https://kitspace.org/)` doest work as expected.

set-soft, 2021-03-13:
The repo is very incosistent about the images.

I don't know which OS are you using @hildogjr . If you are using Linux I'll suggest using symlinks to avoid multiple copies of the same image.

The current status is:
- KitSpace logo:
  - ./kicost/kitspace.png
  - ./docs/kitspace.png
- GUI capture:
  - ./gui.png
  - ./docs/gui.png
- Block diagram:
  - ./docs/block_diag.png
  
The three images are used by README.rst and README.md. I know you want to remove README.md (redundant), but the relative path for the images is different when README.rst is used in `./` and `./doc`. This can be solved using symlinks.

I managed to solve what @leoheck reports, but the cost is that README.rst is more different than README.md. And I don't like the idea of creating a third copy of the KitSpace logo.

Using symlinks isn't good for Windows users. Windows currently supports links, but I found that at least one git client doesn't. What the client does is make another copy of the file.

If symlinks are good for you I can try to create a PR to solve it.

hildogjr, 2021-03-13:
I am using Linux-Ubuntu, but I never made a revision on the user manuals... I just added few sections concomitantly with the new features.

README.md was hard coded in the past by @xesscorp/@devbisme when some change on PyPI forced to use md as rst as default file. But we never really fix by deleting one of the formats.

leoheck, 2021-03-23:
@set-soft symlinks work on windows too. They are just not the default configuration. I am not sure it is needed here since is just a matter of adjusting duplicated references to use the right and single path.

leoheck, 2021-03-23:
Ah, since images are fixed on README.md may I close this issue?

hildogjr, 2021-03-23:
I think pages and documentation still using the .rst files. We just need to check and standardize this.

set-soft, 2021-05-28:
I'm closing this issue now the on-line docs also looks ok: http://xess.com/KiCost/docs/_build/singlehtml/index.html

set-soft, 2021-05-28:
Oh! BTW, I used symlinks. Any Windows user with problems just blame @leoheck :-)

closed 2021-05-28 by set-soft

leoheck, 2021-05-28:
Symlinks are beautiful, don't blame me if Windows sucks. I don't recommend it haha. 

However, it is possible to use symlinks on Windows nowadays, they are just not there by default.
https://www.howtogeek.com/howto/16226/complete-guide-to-symbolic-links-symlinks-on-windows-or-linux/

set-soft, 2021-05-28:
Hi @leoheck !
I used symlinks on Windows (hideous Power Shell syntax), the problem is if the git client will use them.
One of our clients did a mess in a GitLab repo where I used symlinks for KiCad footprints ...