hildogjr/KiCost #415: How to add multiples PartNumbers

leoheck, 2020-06-09:
I found this here:

```
Manufacturer’s part number field can now be labeled as ‘manf#’, ‘mpn’, ‘pn’, ‘#’, etc. (See documentation.)
```

[image]

From here: https://xesscorp.github.io/KiCost/docs/_build/singlehtml/index.html#id1

Which documentation that I should read to know how to use multiple part numbers for each part?

hildogjr, 2020-06-09:
This feature doesn't exist. And there is no plan to code it because, after "use next alternative" some users will ask for ship one e etc... And some of than need some change at the spreadsheet format.
A work around is use the multiple part case and choose, after at the spreadsheet.

robot-army, 2021-01-27:
Hmm, frustrated to see a 'wont fix' label on this. I'd love it if multiple MPNs per ref could be handled somehow.

robot-army, 2021-01-27:
Is the suggestion to use 'variants' or 'subparts' to handle this? Does anyone have a suggested workflow for this?

hildogjr, 2021-01-27:
I use both: variant and sub part (and I chose the subparts that I want after have the spreadsheet).

A multiple choice part will need a new spreadsheet format. At least to force the use to just pick one of the parts. Someone need to propose a viable solution.