hildogjr/KiCost #416: PartNumber with comma

leoheck, 2020-06-09:
I have one part number with a comma, does Kicost work with these?

https://analytics.supplyframe.com/trackingservlet/track/?r=gPVv3S5fz9WS0muSSENj7nlafrT_8Erv5whsN6w-DXL_H20hRDbiwip60DkNBJHrHZ6YZjVqTil-TDq94oCLDOhrCsTJAzbHfOLENgDycJAQcL91C5C6rrhcJR0Z4Kr4Ozf0FAC3cn1KZqYVQ188Kkv67nIqYH5FifL4n0SFIJK8UvqYruCY2e-h_hTNk_OU_UR8qz71GBuHQWG9zpuTWRLGS_TAceKsqUD81xFU7NpF_aFrRW0wTzNhyKjye3UKvlgiba2wZcifR7MTiL_-_s4onBhV5LhNLRZgr-gsV44r9od-tkljNBEuZGiYhL0wL6BxkXQOeNQlk_pWOdzQhUVagQzlAXdnEk5El7fV__oxchGdQQEZYNkP-UM29TpDPG_jGHdpwESxaJ8a-JZ1GvH1nDwhj_a_GJxHlp86kEnuKOsx1o33czQGLkab4H0AsndjVjdn9QmLAyzgSjO2w9hjmVeq4sBl-q_HyhbuuKHqBMJWeVRtQyR3hqbv5vFqin6KVPnbLxsq9eTSWCVkmI96FQ-i8GaKCfH9k_nNCVWOYkggrfwfrfxTIIs1oDyREwflQd1d--_tbvQe9k0P9DLLI090m2ZZYilJNpBQM9w2J1rnchjJHjsCcoKgcSN36SPaVyID-pDyJ6dK2ArPFeDINazI-KBYRx27hoeBNBLkujmS7LKRPQABJqxxo3chZuOU75G6wTkB0GZs7CkVn9DaID0baorM-nUeld-yfobIh5PnBll5WdPeG36b9jW2Wp3DmA3RH4Gr0yzOBnYChmjYfE6IZritjUA_wpXwfljneR_huJ4Hnxkbgs00wYU2gAIPjNg-CNeAFThcdJNXK-K_TJ3hkCwdMlnXrodBaCwQoCqHHIoPdak1FGINuXm5MyeutYlt4-fnKpkkxyDOSTD1kL2yHDlmnuj5X5pxV6A

romain145, 2020-06-10:
I think the workaround is to protect then between quotes:
[image]

Although now that I'm testing it, looks like it's working anymore...

```
C:\Users\Romain\projects\jal>kicost -i jal.xml
Progress:   0%|                                                                               | 0/42 [00:00<?, ?part/s]Traceback (most recent call last):
  File "C:\Users\Romain\AppData\Local\Programs\Python\Python38-32\Scripts\kicost-script.py", line 11, in <module>
    load_entry_point('kicost==1.1.4', 'console_scripts', 'kicost')()
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\__main__.py", line 306, in main
    kicost(in_file=args.input, eda_name=args.eda,
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\distributors\api_partinfo_kitspace.py", line 353, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\distributors\api_partinfo_kitspace.py", line 197, in get_part_info
    results = api_partinfo_kitspace.query(query, distributors)
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\distributors\api_partinfo_kitspace.py", line 133, in query
    raise Exception('Bad request to Kitspace server probably due to an incorrect string format check your `manf#` codes and contact the suport team.')
Exception: Bad request to Kitspace server probably due to an incorrect string format check your `manf#` codes and contact the suport team.
Progress:   0%|                                                                               | 0/42 [00:00<?, ?part/s]

```

mdeweerd, 2020-06-10:
Yes, I use a `BT139B-600,118` and you need to add a backslash `BT139B-600\,118`

hildogjr, 2020-06-10:
As @mdeweerd said:

https://xesscorp.github.io/KiCost
> Be careful if the part have a manf# or distributor code with the `,`, `;` or `\ characters`, these have specific propose as shown in Parts With Subparts. They have to be preceded by `\`, example, for the `NUP1301,215` code, use `NUP1301\,215`.

closed 2020-06-10 by hildogjr