hildogjr/KiCost #417: Is it possible skip instead of break when a url is not working?

leoheck, 2020-06-09:
Using v1.1.5

```
# Getting part data from KitSpace...
Starting new HTTPS connection (1): dev-partinfo.kitspace.org:443                                                                              
https://dev-partinfo.kitspace.org:443 "POST /graphql HTTP/1.1" 504 192                                                                        
Progress:   0%|                                                                                                      | 0/52 [01:01<?, ?part/s]Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/kicost/__main__.py", line 306, in main
    kicost(in_file=args.input, eda_name=args.eda,
  File "/usr/local/lib/python3.8/dist-packages/kicost/kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "/usr/local/lib/python3.8/dist-packages/kicost/distributors/api_partinfo_kitspace.py", line 366, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "/usr/local/lib/python3.8/dist-packages/kicost/distributors/api_partinfo_kitspace.py", line 208, in get_part_info
    results = api_partinfo_kitspace.query(query, distributors)
  File "/usr/local/lib/python3.8/dist-packages/kicost/distributors/api_partinfo_kitspace.py", line 146, in query
    raise Exception('One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.')
Exception: One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.
Progress:   0%|                                                                                                      | 0/52 [01:01<?, ?part/s]
```

leoheck, 2020-06-09:
I cant use the GUI, so, does kicost has any flag to help here?

mdeweerd, 2020-06-10:
I always run with ` --debug 10` to get some hint at which part is failing.

The number of requests to Kitspace is shared amongst the users.
Therefor, I also recommend using  `--no_price` while you are cleaning up your BOM and check prices only later when you are confident that you part numbers and variants are correctly set up.

miili, 2020-08-02:
Is there an insight how many requests are available and in the pool?

set-soft, 2021-05-24:
> Is there an insight how many requests are available and in the pool?

This is a question for @kasbah and I have another:
As we are discussing it: Octopart API (what KitSpace uses) replies with 429 (Too many requests) when you reach the limit.
What @leoheck  posted seems to be a 504 (Gateway timeout) error.
Does KitSpace also reply with 429 when Octopart says 429 to KitSpace?
I can add a case for 429 to KiCost.

About the `--price`: I don't agree, the KitSpace queries are really slow, the user should get bored and choose `--no_price`. Is in the `--help` and you can deselect all distributors in the GUI. The main point of KiCost is the price. If you get an empty spreadsheet you'll think the tool doesn't work. In fact I want to make KiCost fill the cells to avoid the impression that they are empty (until you force a recomputation).

And about the stack trace: Now KiCost shows a red error and no stack trace. So the main issue here (friendly message) is solved. If KitSpace can report 429 this will be much better.

kasbah, 2021-05-24:
The pool is currently 5000 requests per month. 

We don't yet give a proper response when we hit the limit. We just error and then let it timeout. The most [relevant code is in our API is here](https://github.com/kitspace/partinfo/blob/f80a9da8cd1e790f8177a66e5df7c91c0b1fe870/src/octopart.js#L61-L95). I've not gotten around to adding the the ability to give a proper response there.

leoheck, 2021-05-24:
> About the --price: I don't agree, 

Sure, @set-soft if the tool works every time the user will use it, I agree. But since the tool does not work anymore after a couple of tries, this can be an issue. This happened was for me when trying to use it for the first time. Since I was testing and experimenting to make it work, I had to run it too many times. And then I realized that right after being able to use it, I could not use it anymore since I could not make more requests to that server.  

This may also be a problem if you are working in a place where you have many people using the tool and you just have a single external IP, as a normal company is.

The idea for an inverted flag is just to make sure you will be able to run when needed.

leoheck, 2021-05-24:
> What @leoheck posted seems to be a 504 (Gateway timeout) error.
Does KitSpace also reply with 429 when Octopart says 429 to KitSpace?
I can add a case for 429 to KiCost.

Well, this can be right. This issue was created one year ago.

kasbah, 2021-05-24:
What do you mean by "this can be right" ?

leoheck, 2021-05-24:
This means that @set-soft might be right assuming that because I have no idea why it was not working for me after using for a couple of times that time.

mdeweerd, 2021-05-24:
> About the `--price`: I don't agree, the KitSpace queries are really slow, the user should get bored and choose `--no_price`. 

Actually, one thing about kicost is the price...  Getting nices BOMs with the variants feature is a pretty strong feature as well.  
I am often surprised about how lazy a user is to examine the manual and does not seek to optimize because "it works".  
I am in favor of a `--price` option rather than a `--no_price` option.  
This could be a question a the end of reading in the files for instance:
```
KiCost V...
All files were read succesfully.
10 out of 23 components are missing a manufacturer number:
- R1, 10k,  [...]
- C10, 10u, [...]
- ...

Continue with price calculation ? (the number of API calls is limited, please make fair use of it!) ?
You can add `--price` as an option or Type "PRICE" to compute it now (PRICE/<ENTER>):
```

set-soft, 2021-05-24:
> > About the --price: I don't agree,
> 
> Sure, @set-soft if the tool works every time the user will use it, I agree. But since the tool does not work anymore after a couple of tries, this can be an issue. 

I created the regression tests without this problem. And they are far more than a couple ;-)

Lamentably Octopart asks for a credit card to give a free key. I don't want to take the risk of being "accidentally" charged, so I didn't asked for a free key. But you can try it. They give 500 queries a month for free.

Another option to offload KitSpace could be the implementation of the APIs for the major distributors. Digi-key provides a free key with 1000 queries **by day** this is more than enough for regular use. The API is already implemented in Python https://github.com/peeter123/digikey-api   I'm doing a lot of work on KiCost and I'm very delayed with the KiBot integration, so I can't say I'll add Digi-Key support.

We can try adding a warning, enabled by default and disabled using an environment variable (or a command line option for people too lazy to define environment variables). Hey @hildogjr ! What do you think about such a warning? Something like

```
KitSpace resources are limited, please include prices only after you finished fine tunning the BoM.
In the meantime use --no_price (command line) or deselect all distributors (GUI).
```

set-soft, 2021-05-24:
> The pool is currently 5000 requests per month.

Thanks for the info

> We don't yet give a proper response when we hit the limit. We just error and then let it timeout. The most [relevant code in our API is here](https://github.com/kitspace/partinfo/blob/f80a9da8cd1e790f8177a66e5df7c91c0b1fe870/src/octopart.js#L61-L95). I've not gotten around to adding the the ability to give a proper response there.

I don't know enough Java Script to help with it. If you manage to reply 429 just tell me and I'll add a nice message to KiCost explianing the probem.