hildogjr/KiCost #418: SPAM From CIRCUITHUB !

mdeweerd, 2020-07-03:
I am reporting this as an issue to inform contributers about the abuse of the contact information left in the project commit information.   This is almost certainly not the responsibility of KiCost, Kitspace or Github.

I was contacted by Circuithub on the address used for my commits to invite me to use their services.
In Europe this is against GDPR regulations.

I filed a request with them regarding my personal data and they replied.  I am not expecting to received more detailed information about the company that provided them with this information that was collected based on their explicit request.

I now have to check how to remove this information from my git commits and I have already taken action to disable the email address I used.

hildogjr, 2020-07-04:
I had some annoying spams related with some information that I used in a post academic article.

I never got something that appear related with my GitHub data, I know that's possible get the signed email on commits but hope that is not the case.

mdeweerd, 2020-07-04:
In this case my email address was taken from a github repository and very likely KiCost.  I did not use that email only for github repositories.

It's simple to get the emails of authors `git log --all  | grep -i "Author:" | sort -u` .

I have now cleaned all the repositories and gists that I have control over using a script like this:

```
#!/bin/bash

# Note: Need to `pip install git-filter-repo`
# Several lines commented in this script

#Example of GIST:  REPO=git@gist.github.com:911b254b5c2d7ca17db7.git
# Example with KiCost:
REPO=git@github.com:mdeweerd/KiCost.git

# Get the base name from the repository
BASE=$(basename $REPO)
# Remove up to ':' include (was required for gists that do not have '/')
BASE=${BASE#*:}

if [[ ! -r ""$(basename $REPO) ]] ; then
  # Get the remository only if we did not get it yet
  git clone --bare $REPO
fi
cd "./"$BASE

# If test allows disabling this step by changing to "0 == 1"
if [[ 1 == 1 ]] ; then
  git-filter-repo --name-callback 'return name.replace(b"USER1", b"NEWUSERNAME")' \
      --email-callback 'return email.replace(b"MAIL1@example.com", b"")'
  git-filter-repo --name-callback 'return name.replace(b"USER2", b"NEWUSERNAME")' \
      --email-callback 'return email.replace(b"MAIL2@example.com", b"")'
  # ...
  # Finished in my case with a line without "--name-callback"
  git-filter-repo --email-callback 'return email.replace(b"MAILN@example.com", b"")'
fi

# Push only if you are certain it works (after validating your filters at least once)
```

hildogjr, 2020-07-04:
@devbisme, that do you think? Do we proceed?

xesscorp, 2020-07-04:
From what I've read:
1. There was a single spam message from CircuitHub.
2. The email account to which that email was sent has been disabled.
3. The user has removed their email from future Github commits.

Why is there a need to do anything else?

hildogjr, 2020-07-04:
The old commits on master still with the e-mail info. It possible to remove from them too but quite laborious.

xesscorp, 2020-07-04:
But the user has disabled their old email address, so why remove it?

hildogjr, 2020-07-04:
Agreed. I will close this ant keep the mdeweerd warning of spams in mind.

closed 2020-07-04 by hildogjr