hildogjr/KiCost #421: KiCost 1.1.4, KiCAD 5.1.6-1 Install error under Windows 10 Pro x64

craigarno, 2020-08-12:
Saw the KiCost video, was really excited until I ran into problems I don't know how to solve in the Python environment.  Help is requested.

After installing KiCost 1.1.4, KiCost won't start.

```
C:\WINDOWS\system32>kicost --info
KiCost v.1.1.4.at Python 3.7.3.on Windows-10-10.0.18362-SP0(64bit).Graphical
library: 4.1.0 msw (phoenix) wxWidgets 3.1.4.
```

KiCost 1.1.4 won't start.
```
PS C:\Users\craig\Desktop\PicoTAP> kicost                                                                                                                                                                    Traceback (most recent call last):
  File "C:\Users\craig\AppData\Local\Programs\Python\Python37\Scripts\kicost-script.py", line 11, in <module>
    load_entry_point('kicost==1.1.4', 'console_scripts', 'kicost')()
  File "c:\users\craig\appdata\local\programs\python\python37\lib\site-packages\kicost-1.1.4-py3.7.egg\kicost\__main__.py", line 253, in main
    kicost_gui() # Use the user gui if no input is given.
  File "c:\users\craig\appdata\local\programs\python\python37\lib\site-packages\kicost-1.1.4-py3.7.egg\kicost\kicost_gui.py", line 1065, in kicost_gui
    frame = formKiCost(None)
  File "c:\users\craig\appdata\local\programs\python\python37\lib\site-packages\kicost-1.1.4-py3.7.egg\kicost\kicost_gui.py", line 367, in __init__
    fgSizer1.Add(self.m_gauge_process, 1, wx.ALL|wx.EXPAND|wx.ALIGN_CENTER_HORIZONTAL|wx.ALIGN_CENTER_VERTICAL, 5)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL)) || !(flags & (wxALIGN_RIGHT | wxALIGN_CENTRE_HORIZONTAL))" failed at ..\..\src\common\sizer.cpp(1453) in wxGridSizer::DoInsert(): wxEXPAND flag will be overridden by alignment flags
```

During install I see lots of
```
    def Run(self):
    ^
SyntaxError: import * only allowed at module level
```

MS Windows 10:
Fresh install of KiCAD 5.1.6-1, then open CMD Window as Admin:

How to reproduce:
```
> C:\WINDOWS\system32>easy_install kicost
> Searching for kicost
> Reading https://pypi.org/simple/kicost/
> Downloading https://files.pythonhosted.org/packages/62/64/27e7153c1ad4c3d80ebb12b889f54bc36cdbc573774ab2601970b561d150/kicost-1.1.4.tar.gz#sha256=acf68ca75216986899acabe07b535ebb9cade0877740aa8892a48c42e6afc559
> Best match: kicost 1.1.4
> Processing kicost-1.1.4.tar.gz
> Writing C:\Users\craig\AppData\Local\Temp\easy_install-nyt81s24\kicost-1.1.4\setup.cfg
> Running kicost-1.1.4\setup.py -q bdist_egg --dist-dir C:\Users\craig\AppData\Local\Temp\easy_install-nyt81s24\kicost-1.1.4\egg-dist-tmp-ojiqk9xa
> warning: no files found matching 'block_diag.png'
> warning: no previously-included files matching '__pycache__' found under directory '*'
> warning: no previously-included files matching '*.py[co]' found under directory '*'
>   File "build\bdist.win-amd64\egg\kicost\kicost_kicadplugin.py", line 56
>     def Run(self):
>     ^
> SyntaxError: import * only allowed at module level
> 
> creating c:\users\craig\appdata\local\programs\python\python37\lib\site-packages\kicost-1.1.4-py3.7.egg
> Extracting kicost-1.1.4-py3.7.egg to c:\users\craig\appdata\local\programs\python\python37\lib\site-packages
>   File "c:\users\craig\appdata\local\programs\python\python37\lib\site-packages\kicost-1.1.4-py3.7.egg\kicost\kicost_kicadplugin.py", line 56
>     def Run(self):
>     ^
> SyntaxError: import * only allowed at module level
> 
> Adding kicost 1.1.4 to easy-install.pth file
> Installing kicost-script.py script to c:\users\craig\appdata\local\programs\python\python37\Scripts
> Installing kicost.exe script to c:\users\craig\appdata\local\programs\python\python37\Scripts
> 
> Installed c:\users\craig\appdata\local\programs\python\python37\lib\site-packages\kicost-1.1.4-py3.7.egg
> Processing dependencies for kicost
> Searching for babel>=2.6
> Reading https://pypi.org/simple/babel/
> Downloading https://files.pythonhosted.org/packages/15/a1/522dccd23e5d2e47aed4b6a16795b8213e3272c7506e625f2425ad025a19/Babel-2.8.0-py2.py3-none-any.whl#sha256=d670ea0b10f8b723672d3a6abeb87b565b244da220d76b4dba1b66269ec152d4
> Best match: Babel 2.8.0
> Processing Babel-2.8.0-py2.py3-none-any.whl
[... 786 lines cut ...]
Requirement already satisfied: numpy; python_version >= "3.0" in c:\users\craig\appdata\local\programs\python\python37\lib\site-packages (from wxpython) (1.17.4)
Requirement already satisfied: pillow in c:\users\craig\appdata\local\programs\python\python37\lib\site-packages (from wxpython) (7.2.0)
Requirement already satisfied: six in c:\users\craig\appdata\local\programs\python\python37\lib\site-packages (from wxpython) (1.15.0)

C:\Program Files\KiCad\bin>kicost
Traceback (most recent call last):
  File "C:\Program Files\KiCad\bin\kicost-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pkg_resources/__init__.py", line 3250, in <module>
    @_call_aside
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pkg_resources/__init__.py", line 3234, in _call_aside
    f(*args, **kwargs)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pkg_resources/__init__.py", line 3263, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'soupsieve>1.2' distribution was not found and is required by beautifulsoup4
```
This is a new design created using KiCAD 5.1.6-1 under Windows 10.  Schematic done, aiming for PCB layout.

I'm stumped.  Any help to get KiCost running will be appreciated.  KiCost looks like a really amazing time saver.

hildogjr, 2020-08-12:
Related with #412 

We are right with low people number to fix all the issues.

You still can use by the CLI commands: `kicost -i FILE.xml`.

Also, try to install using `pip kicost` instead `easy_install kicost`, there is a old issue (I think not present anymore).

craigarno, 2020-08-12:
I understand not enough people/hours in day to fix things...

Tried **pip kicost** and **easy_install kicost**.  Both say everything is fine, then KiCost doesn't work.  KiCAD and the schematic in eeschema look great on both systems.

I tried **kicost -i PicoTAP.xml** and end up with a spreadsheet containing BOM information, but not part vendor/cost information from distributors.

```
craig@pluto:~/Project/PicoTAP% kicost -i PicoTAP.xml --overwrite -o PicoTAP.xlsx
No 'manf#' and 'rs#' field in any part: no information by 'RS Components'.
No 'manf#' and 'newark#' field in any part: no information by 'Newark'.
No 'manf#' and 'tme#' field in any part: no information by 'TME'.
No 'manf#' and 'arrow#' field in any part: no information by 'Arrow'.
No 'manf#' and 'farnell#' field in any part: no information by 'Farnell'.
No 'manf#' and 'digikey#' field in any part: no information by 'Digi-Key'.
No 'manf#' and 'mouser#' field in any part: no information by 'Mouser'.
No 'manf#' and 'lcsc#' field in any part: no information by 'LCSC'.
Progress: 0part [00:00, ?part/s]
craig@pluto:~/Project/PicoTAP%
```

I just tried kicost under Ubuntu 16.04 LTS.  Similar if not the same problems:
craig@pluto:~/Project/PicoTAP% kicost --info
KiCost v.1.1.4.at Python 2.7.12.on Linux-4.4.0-186-generic-x86_64-with-
Ubuntu-16.04-xenial(64bit).Graphical library: 3.0.2.0 gtk2 (classic).

```
craig@pluto:~/Project/PicoTAP% kicost                               
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/kicost/__main__.py", line 253, in main
    kicost_gui() # Use the user gui if no input is given.
  File "/usr/local/lib/python2.7/dist-packages/kicost/kicost_gui.py", line 1065, in kicost_gui
    frame = formKiCost(None)
  File "/usr/local/lib/python2.7/dist-packages/kicost/kicost_gui.py", line 540, in __init__
    self.set_properties()
  File "/usr/local/lib/python2.7/dist-packages/kicost/kicost_gui.py", line 866, in set_properties
    self.m_comboBox_currency.Insert(currencyList, 0)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 12875, in Insert
    return _core_.ItemContainer_Insert(*args, **kwargs)
TypeError: String or Unicode type required

```

KiCost seems an important enough feature that I wonder if you could convince the KiCAD development team to support KiCost, kind of like Thunderbird did with Lightning calendar?  KiCost is just as, if not more important, than 3D viewing.  KiCost ranks up there with DRC and Annotation in my book.

hildogjr, 2020-08-13:
The first error appear that kicost is not recognizing your manucture code field.

Did you fill some "manf#" field at Eschema?

hildogjr, 2020-08-13:
Kicost support some distributors and right now depends on KiSpace service or Octopart account.

I think it dependency doesn't fill on KiCad idea.

sslupsky, 2020-09-11:
I think I am seeing the same error of macOS:
```
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/kicost/__main__.py", line 253, in main
    kicost_gui() # Use the user gui if no input is given.
  File "/usr/local/lib/python3.7/site-packages/kicost/kicost_gui.py", line 1065, in kicost_gui
    frame = formKiCost(None)
  File "/usr/local/lib/python3.7/site-packages/kicost/kicost_gui.py", line 367, in __init__
    fgSizer1.Add(self.m_gauge_process, 1, wx.ALL|wx.EXPAND|wx.ALIGN_CENTER_HORIZONTAL|wx.ALIGN_CENTER_VERTICAL, 5)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL)) || !(flags & (wxALIGN_RIGHT | wxALIGN_CENTRE_HORIZONTAL))" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/common/sizer.cpp(1453) in DoInsert(): wxEXPAND flag will be overridden by alignment flags
```

kicost --info:
```
KiCost v.1.1.4.at Python 3.7.7.on
Darwin-19.6.0-x86_64-i386-64bit(64bit).Graphical library: 4.1.0 osx-cocoa
(phoenix) wxWidgets 3.1.4.
```

sslupsky, 2020-09-11:
I am however able to run it from the command line:
```
No information found for parts 'Q1,Q3,Q5' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'Si2333DDS'}}`                                                      
No information found for parts 'F1-F6' query `{'mpn': {'manufacturer': 'Bel Fuse', 'part': '0ZCK0035FF2G'}}`                                                    
No information found for parts 'R2-R8,R15,R21,R36' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW0402499KFK'}}`                                        
No information found for parts 'R9-R12,R37,R38,R46' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW04021M00FK'}}`                                       
No information found for parts 'R17' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW0402357KFK'}}`                                                      
No information found for parts 'R18' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW0402274KFK'}}`                                                      
No information found for parts 'Q2,Q4,Q6-Q8' query `{'mpn': {'manufacturer': 'Diodes Inc.', 'part': 'DMG2302UK-7'}}`                                            
No information found for parts 'R16,R22' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW06030000Z0'}}`                                                  
No information found for parts 'D1-D3' query `{'mpn': {'manufacturer': 'Diodes Inc.', 'part': 'D4V5H1U2LP1610-7'}}`                                             
No information found for parts 'R1' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW0402191KFK'}}`                                                       
No information found for parts 'U4-U8,U16' query `{'mpn': {'manufacturer': 'Texas Instruments', 'part': 'TPS22917DBVR'}}`                                       
No information found for parts 'U1#2,U2#2,U3#2' query `{'mpn': {'manufacturer': 'Nexperia', 'part': '125'}}`                                                    
No information found for parts 'J3-J6' query `{'mpn': {'manufacturer': 'JST Connector', 'part': 'B2B-PH-K-S'}}`                                                 
No information found for parts 'J1,J2' query `{'mpn': {'manufacturer': 'Sullins Connector Solutions', 'part': 'PPPC141LFBN-RC'}}`                               
No information found for parts 'U9-U14' query `{'mpn': {'manufacturer': 'Diodes Inc.', 'part': 'DT2041-04SO-7'}}`                                               
No information found for parts 'J7,J8' query `{'mpn': {'manufacturer': 'JST Connector', 'part': 'B3B-PH-K-S'}}`                                                 
No information found for parts 'R23,R24,R27' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW040210K0FK'}}`                                              
No information found for parts 'J10' query `{'mpn': {'manufacturer': 'JST Connector', 'part': 'BM08B-SRSS-TB'}}`                                                
No information found for parts 'U15' query `{'mpn': {'manufacturer': 'Texas Instruments', 'part': 'TPS62740DSSR'}}`                                             
No information found for parts 'R31' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW040210M0FK'}}`                                                      
No information found for parts 'U19' query `{'mpn': {'manufacturer': 'Fujitsu', 'part': 'MB85RS1MTPNF-G'}}`                                                     
No information found for parts 'U20' query `{'mpn': {'manufacturer': 'Texas Instruments', 'part': 'TMP116AIDRVR'}}`                                             
No information found for parts 'U21,U22' query `{'mpn': {'manufacturer': 'Texas Instruments', 'part': 'ADS1115IDGS'}}`                                          
No information found for parts 'R49-R56' query `{'mpn': {'manufacturer': 'Vishay', 'part': 'CRCW0402100RFK'}}`                                                  
No information found for parts 'J12' query `{'mpn': {'manufacturer': 'METZ Connect', 'part': 'PR20B05VBDN'}}`                                                   
Progress: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████| 37/37 [00:11<00:00,  3.20part/s]
```

Are those queries returning "No information found" specific to a vendor or does it mean no info from any vendor?

Each part has a Digikey PN already so I will need to investigate why I am seeing the query problem.

hildogjr, 2020-09-11:
Some problems can happens when the manufacture name on schematic is slightly different from real one (Vishay/vishay). `--ignore_fields manf` on KiCost call could some it.

I just catch in your log:

    No information found for parts 'U1#2,U2#2,U3#2' query `{'mpn': {'manufacturer': 'Nexperia', 'part': '125'}}`

Have a look on https://github.com/xesscorp/KiCost/issues/416

sslupsky, 2020-09-11:
Yeah, that line is weird.  There are actually two lines in the generated excel file:
```

U1#1,U2#1,U3#1 | 74AUP1G98 - p1/2 |   | SOT-363_SC-70-6 | Nexperia | 74AUP1G98GW | 300 | $0.00 | $0.00 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
U1#2,U2#2,U3#2 | 74AUP1G98 - p2/2 |   | SOT-363_SC-70-6 | Nexperia | 125 | 300 | $0.00 | $0.00 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  

```

I am not sure why.  Those parts are single "gate" devices.  So, one part per package.  Looking at the line in the excel file seems to suggest that there are multiple parts per device package?

hildogjr, 2020-09-11:
https://github.com/xesscorp/KiCost/issues/416#issuecomment-641929079

hildogjr, 2020-09-11:
Duplicated of #412

closed 2020-09-11 by hildogjr

sslupsky, 2020-09-11:
@hildogjr Yeah, I see that now.  You can escape the comma with a backslash.  I was wondering if that comma was going to cause a problem somewhere ;-)  Thank you.

hildogjr, 2020-09-11:
The comma is intended to create parts will "multiple parts", useful for connectors and other assembled parts.

sslupsky, 2020-09-11:
Unfortunately, that creates a problem when using the other BOM tools ... they do not seem to recognize the escape and so the back slash is included the part number.

sslupsky, 2020-09-11:
Seems like a KiCAD issue.

hildogjr, 2020-09-11:
Other BOM tools will also not recognize the multiple part system.

So, maybe, que KiCost need some improvements in this cases.

sslupsky, 2020-09-11:
Ah, ok.  So, the comma in the part name is a KiCost feature, not an issue with KiCAD itself.  Maybe KiCost could use a dedicated field for this purpose instead of using the part number?

hildogjr, 2020-09-11:
Questions:
1. Which field? Should be expansible for N multiple parts.
2. How it will work? The actual system allow multiple parts with different quantities.
3. Who could propose the implementation? We are lower of programmers.

sslupsky, 2020-09-11:
The user could specify the field with a command line parameter?  Something like "--multipartfield" or something that makes sense.  I'm not too familiar with the feature so do not have a good suggestion on exactly what to call it.

Is there a kicost.ini file?  The KiBOM guys use an ini file for configuration of the tool for the project.  So, for instance, you specify which field to use for the "do not fit" field in the ini file.  They also use a field for linking to a part on the Digikey website.  I've included a portion of the ini file below.  The fit_field option in this example is "Config".

I think the feature would work the same way you do it now.  I do not see any need to change the functionality.  Just use a custom field specified by the user.

```
[BOM_OPTIONS]
; General BoM options here
; If 'ignore_dnf' option is set to 1, rows that are not to be fitted on the PCB will not be written to the BoM file
ignore_dnf = 1
; If '{opt}' option is set to 1, the schematic version number will be appended to the filename.
include_version_number = 1
; Field name used to determine if a particular part is to be fitted
fit_field = Config
; Make column to contain links to the Digi-Key web page for a component. (Note: Digi-Key links will only be generated for the HTML output format)
digikey_link = Digikey PN
; Specify output file name format, %O is the defined output name, %v is the version, %V is the variant name which will be ammended according to 'variant_file_name_format'.
output_file_name = %O_bom_%v%V
; Specify the variant file name format, this is a unique field as the variant is not always used/specified. When it is unused you will want to strip all of this.
variant_file_name_format = _(%V)
; Default number of boards to produce if none given on CLI with -n
number_boards = 1
; Default PCB variant if none given on CLI with -r
board_variant = "default"
; When set to 1, suppresses table/column headers and legends in the output file.
; May be useful for testing purposes.
hide_headers = 0
; When set to 1, PCB information (version, component count, etc) is not shown in the output file.
; Useful for saving space in the HTML output and for ensuring CSV output is machine-parseable.
hide_pcb_info = 0
```

hildogjr, 2020-09-11:
This INI file was a good propose in the past. KiCost started to require a lot of parameters (in not straightforward use).

In that time, I made the GUI (with memory of the parameters). But it is now broken.

sslupsky, 2020-09-11:
If one could ignore the "do not fit" components, that would be helpful as well.  Or better yet, use a different field for "do not cost" and these components would be ignored and not included in the spreadsheet.

hildogjr, 2020-09-11:
The first already exists `--ignore_fileds NOPOP`.

sslupsky, 2020-09-11:
I misunderstood that --ignore_fields just ignored the field, not the entire part?

hildogjr, 2020-09-11:
Yes, correct just the field. But ignoring the DNP/NOPOP (filed used for not populated information yours), the part will be listed in the spreadsheet.

hildogjr, 2020-09-11:
Yes. NOPOP / DNP fields as 1.
KiCad Nightly ans next v6 will already provide this information (not as field). Until now, if NOPOP / DNP are defined, the part is excluded from spreadsheet. Butt will be include with used `--ignore_fields` to ignore this info.