hildogjr/KiCost #423: Failed to find specfici parts

sebo83910, 2020-09-04:
Hello,

First of all, thanks for this nice piece of software. It's very useful!
I'm using a csv bom file and version 1.1.15 of kicost.
I'd like to report two specific parts for which the tool is unable to complete part search:

- BAT54H,115: diode from Nexperia. If I replace the ',' by '-' it finds it but it's not the correct manf# information. I guess a simple regexp in the manf# parse could solve this issue.
- 9551670000000: DB9 connector. Excel somehow represents the number in exponential (9.55E+12) and fails to search. 

Thanks.
Sébastien

hildogjr, 2020-09-04:
See section **Parts With Subparts** in https://xess.com/KiCost/docs/_build/singlehtml/index.html.
The `,` is used to indicate multiple manufacture part numbers. If it is present a single part number, it must be preceded by escape character: `\,`.

The other issue, please attache a BOM example and the commands execute to call KiCost.

sebo83910, 2020-09-04:
github won't let me upload a csv file... :(
here is the xlsx:
[github-issue423.xlsx](https://github.com/xesscorp/KiCost/files/5175728/github-issue423.xlsx)

In the csv, here are the values used:
manf# | refs
-- | --
9.55167E+12 | J6

The DB9 connector [(Mouser ref)](https://www.mouser.fr/ProductDetail/HARTING/09551667812741?qs=%2Fha2pyFadugXu7EeJ5ifqc%2FJRmli2xNtqUs0vKT5IAlL2NeX%252Bk5ZOA%3D%3D) is 9551667812741 and I used this value within the csv file. 
Excel shows it in scientific mode.

sebo83910, 2020-09-04:
The command line is the following: 
`python3 -m kicost -i github-issue423.csv --exclude arrow lcsc tme --overwrite`

hildogjr, 2020-09-04:
Is this spreadsheet generated by a CSV or XML file?
I could not reproduce this on Linux-LIbreOffice.

Please, attach the file example (changing the extension for GitHub ignore it).

sebo83910, 2020-09-07:
Here it is.
The file has been created with MS Excel 2016. 
[github-issue423.csv.xlsx](https://github.com/xesscorp/KiCost/files/5181478/github-issue423.csv.xlsx)

closed 2021-05-05 by set-soft

set-soft, 2021-05-05:
The problem here seems to be a wrong part number. The correct part number is "09551667812741", not "9551667812741". Using the correct value I get data from KitSpace API.

KiCost writes it as an string, I guess the "9.55167E+12" comes from some mistake when importing the CSV in Excel. I tried loading the KiCost XLSX in Excel from Office 365 and it shows the correct P/N.

commit e5d3f089 (2021-05-05): Better #423 test

- Added test using the correct manf#
- Added tests to check we send the correct value to KitSpace
- Closes #423