hildogjr/KiCost #424: Rapid Electronics

alicephilippa, 2020-10-23:
Is it possible for Rapid Electronics (https://www.rapidonline.com/) and/or CPC (https://cpc.farnell.com/)  to be added to the list of suppliers? 

KiCost 1.1.4
Windows 10
KiCad 5.1.7-1

hildogjr, 2020-10-23:
Yes, sure.

Two approaches can be taken:

1. The default configuration of KiCost  rely on `kicost/distributors/api_partinfo_kitspace.py` web service. If this service already aggregate the two distributors information you just need to add them to lines 89 (list of distributors) and 106 (dict of translation: user name - server name);
2. Develop the own `api_*.py` file for each distributor and put it on `distributors` folder.

set-soft, 2021-05-24:
They aren't listed as supported by Octopart:

https://octopart.com/api/v4/values#sellers
https://octopart.com/distributors

hildogjr, 2021-05-24:
@alicephilippa, do these websites have it own API system for requests? I didn't find this information.