hildogjr/KiCost #425: usage of --currency switch on prices and excel sheet

sebo83910, 2020-10-26:
Hello,

I use kicost 1.1.4 with the switch currency set to EUR.
I have two remarks.
- for a specific part, if i look at the distributor web site, the price given on the xls is slightly different. Is that a currency conversion is made within kicost itself?
- for each distributor part number, there is a comment made showing the Qty/price break. The price are still in USD and not in EUR. More over, is it possible to enlarge the comment area? most of the time, there is not enough space to show the entire information.

Thanks for your excellent work!!

hildogjr, 2020-10-26:
Yes, the conversion was introduced by me, it prioritize the values passed by the distributors and made conversion by `CurrencyConverter` python module, when not present. Usually, by the API, the distributors just provide USD or USD+EUR (but not always).

You can update `CurrencyConverter`, that have a static conversion database. Also, the spreadsheet have the currency rate listed bellow all your parts (edit them to update the values).

Because of this behavior: the currency rate may not be up-to-date and to use a online server to do that was necessary an API and user login and, the comments-balloons are static (doesn't change with the update of the cell with the current rate). I kept them as original distributor currency, it is not the best but it was what I though at the moment.

sebo83910, 2021-01-18:
Hi @hildogjr,
Is it possible to retrieve the data in EUR instead of USD? I mean without converting the price within the python? If look at mouser for instance, the prices migth be significantly different when I look at the website... The available quantity as well...

hildogjr, 2021-01-18:
Yes, just not use `--currency` on call.

Some distributor have native USD and EUR and KiCost prioritize USD, but I didn't remember if EUR get deleted in this case on `api_partinfo_kitspace.py` functions.

The difference in qty is because we use KitSpace API, that update it database time-by-time on demand. The ffinal solution to this is have suport for all distributors APIs, but it demands some code https://github.com/xesscorp/KiCost/issues/386.

sebo83910, 2021-01-18:
Now I'm confused. I removed the switch --currency and I have the same results. I thought everything would be in USD in this case...

sebo83910, 2021-01-18:
No sorry. I talked too fast sorry. USD by default without the switch.
I've been looking at the code but I can't find a way to retrieve the data in EUR form Mouser (as an example).
Moreother, I see a py file for octopart. Does the tool support octopart queries instead of kitspace?

hildogjr, 2021-01-18:
I think, specific Mouser (I am not sure) just have EUR in the KitSpace API, so USD is internally generated.

sebo83910, 2021-01-19:
For Mouser, I don't know. it seems to me everything out of the call to KitSpace API is in USD or GBP.

I tried to switch to Octopart as you suggested:
api_octopart.query_part_info(parts, distributor_dict, currency).
It crashes so far.

hildogjr, 2021-01-19:
You need you on KEY / license / registration to run Octopart.

GBP is the base currency for RS, if I remember...

sebo83910, 2021-01-19:
yes indeed, there is a key required along with a registration.
So, what's the current status for kicost? do you have a roadmap for future improvement of the tool?

hildogjr, 2021-01-19:
I have a personal roadmap
https://github.com/xesscorp/KiCost/milestones
but not time to develop all by my self.

hildogjr, 2021-05-05:
There was some changes @set-soft  of how the prices currencies are converted. @sebo83910, could you check the last code on this repository?

set-soft, 2021-05-24:
1. Now we update the currency conversion. So differences should be (now) some distributor detail.
2. Now Octopart can be enabled from command line and/or environment. Seems to be working.
3. I verified that using `--currency EUR` we are selecting the values in EUR (when available of course)

For the above reasons, and the lack of reply, I'm closing this issue

closed 2021-05-24 by set-soft