hildogjr/KiCost #427: how to fix kicost or how to uninstall then new install it?

enhee2706, 2020-12-02:
Sorry all. How to fix it? Please help me.
enhee@M92p-1:~/regulator/pcb_dzd1$ kicost --version
KiCost v.1.1.4
enhee@M92p-1:~/regulator/pcb_dzd1$ kicost -i dzd.xml
Progress:   0%|                                        | 0/29 [00:00<?, ?part/s]Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 159, in _new_conn
    conn = connection.create_connection(
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 84, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 74, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 665, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 376, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 996, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 314, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 171, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x7ff619d77a00>: Failed to establish a new connection: [Errno 110] Connection timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 719, in urlopen
    retries = retries.increment(
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 436, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='dev-partinfo.kitspace.org', port=443): Max retries exceeded with url: /graphql (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7ff619d77a00>: Failed to establish a new connection: [Errno 110] Connection timed out'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/enhee/.local/bin/kicost", line 8, in <module>
    sys.exit(main())
  File "/home/enhee/.local/lib/python3.8/site-packages/kicost/__main__.py", line 306, in main
    kicost(in_file=args.input, eda_name=args.eda,
  File "/home/enhee/.local/lib/python3.8/site-packages/kicost/kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "/home/enhee/.local/lib/python3.8/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 353, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "/home/enhee/.local/lib/python3.8/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 197, in get_part_info
    results = api_partinfo_kitspace.query(query, distributors)
  File "/home/enhee/.local/lib/python3.8/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 124, in query
    response = requests.post(QUERY_URL, {'query': query_type, "variables": variables})
  File "/usr/lib/python3/dist-packages/requests/api.py", line 116, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='dev-partinfo.kitspace.org', port=443): Max retries exceeded with url: /graphql (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7ff619d77a00>: Failed to establish a new connection: [Errno 110] Connection timed out'))
Progress:   0%|                                        | 0/29 [02:09<?, ?part/s]

hildogjr, 2020-12-02:
Try:

pip3 uninstall kicost # To remove it.
pip3 install kicost # To reinstall it.

If you want it present to all user, use
sudo -H pip3 install kicost

closed 2020-12-03 by enhee2706

enhee2706, 2020-12-03:
I did it. But still showing this errors.

hildogjr, 2020-12-03:
```
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='dev-partinfo.kitspace.org', port=443): Max retries exceeded with url: /graphql (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7ff619d77a00>: Failed to establish a new connection: [Errno 110] Connection timed out'))
```

Means that the KitSpace server is offline. At the moment KiCost just rely on KitSpace into but is possible to change it to direct use Octopart (./kicost/distributos/api_octopart.py) or develop other API/scrape modules.

enhee2706, 2020-12-03:
I received this. I will change HTTPSConnectionPool(host='dev-partinfo.kitspace.org', port=443) to Octopart (./kicost/distributos/api_octopart.py). But where is this function HTTPSConnectionPool(host='dev-partinfo.kitspace.org', port=443).

hildogjr, 2020-12-03:
This is just a internal error of the library.

The function responsible to do the work is `query_part_info(foo)` on both files. To use Octopart you will need to have your on registered KEY (line 92 of `api_octopart.py`).