hildogjr/KiCost #431: feature request: data export to json

sebo83910, 2021-01-09:
Hello,

I'm currently struggling to understand the data (parts) structure within the projet. I'm interested by adding an export feature to the tool as I'd like to use output of the tool with another custom tool.
So far, I guess I'm not good enough at Python to do this. Because, I'm trying to build a "complex" dictionary but I encounter a structure I'm not familiar with.

What I'd like to have is a list/dictionary like this:

Part1:
 - dist1
     - url
     - qty
     - currency
     - price_tiers
     - datasheet  
 - dist2
     - ...
Part2:
 - dist1
 - ...

so far, no luck.
What do you think?

thanks.

hildogjr, 2021-01-11:
Hi @sebo83910.
Nice to ear some other used of enlargement of the output format.

# General aspects

You could use the same software flux control of `collapse_refs` and `suppress_cat_url` to control the choice (spreadsheet or your new format), check how they affect `__main__.py` and `kicost.py`.

- Create some `--new_format` at the `parser.add_argument(foo)` directives of `__main__.py`.
- Modify `kicost.py` line 254 to use a `create_new_format` instead `create_spreadsheet` when passed `kicost --new_format`.
- Create some `new_format.py` file with the function `create_new_format(foo)` that receive the same parameters as `create_spreadsheet(foo)`

# Variables

The `parts` variable (passed to `create_spreadsheet(foo)`) is a list of elements, each one is a dictionary with two parts:

- 1st: a global part of the spreadsheet/BOM, the fields: refs, values, desc, footprint, manf (optional), manf# . See the `columns` variable of line 328 of `spreadsheet.py`. Each `columns.keys()` defines that. Note: 'unit_price' and 'ext_price' will be just formulas of Excel;
- 2nd: distributors and prices. For each part (row of the spreadsheet / element of the list of variable `parts`) you have `part_num` (stock number), `price_tiers` (a dict with pairs:  "tier qty - price tier") and `dist_currency`. I do not remember exactly the structure now but it can be checked/debugged at line 865 of `spreadsheet.py`.

sebo83910, 2021-01-11:
Thanks @hildogjr for your inputs.
I've been able to make some progress. But it's more a hack than a nice solution so far. I'll see how I could adapt my code based on the elements you've desribed in your comment.

hildogjr, 2021-01-11:
Which tool you aim to integrate? Whats is the output format need?

(If is of your interest) keep the code as branch to we keep on track here.

sebo83910, 2021-01-11:
@hildogjr I'm not very familier either with github. I'd like to make sure I understand, you'd allow me to use a branch on your project where I could commit this feature? If you're ok, i'd be more than happy to share this with you. Let me know.

hildogjr, 2021-01-11:
At `https://github.com/xesscorp/KiCost` you **fork** the project and edit it was you want, you will be a owner of `https://github.com/sebo83910/KiCost`.

When finished the changes you can **merge it** (pull request) to be merged to this main repository.

sebo83910, 2021-01-11:
Ok thanks. I'll do it when I finish some tests.

closed 2021-01-19 by hildogjr