hildogjr/KiCost #432: KiCost is not finding some parts

cbrake, 2021-01-26:
#### Issue / Problem report
1. Add the KiCost version. Type ``kicost --version`` at terminal.

KiCost v.1.1.4

3. Write the command used to call KiCost (or the graphical interface configuration);

```
[cbrake@mars is-interface]$ kicost -i is-interface.xml 
No information found for parts 'D1-D3,D14-D17' query `{'mpn': {'manufacturer': 'Diodes Inc', 'part': 'US1M-13-F'}}`                                                                                                                                            
No information found for parts 'PS1' query `{'mpn': {'manufacturer': 'Mean Well', 'part': 'IRM-30-12'}}`                                                                                                                                                       
No information found for parts 'C4' query `{'mpn': {'manufacturer': 'TDK Corporation', 'part': 'B32672P4155K000'}}`                                                                                                                                            
No information found for parts 'R7-R12' query `{'mpn': {'manufacturer': 'Stackpole Electronics', 'part': 'JW60ZT0R00'}}`                                                                                                                                       
No information found for parts 'J2' query `{'mpn': {'manufacturer': 'TE', 'part': '1-744048-0'}}`                                                                                                                                                              
No information found for parts 'F1' query `{'mpn': {'manufacturer': 'Bel Fuse', 'part': '56200001009'}}`                                                                                                                                                       
No information found for parts 'D4-D6,D18-D20' query `{'mpn': {'manufacturer': 'Diodes Incorporated', 'part': '1N4148WS-7-F'}}`                                                                                                                                
No information found for parts 'Q1-Q3' query `{'mpn': {'manufacturer': 'Diodes Inc.', 'part': 'FMMT734TA'}}`                                                                                                                                                   
No information found for parts 'R5,R6,R20' query `{'mpn': {'manufacturer': 'Vishay Dale', 'part': 'CRCW060347K0FKEAC'}}`                                                                                                                                       
No information found for parts 'R22-R24' query `{'mpn': {'manufacturer': 'Vishay Dale', 'part': 'CRCW060313K0FKEAC'}}`    
```

4. One BoM to reproduce the error (with the EDA version).

[is-interface.zip](https://github.com/xesscorp/KiCost/files/5875621/is-interface.zip)

#### Feature / Enhancement request

I'm puzzled why parts like B32672P4155K000 are not found at digikey/mouser, where a search at those sites finds the parts:

https://www.mouser.com/ProductDetail/EPCOS-TDK/B32672P4155K000/?qs=%2Fha2pyFadujue6vbtQaeYPj6jk5kCLgId7Mk00UFarJVK10yhKieZQ%3D%3D

Appreciate any thoughts as to what I'm doing wrong.

cbrake, 2021-01-27:
Here is what the spreadsheet looks like:

[image]

hildogjr, 2021-01-27:
Possible remove de manufacture field it use `kicost --ignore_field manf`.

It is given, KiCost try to match both, manf and manf\#, and sometimes the first have typing errors or just different spelling on server answer.

cbrake, 2021-01-28:
That helped a lot.

[image]

It is still not finding 1-744048-0, which is available on mouser and digikey:

https://www.mouser.com/ProductDetail/TE-Connectivity-AMP/1-1744048-0/?qs=%2Fha2pyFadujJMp3AMHUiHuiMT2R%2FLcQVUHEiIbFC1iKB0%252BmAgyihBw%3D%3D

https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/1-1744048-0/4730081?s=N4IgTCBcDaIIwFoDsAWFAGFAOB6QF0BfIA

Any thoughts what might be going on there?

hildogjr, 2021-01-28:
"1-1744048-0" could be a too generic part name and not recognized into the API system.

For those disambiguation case add the specific stock number on each part using "digikey#"  and "mouser#" fields (and keep the "manf#"/"MPN" field) on KiCad (or other EDA that are you using).

cbrake, 2021-01-28:
Got it -- Thanks for your help!

closed 2021-01-28 by cbrake