hildogjr/KiCost #457: kicost 504 error when connecting to https://dev-partinfo.kitspace.org

tinoest, 2021-03-24:
I am not sure if this is an issue with my end or kitspace, it used to work when I last tried it but that was over a year ago now. 

I am getting a 504 error when attempting to get the prices for the attached csv BoM. If anyone could help it would be much appreciated please see the debug output below. 

kicost 1.1.4

```
/usr/local/bin/python3 -m kicost -i test.csv --eda csv --overwrite --include farnell --debug 15
Started KiCost v.1.1.4 on macOS-10.14.6-x86_64-i386-64bit(64bit) Python 3.9.2
# Getting from CSV 'test.csv' BoM...
Getting CSV header...
Getting parts...
# Removing do not populate parts...
Splitting subparts in the manufacture / distributors codes...
C1 >> ['manf#']
part/qty>> MKS2D031001A00KI00		part>>MKS2D031001A00KI00	qty>>
{'manf#': 'MKS2D031001A00KI00', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C6 >> ['manf#']
part/qty>> MKS2D031001A00KI00		part>>MKS2D031001A00KI00	qty>>
{'manf#': 'MKS2D031001A00KI00', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C11 >> ['manf#']
part/qty>> MC0805F105Z160CT		part>>MC0805F105Z160CT	qty>>
{'manf#': 'MC0805F105Z160CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C12 >> ['manf#']
part/qty>> MC0805F105Z160CT		part>>MC0805F105Z160CT	qty>>
{'manf#': 'MC0805F105Z160CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C14 >> ['manf#']
part/qty>> MC0805F105Z160CT		part>>MC0805F105Z160CT	qty>>
{'manf#': 'MC0805F105Z160CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C15 >> ['manf#']
part/qty>> MC0805F105Z160CT		part>>MC0805F105Z160CT	qty>>
{'manf#': 'MC0805F105Z160CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C16 >> ['manf#']
part/qty>> MC0805F105Z160CT		part>>MC0805F105Z160CT	qty>>
{'manf#': 'MC0805F105Z160CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C19 >> ['manf#']
part/qty>> MC0805F105Z160CT		part>>MC0805F105Z160CT	qty>>
{'manf#': 'MC0805F105Z160CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C13 >> ['manf#']
part/qty>> EEEFC1V100R		part>>EEEFC1V100R	qty>>
{'manf#': 'EEEFC1V100R', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C17 >> ['manf#']
part/qty>> MC0805N220J500CT		part>>MC0805N220J500CT	qty>>
{'manf#': 'MC0805N220J500CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C18 >> ['manf#']
part/qty>> MC0805N220J500CT		part>>MC0805N220J500CT	qty>>
{'manf#': 'MC0805N220J500CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C2 >> ['manf#']
part/qty>> 865080345012		part>>865080345012	qty>>
{'manf#': '865080345012', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C4 >> ['manf#']
part/qty>> 865080345012		part>>865080345012	qty>>
{'manf#': '865080345012', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C3 >> ['manf#']
part/qty>> MKS0F016800B00MSSD		part>>MKS0F016800B00MSSD	qty>>
{'manf#': 'MKS0F016800B00MSSD', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C5 >> ['manf#']
part/qty>> MKS0F016800B00MSSD		part>>MKS0F016800B00MSSD	qty>>
{'manf#': 'MKS0F016800B00MSSD', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C7 >> ['manf#']
part/qty>> EEEFC0J101P		part>>EEEFC0J101P	qty>>
{'manf#': 'EEEFC0J101P', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C8 >> ['manf#']
part/qty>> EEEFC0J101P		part>>EEEFC0J101P	qty>>
{'manf#': 'EEEFC0J101P', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C9 >> ['manf#']
part/qty>> MC1210B223K501CT		part>>MC1210B223K501CT	qty>>
{'manf#': 'MC1210B223K501CT', 'libpart': 'Lib:???', 'footprint': 'Foot:???', 'value': '???'}
C10 >> ['manf#']
part/qty>> MC1210B223K501CT		part>>MC1210B223K501CT	qty>>
[... 109 lines cut ...]
                              'delimiter': ' '},
                    'type': 'local'}}
# Getting part data from KitSpace...
Starting new HTTPS connection (1): dev-partinfo.kitspace.org:443                                                                                                                                                                                                                                                                                                                                                                      
https://dev-partinfo.kitspace.org:443 "POST /graphql HTTP/1.1" 504 192                                                                                                                                                                                                                                                                                                                                                                
Progress:   0%|                                                                                                                                                                                                                                                                                                                                                                                              | 0/26 [01:00<?, ?part/s]Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kicost/__main__.py", line 320, in <module>
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kicost/__main__.py", line 306, in main
    kicost(in_file=args.input, eda_name=args.eda,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kicost/kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 353, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 197, in get_part_info
    results = api_partinfo_kitspace.query(query, distributors)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 135, in query
    raise Exception('One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.')
Exception: One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.
Progress:   0%|          
```

mdeweerd, 2021-03-24:
After updating to the current master version, I have the same on my end - the process I get 
```log
# Getting part data from KitSpace...
Starting new HTTPS connection (1): dev-partinfo.kitspace.org:443
https://dev-partinfo.kitspace.org:443 "POST /graphql HTTP/1.1" 504 192
Progress:   0%|          | 0/12 [01:00<?, ?part/s]Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 33, in <module>
    sys.exit(load_entry_point('kicost==1.1.5', 'console_scripts', 'kicost')())
  File "/usr/local/lib/python3.8/site-packages/kicost/__main__.py", line 318, in main
    kicost(in_file=args.input, eda_name=args.eda,
  File "/usr/local/lib/python3.8/site-packages/kicost/kicost.py", line 251, in kicost
    api_partinfo_kitspace.query_part_info(parts, distributor_dict, currency)
  File "/usr/local/lib/python3.8/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 369, in query_part_info
    get_part_info(queries[slc], query_parts[slc], query_part_stock_code[slc])
  File "/usr/local/lib/python3.8/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 214, in get_part_info
    results = api_partinfo_kitspace.query(query, distributors)
  File "/usr/local/lib/python3.8/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 151, in query
    raise Exception('One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.')
Exception: One of the internal Kitspace services may experiencing problems. Contact the Kitspace support.
Progress:   0%|          | 0/12 [01:00<?, ?part/s]
```

I am able to get BOMs without prices using `--no_price` as an option.

snhobbs, 2021-04-06:
I have the same issue and get no response while pinging dev-partinfo.kitspace.org. Kitspace says they're in a "closed beta phase" for their Bom-Builder. Did they take the API offline?

mdeweerd, 2021-04-06:
partinfo.kitspace.org is pingable - maybe we need to remove the "dev-" part!

mdeweerd, 2021-04-06:
Well, that worked!

hildogjr, 2021-04-06:
Just following the discussion. Please, test the last code version on Git.

closed 2021-04-06 by hildogjr

mdeweerd, 2021-04-06:
@hidogjr Updated several times to the last code version before, but this time it is fine!

hildogjr, 2021-04-06:
There was several updates from @set-soft to pass on all automated tests and better internal structures.

I need some revision and tests before release at PyPI.

commit bb5136fb (2021-04-06): Fix #457