hildogjr/KiCost #466: "Local" distributor not working

set-soft, 2021-04-29:
This is related to the following part of the documentaion: "Custom Part Data":

-----------------------
The price breaks on some parts can't be obtained automatically because:

* they're not offered by one of the distributors whose web pages KiCost can scrape/query, or
* they're custom parts.

For these parts, you can manually enter price information as follows:

#. Create a new field for the part named ``kicost:pricing`` in either the schematic or library.
#. For the field value, enter a semicolon-separated list of quantities and prices which
   are separated by colons like so::

      1:$1.50; 10:$1.00; 25:$0.90; 100:$0.75
      
   (You can put spaces and currency symbols in the field value. KiCost will
   strip everything except digits, decimal points, semicolons, and colons.
   Others currency are acepted by use of the standardize ISO 4217 alpha3 format,
   e.g. ``USD1.50``, ``EUR1.00``)
   
You can also enter a link to documentation for the part using a field named ``kicost:link``.
The value of this field will be a web address like::

    www.reallyweirdparts.com/products/weird_product.html
   
After KiCost is run, the price information and clickable link to documentation
for the part are shown in a section of the spreadsheet labeled **Local**.
If you want to associate the pricing and/or documentation link to a particular
source or distributor, just place an extra label within the field key to indicate
the source like so:

    kicost:My_Weird_Parts:pricing
    kicost:My_Weird_Parts:link

Then the pricing and documentation link for that part will appear in a section of the spreadsheet labeled `My_Weird_Parts`.

------

Current code in `master` fails to do it. I think the original implementation was by @devbisme and then @mdeweerd tried to fix something related to variants and all started to fail.

I changed the code in this patch: 343297c0a3aa9a58b7ac914ef9a80ad9d5eb1519

I'm looking for comments about it. After applying this patch I found various interesting differences in the tests. You can see a big difference in `fitting_test.xlsx`. 

The current result can be observed here: https://github.com/xesscorp/KiCost/suites/2616517652/artifacts/57324546
And the current `master` is here: https://github.com/xesscorp/KiCost/suites/2579090530/artifacts/56243325

You'll see that in `master` JP1 is separated from JP2 and JP3. This is because the code fails to interpret `kicost:pricing`. You can also note that we don't get the `Local` distributor as stated in the documentation.

Is the new version correct? I guess that the `local` namespace is there to avoid collision with other tools. As an example: if another tool makes a different use of the `DNP` field you can use `kicost:DNP` to avoid problems. Is this correct?

mdeweerd, 2021-05-02:
I added a test case with regards to multipart and "local" distributor in https://github.com/mdeweerd/KiCost/tree/complex_multipart .
I do not know the best method to make it work, as it also gets the pricing from the kitspace API.

This also demonstrates that it is not currently possible to have a multipart that has a pricing and different distributors.
In this example it's a fuse: the fuse holder and casing are provided by one manufacturer that proposes a direct pricing not available through "scraping".  The fuse itself is available elsewhere.

The multipart separator is ";", which is the same as the pricing separator.  
If the pricing separator would have been "#" for instance, then the multipart could have been 'kicost:distributor:pricing' => '1:EUR0.07;0:0;1:EUR0.03'  as given in the example where the pricing is 0.07 for the first part, 0 for the second and 0.03.  
Instead  This is interpreted as the pricing for all the parts in the multiparts.  

Either this is case it treated by depending on the price break quantity: as long as it strictly increases from >1, it belongs to the same part, when it remains the same or is less than the previous quantity break it is considered for the second part, and considering a quantity of 0 as "unavailable".  The above exemple would then work as well.  The quantities are 1,0 and 1. 
A value of '1:EUR0.10;10:EUR0.09;1000:EUR0.07;0:0;1:EUR0.03' would also work as 1,10 and 100 are increasing, 0 marks a new start as well as the end of a pricing and 1 marks a new start.

set-soft, 2021-05-04:
> I added a test case with regards to multipart and "local" distributor in https://github.com/mdeweerd/KiCost/tree/complex_multipart .
> I do not know the best method to make it work, as it also gets the pricing from the kitspace API.

To get the price from KitSpace:
1) In `test_kicost.py` define `ADD_QUERY_TO_KNOWN` to `True`
2) Run the test
3) Check you got the queries recorded in `tests/kitspace_queries.txt`
4) Revert `ADD_QUERY_TO_KNOWN` to `False`
5) Now you'll get the recorded replies

> This also demonstrates that it is not currently possible to have a multipart that has a pricing and different distributors.
> In this example it's a fuse: the fuse holder and casing are provided by [...]

IMHO multiparts are a nasty hack. I understand they can save time, but if you can't solve the problem using it ... do it in the right way ;-)
With right way I mean: put explicit parts in your schematic, then mark them to be outside the PCB (i.e. virtual components).

We could also make some separators configurable.

mdeweerd, 2021-05-04:
Thanks for the procedure to set up a test with parts.

This could be added to the readme.md I started in the tests directory.

> IMHO multiparts are a nasty hack. I understand they can save time, but if you can't solve the problem using it ... do it in the right way ;-)
> With right way I mean: put explicit parts in your schematic, then mark them to be outside the PCB (i.e. virtual components).
Virtual, empty components on the PBC seem like a nasty hack to me.  
It's also nice to keep the components together, also useful when they are connectors, mating connectors and cables.  
Multipart components are working pretty well overall but some functionnalities have to be tweaked.

Sidenote: Something that would also be usefull is the examination of alternatives (different options for the manf/manf#), but that is more complex to solve.

> We could also make some separators configurable.
What do you think about the proposal to avoid a specific separator for the pricing?
Making separators configurable is an idea, but then I'ld prefer to be able to configure that from within the sheet - for instance one (or more) of the component fields could define the separators for that part .

mdeweerd, 2021-05-04:
I checked the updated code.

The comment column is now missing with the following options:
```
--ignore_fields h comment --no_price --no_collapse -f comment S1MN S1PN S
2MN S2PN
```

The intent is to not use the comment column to conclude that components are distinct, but still have it in the resulting BOM.

set-soft, 2021-05-04:
> What do you think about the proposal to avoid a specific separator for the pricing?

In general the mechanism used for multiparts looks uterly complex to me. Sure, we can add some "intelligence". Or we could just introduce some rules, like:

```
The multipart separator is `;` or `|`. When providing a pricing list please use `|` to avoid confusing it with the price separator (also `;`).
```

I say `|` just because this character is quite uncommon and also works very well as a visual separator. But it could be another. 

This rule could be disabled using some command line option, just in case it collides with some old use.

set-soft, 2021-05-04:
> ```
> --ignore_fields h comment --no_price --no_collapse -f comment S1MN S1PN S
> 2MN S2PN
> ```
> 
> The intent is to not use the comment column to conclude that components are distinct, but still have it in the resulting BOM.

The help says:

```
--ignore_fields NAME [NAME ...]
                        Declare part fields to ignore when reading the BoM file.
```

So if `comment` is ignored when you read the BoM you'll get an empty column.

set-soft, 2021-05-04:
I think you want to use `--group_fields` I'm committing a change with it.

set-soft, 2021-05-04:
If the only remaining issue here is the multipart separator thing I think we should close it and open another for the separator.

mdeweerd, 2021-05-04:
We also need to read what is written elsewhere:
```
It is possible that there are identical parts in your schematic that have differing data and, hence, wouldn’t be grouped together. For example, you might store information about a part in a “notes” field, but that shouldn’t exclude the part from a group that has none or different notes. *There are three ways to prevent this:*

Use the --ignore_fields command-line option to make KiCost ignore part fields with certain names:
```

The '--ignore_fields' option is there so that KiCost does not use the field in the grouping algorithm, but it does not mean that it should skip the field.

Interpreting that option there would be no method to have a specific comment for a component different from the others of the same type.
In the end, not ignoring Comment (or other fields that should not prevent from grouping) generates multiple lines of the same component in the BOM, so incorrect pricing and difficulties when submitting this BOM to the EMS.

This option worked for me before, and does not work anymore.

mdeweerd, 2021-05-04:
It seems that the grouping option works better than in the past.  Still, IMHO 'ignore' should not imply 'empty'.

I added a test case covering some cases.

set-soft, 2021-05-04:
> We also need to read what is written elsewhere:
>
> ```
> It is possible that there are identical parts in your schematic that have differing data and, hence, wouldn’t be grouped together. For example, [...]

And this is right, you can do it. But then you can't use this field because you ignored it. This is why it you have option 2.

> The '--ignore_fields' option is there so that KiCost does not use the field in the grouping algorithm, but it does not mean that it should skip the field.

I don't agree, the help says:

```
  --ignore_fields NAME [NAME ...]
                        Declare part fields to ignore when reading the BoM file.
  --group_fields NAME [NAME ...]
                        Declare part fields to merge when grouping parts.
```

If you want to **ignore** the field **when reading the BoM** use `--ignore_fields`. If you want to avoid a group split use `--group_fields`.

> Interpreting that option there would be no method to have a specific comment for a component different from the others of the same type.
> In the end, not ignoring Comment (or other fields that should [...]

This is why you have another option.

> This option worked for me before, and does not work anymore.

I don't know what worked in the past or what the docs said. But the current code and docs can be used without problems and are quite clear.

mdeweerd, 2021-05-04:
I'll modify my BOM generation scripts and report back in case it does not fullfill the needs ;-).

My experience goes as far back as #102 and #195 .  I had no idea that was already 3 years !

mdeweerd, 2021-05-04:
> I added a test case with regards to multipart and "local" distributor in https://github.com/mdeweerd/KiCost/tree/complex_multipart .

I now updated the branch with scraped information.

Currently this generates the same pricing for the three parts in the multipart component.
This is not what is expected, but at least its a test case that can be used (and improved ;-) ).

There is also a small update to test_kicost.pl so that storing kitspace information by uncommenting a line (faster than copying and editing).

I will make a pull request.

set-soft, 2021-05-05:
About the multipart pricing: I took a look at the code and there are more than one problem with this, isn't just the `;`separator. I found two more problems:

1. The `pricing` field isn't split. Only `DISTRIBUTOR#` and `manf#` are split
2. The `:` separator collides with the quantiry multiplier separator.

mdeweerd, 2021-05-05:
@set-soft Thanks for looking.

I proposed to split the pricing field without a separator.  
The proposal was to apply a split when the pricing quantity is less or equal to the previous break quantity or when the pricing is 0.  

That seems the easiest way.  The question is what to do about cases where the quantity for the next component is higher than the first.  Simple: repeat the same price for a quantity of 1.
A quantity of 0 means no pricing and ends the pricing for the current component.

closed 2021-05-05 by set-soft

set-soft, 2021-05-05:
Hi @mdeweerd !

I implemented it in another way. The patch is more general. In previous code you could use `;` to separate subparts, but you could also use `,`. Taking advantage of this I implemented this:

- You can specify a list of fields to be included in the split mechanism using `--split_extra_fields`
- The `pricing` field is always added to this list
- Subparts separator is restricted to `,`, you can't use `;`
- No multiplier rules are applied to these fields

With this implementation I was able to generate something that IMHO is more similar to what you had in mind when you created the `complex_multipart` test. I just changed `;` by `,`. Note that now we also get splitted 'S1MN', 'S1PN', 'S2MN' and 'S2PN', something impossible before this patch. Also note the patch has a minor impact in the code, just a new list with a very simple processing (compared with all the nightmare around the subparts code).

If this doesn't generate a correct output please open another issue.

mdeweerd, 2021-05-05:
@set-soft Just tried the master - the Excel file looks good !

mdeweerd, 2021-05-05:
@set-soft With regards to the description: is that also ',' seperated or ';' ?  Descriptions tend to have ',' to list properties of the component, so it seems best to maintain ';' there.  
Are descriptions part of the automatic list?

I think it is best to amend the test case to also show 'desc' splitting.

mdeweerd, 2021-05-05:
@set-soft Good work, I tested on a real project and I could amend it to split some parts/

Description splitting with "," works, ";" does not.  I guess that escaping the ',' would work, but the ';' separator would be better for the description (and comment) fields.

Anyway, it's already great that this works, the BOMS just got better ;-).

set-soft, 2021-05-05:
@mdeweerd we can't use `;` as separator because this is the only supported for price lists.

mdeweerd, 2021-05-06:
@set-soft ';' is available for manf# and maybe some other fields (I did not check the code).  I am not suggesting to use it as a separator for 'pricing', but as a separator for "desc", "comment".  Desc could contain texts like "10uF, 10V, X7R" and comment would easily have commas as well.  Escaping those commas is not compatible with the standard use of these fields.

set-soft, 2021-05-06:
This will unnecesary complicate the code. What's the problem with escaping `,`.

Currently `pricing` is just added to the list provided by `--split_extra_fields`.

Also note that this is processed only by user request. So I don't see a dangerous situation, is not that KiCost will try to split multiple subarts accidentally. BTW: "10uF, 10V, X7R" looks a really bad comment. These values belong to 3 different fields. 

I'm not sure about complicating the code because the user:

1. Wants to split a comment with commas
2. Can't replace them by `;` or just escape them

A comment for multiple components will look complex anyways, and will need some thinking from the user.

I don't like the current situation: the same separator is used for more than one thing, and this is for more than one separator. But this is the way it worked for years, and nobody even reported this as a problem. Of course we can add more and more cases and exceptions, but I would like to keep them as low as possible.

mdeweerd, 2021-05-06:
- Escaping "," in those fields that are otherwise used (in KiCAD) without escaping is necessarily going to induce errors.
- I would go for using the ';' as a separator for all fields, but the pricing field has been built as an exception to that.
- Complaints?  I was working on the assumption the ';' was the main separator, not the ',' which was an alternative separator.  https://github.com/xesscorp/KiCost/issues/34#issuecomment-326855921 is in favor of using a not so common character as the separator.  I agree with that and I just think that ';' is less common than ',' in descriptions and comments.  We have to go for one of those because none of the other special characters has been used and it is good to be somewhat consistent.  Very recent evolutions are aligning on ',' as a separator, it is still time to align on ';'.  That's the only separator I was using.  Once a choice is made, it is difficult to revert because of backward compatibility.  1.1.5 is not officially released yet, the improved multipart functionnality is still fresh.
- The only exception should be price, the rest should be ';' (and ',' optionally accepted for fields where it currently already exists).  Price could be changed though, two regexes can easily help to detect which implementation is used : ',' between prices or ';' - I would move to use ',' between prices in stead of ';', and use ';' as the consistent multipart separator.

set-soft, 2021-05-06:
I implemented a very simple patch that implements what you say.

I don't like having 3 different rules, depending on which field the rule is applied.

Test if it work as spected and  submit patches for the docs explaining it.

set-soft, 2021-05-06:
I changed my mind, previous patch wasn't clean. I think this new patch solves the real problem:

```
Now the separator is | 
```

For compatibility we also allow `;` and `,` for `manf`, `manf#` and `DISTRIBUTOR#`. All the test pass, I only modified the `complex_multipart.xml` file.

What do you think @hildogjr ?

hildogjr, 2021-05-06:
`|` could be a good general separator. `,` is present in few `manf#` and created a lot of misunderstanding of the users. Has been some time that I wanted to remove it, but I would like this discussion.

Just thinking: `|` could be the multipart separator since is a rare character and `,` or `;` could stay for `localDistributor` price tier separator. This could open the possibility of use local multipart.

set-soft, 2021-05-06:
Take a look at the last patch @hildogjr it defines:

```python
PART_SEPRTR = r'(?<!\\)\s*\|\s*'             # Separator for the part numbers in a list, remove the lateral spaces.
PART_SEPRTR_LEGACY = r'(?<!\\)\s*[;,\|]\s*'  # Legacy version
```

The `PART_SEPRTR_LEGACY` is used to split `manf`, `manf#` and `DISTRIBUTOR#`. In this way the user can use the old `;` and `,` separators, and the new one `|`.

But for `pricing`, and any user field specified by the user, the only available separator is `|`. And yes, it makes the local pricing work for the `complex_multipart.xml` test. In addition to comments fields.

hildogjr, 2021-05-07:
```xml
<field name="kicost:we:pricing">1:EUR0.07|0:0|1:EUR0.03;100:EUR0.01</field>
<field name="manf">Wurth Electronik|Bel Fuse|Wurth Electronik</field>
<field name="manf#">696101000002|5MF 6-R|695120004001</field>
<field name="mouser#">710-696101000002|530-5MF6-R|</field>
```

I liked this new `|` separator, I think it is more clear.

set-soft, 2021-05-07:
> I liked this new `|` separator, I think it is more clear.

Yes this is very good visually. Good for multiparts where the fields becomes really long.

commit b8c519db (2021-05-05): Added a mechanism to include more fields in the subparts split.

- New command line option --split_extra_fields
- The pricing is always included
- Closes #466 well in fact the discussion it derived.