hildogjr/KiCost #467: BOM for Farnell has too many fields?

mdeweerd, 2021-04-30:
The feature to copy paste the BOM from the excel sheet to the distributor is a great help, but did not work "as is" for "https://fr.farnell.com/quick-order?isQuickPaste=true" .

I had to reduce:
```csv
3416650,5,C1, 10u 50V X7S Ceramic Capacitor RADIAL,TDK,,,,,
1650914,10,C2;C6,100n ±20% 50V Z5U Ceramic Capacitors axial,KEMET,,,,,
2483339,2,LCD1,6800 16x02 character LCD https://www.winstar.com.tw/products/character-lcd-display-module/wh1602b2-tmi-et.html,Winstar,,,,,
9845178,5,Q1,Mosfet N-CH 60V 200mA TO92-3,,,,,,
1574372,5,Q3,Transistor PNP 40V 200mA TO92-3,,,,,,
1357143,5,RV1,Potentiometer 10k axial,Vishay,BOURNS,,BI Technologies,T73YP103KT20,3362P-1-103LF
2065107,25,SW1-SW5,Tactile Switch 6x6mm Height 17mm SMD or THT,ALPSALPINE,,,,,
3124747,5,U1,I/O Expander 8 I²C 16-PDIP,TEXAS INSTRUMENT,,,,,
```

to

```csv
3416650,5,C1, 10u 50V X7S Ceramic Capacitor RADIAL,TDK
1650914,10,C2;C6,100n ±20% 50V Z5U Ceramic Capacitors axial
2483339,2,LCD1,6800 16x02 character LCD
9845178,5,Q1,Mosfet N-CH 60V 200mA TO92-3
1574372,5,Q3,Transistor PNP 40V 200mA TO92-3
1357143,5,RV1,Potentiometer 10k axial
2065107,25,SW1-SW5,Tactile Switch 6x6mm Height 17mm SMD or THT
3124747,5,U1,I/O Expander 8 I2C 16-PDIP
```

The latter list passes.

closed 2021-04-30 by hildogjr

hildogjr, 2021-04-30:
The intention of this be configurable is that some distributors print the "column" comment in ticket of only parts package. This could help check.

Tell us if some information can be added to this or any other distributor.

mdeweerd, 2021-04-30:
It turns out that the description in
"2483339,2,LCD1,6800 16x02 character LCD https://www.winstar.com.tw/products/character-lcd-display-module/wh1602b2-tmi-et.html"
is to long .

BTW, the link for farnell points to newark - the "issue" for adding a link in the tool is that one has to choose a local site such as "fr.farnell.fr" or "uk.farnell.fr"  -  the rest of the link is fine.

mdeweerd, 2021-05-01:
Thank you for the quick updates!

Regarding "uk" - they're not in the EU anymore so their pricing is in GBP.  As an in between, I propose 'https://fr.farnell.com/en-FR/quick-order?isQuickPaste=true' - still in English, so will be Euro.  It could be made dependent on the main currency, but well, that's more coding.

It seems that the length of the field/text following the first three fields has to be limited to 41 characters.
In the below 'quick paste', the first line is accepted, the 2nd line breaks the import:
```csv
2483339,5,LCD168006800,6800 16x02 character LCD https://www.winstar.com.tw/products/xxx
2483339,5,LCD168006800,6800 16x02 character LCD https://www.winstar.com.tw/products/xxxx
```

The content of the description is the same as desc - it's not a KiCAD issue, this was edited by my trainee (I'll move this to the comment instead).

Editing the distributors_info.py file, I see that the URL for farnell points to newark - it can be pointed to www.farnell.com .
```
        'label': {
            'name': 'Farnell', 'url': 'https://www.newark.com/',
            'format': {'font_color': 'white', 'bg_color': '#FF6600'},  # Farnell/E14 orange.
        },

```

Thanks

Mario

set-soft, 2021-05-01:
Hi @mdeweerd !
I played with it and the limit seems to be 30, extra stuff is just ignored.
I added a fix to this branch: distributors_cleanup
Can you check it?

mdeweerd, 2021-05-01:
I had also played with it and the limit was 42  (on fr.farnell.com...) and I also increased the reference field a bit which did not have an impact on the limit in my test.

But if 30 is the limit on other version, so be it !

I tested the distributors_cleanup branch and it worked for me (and the main branch did not, so that confirms that I had to switch).

Thank you.

mdeweerd, 2021-05-02:
But, in principle reference & description has to be switched:
[image]

The reference appears in the note and the note as the part number currently.

set-soft, 2021-05-02:
> I had also played with it and the limit was 42 (on fr.farnell.com...) and I also increased the reference field a bit which did not have an impact on the limit in my test.
> 
> But if 30 is the limit on other version, so be it !

I'm trying with the french URL, I got:
`Your line notes have exceeded the maximum number of characters allowed. Please revise your line notes to be under 30 characters.`

In the other field I can enter 40 character, but only the first 30 are retained.

 
> I tested the distributors_cleanup branch and it worked for me (and the main branch did not, so that confirms that I had to switch).

Thanks for testing it.

set-soft, 2021-05-02:
> But, in principle reference & description has to be switched:
> [image]
> 
> The reference appears in the note and the note as the part number currently.

Thanks, I inverted it.

BTW @mdeweerd please take a look at #466 I found some issues regarding the "Local" distributor mechanism. I traced the problem and it seems to be related to this patch: b50368313ba9b08cc6e6ca0149983c36908e17f0

I'm reverting part of this patch, and adding other mechanisms to ensure KiCost behaves as stated in the docs. Do you have any test case that I can use to be sure that I don't break the whole b50368313ba9b08cc6e6ca0149983c36908e17f0 ?

mdeweerd, 2021-05-02:
@set-soft  Ok, I tied to create a test case,   While building it I ran in another issue.  See https://github.com/mdeweerd/KiCost/tree/test_variants_2 , "test_variants_2.sh" shows the way I ran the test case which results in an exception.

Maybe we should not be continuing to comment in a "closed" case.

set-soft, 2021-05-02:
> Maybe we should not be continuing to comment in a "closed" case.

I opened #474 
BTW the problem is related to `manf` missmatch. The exception is just the way KiCost likes to report errors (IMHO quite confusing).

commit e0d915c4 (2021-04-30): Fix #467 - Farnel and Newark purchase columns
commit d95e9cd0 (2021-04-30): Fix #467 (link)
commit 9451d9db (2021-05-01): Fix #467 - Farnel and Newark purchase columns
commit ac26121f (2021-05-01): Fix #467 (link)