hildogjr/KiCost #479: 1.1.4 KiCost stalling and not loading component data.

BillyPocketNC, 2021-05-04:
I hope this is helpful for other KiCost users when KiCost stalls and will not get prices on components.
I edited the api_partinfo_kitspace.py line 69 to get KiCost working.

Before: QUERY_URL = 'https://dev-partinfo.kitspace.org/graphql'
After: QUERY_URL = 'https://partinfo.kitspace.org/graphql'

Found fix here, but there is not a new release to fix this problem yet.

partinfo.kitspace.org is pingable - maybe we need to remove the "dev-" part!

_Originally posted by @mdeweerd in https://github.com/xesscorp/KiCost/issues/457#issuecomment-814271394_

mdeweerd, 2021-05-04:
It's not clear what is going on because it was also requested to not use 'https://partinfo.kitspace.org/graphql' which is in fact cached data that is never updated.

set-soft, 2021-05-04:
Yes KitSpace maintainer asked not to use the cache server.

I sent an email to @kasbah he will tell use what's the problem.

kasbah, 2021-05-05:
That's right, the main partinfo API (without dev-) is the one used for the component popups on https://kitspace.org itself. We don't use any of the pricing info there so we cache it aggressively. It's not going to be very useful to get current prices.

closed 2021-05-05 by set-soft

jhaand, 2021-05-05:
I'm glad to see that Kicost works again as expected.

kasbah, 2021-05-05:
The main issue with it stopping working is that we have a limit of 5000 Octopart requests per month. This is shared between Kitspace, KiCost and HorizonEDA use. If you we want to contribute towards increasing the limit you can do so through [our Open Collective](https://opencollective.com/kitspace).