hildogjr/KiCost #492: How to use KiCost with personal Octopart API?

sujitth, 2021-05-16:
- Use a simple and clear title.
- Do not mark as "ISSUE", "FEATURE" on title, this will be tagged by the development group.

#### Issue / Problem report
1. Add the KiCost version. Type ``kicost --version`` at terminal.
    KiCost - 1.0.4

2. Add a terminal error report / txt, use ``--debug`` option on KiCost call;
    kicost -w -i %YOUR_FILE% --debug 8
   [Terminal error report.txt](https://github.com/xesscorp/KiCost/files/6489569/Terminal.error.report.txt)
3. Write the command used to call KiCost (or the graphical interface configuration);
4. One BoM to reproduce the error (with the EDA version).
    KiCad 5.1.7

#### Feature / Enhancement request
- Explain how it will help the other users.
Hello Team,

I was trying to use KiCost on a Win7x64 system, and I have installed it successfuly using pip install.
and when used it on a bom xml file, I have got this error:  https://i.imgur.com/yp0FEQs.png

I have read the related posts on the kicad forum and understood the issues and history a bit.

I have registered for an Octopart API and got it(V4)
The error I am getrting is rlated to API right?, how can I configure KiCost to use my personal API?

The team thanks your contribution.

set-soft, 2021-05-16:
Hi @sujitth !

No idea about KiCost 1.0.4.

If you want to get the current git code working (1.1.5) I can help. It have some support for Octopart API v3.

If you get a v3 key you can try the git code defining the `KICOST_OCTOPART_KEY_V3` environment variable with the key value (not sure how to define environment variables on Windows 7, but you can find a lot of explanations on internet).

Then post any errors here and I'll take a look at the code to fix it.

sujitth, 2021-05-16:
Hi @set-soft ,

Thank you very much for the quick response.

This will install the current git version? pip install -U git+https://github.com/xesscorp/KiCost/
do i need to uninstall the current version? and how to do that?

set-soft, 2021-05-16:
Hi @sujitth !

I'm not a big fun of `pip`. So I don't know all the syntax. What I do is:

1. Clone the git repo
2. Enter to the project directory
3. Run `pip install -e . ` The `-e` means **editable**, so when you update the git repo (`git pull`) the installation gets udated, no need to install again. This is good to test git code. The `.` means **this directory**.

Now about your problem with versions. The PNG shows a correct installation of the git code. It shows `pip` uninstalled 1.1.4 and installed the current code (which depends on colorama). So I guess you have 1.0.4 in your PATH, in some place listed before the place where `pip` is installing.

Check the content of PATH (https://superuser.com/questions/341192/how-can-i-display-the-contents-of-an-environment-variable-from-the-command-promp)

sujitth, 2021-05-17:
Hi @set-soft ,

Yes, it's a bit strange regarding the versions https://i.imgur.com/f4XWUHC.png
The pip list shows the correct version but kicost -version returns old one.

Here is the system path dump. nothing much useful there I guess.
[System_Path_Data.txt](https://github.com/xesscorp/KiCost/files/6490641/System_Path_Data.txt)

I have to find the correct way to uninstall the pip package.
The installed location is here:C:\Users\Sujith\AppData\Roaming\Python\Python38\site-packages\kicost
There the file version.py say's  "__version__ = '1.1.5'"

set-soft, 2021-05-17:
Hi @sujitth !

Yes, you can delete the directory inside the Anaconda installation.

Looking at your path I see a lot of Python installations:

```
C:\Program Files\Python38\Scripts\
C:\Program Files\Python38\
C:\Program Files\Python37\Scripts\
C:\Program Files\Python37\
C:\Program Files\Python36\Scripts\
C:\Program Files\Python36\
C:\PYTHON27\
C:\PYTHON27\SCRIPTS
```

You have Python 2.7, 3.6, 3.7 and 3.8 on your path. Plus the one included in Anaconda. I'll suggest trying to change the `C:\Users\Sujith\AppData\Roaming\Python\Python38\Scripts` path to be at the beggining of `PATH`, so you get the packages installed by `pip`.

sujitth, 2021-05-18:
Hi @set-soft ,
Managed to install  the kicost current version by pip3 install -U git+https://github.com/xesscorp/KiCost.git
(Uninstalled Python 37 and 36, but scripts and site-packages was still there, renamed it.)

But could't start the GUI, locale setting error:
[image]

and when tried the CLI, Octopart KEY error came :
[image]

Please tell me how to configure by using kitspace/my Octopart key to work
and how to configure to run from eeschema BoM generate option.

set-soft, 2021-05-18:
> Hi @set-soft ,
> Managed to install the kicost current version by pip3 install -U git+https://github.com/xesscorp/KiCost.git
> (Uninstalled Python 37 and 36, but scripts and site-packages was still there, renamed it.)
> 
> But could't start the GUI, locale setting error:

Ok, I patched the code, should be fixed, please test it. 

> and when tried the CLI, Octopart KEY error came :
... 
> Please tell me how to configure by using kitspace/my Octopart key to work

Are you sure you used a v3 key? Because you started saying you was trying a v4 key.
In order to debug it we need more info. Run `kicost` with the maximum debug level (--debug 10) and capture the `stderr` to a file (2> file). Edit this file to remove your key, replace it by something `*MYKEY*` or similar. Then post it so we can see what exactly is KiCost sending and what are you getting from Octopart as reply.

> and how to configure to run from eeschema BoM generate option.

Did you try `kicost --setup`? This should try to detect your KiCad installation and add kicost to your `eeschema` config file.

I think you can also just ask KiCad to run `kicost -qwi "%I"` and you'll get the same. If the GUI is working you can use `kicost --gui "%I"`

sujitth, 2021-05-18:
Hi @set-soft,

Still the GUI could't start due to the error "Cannot set locale to language en_IN"
[image]

The my API page did't say anything about API version, but their current documentation is about V4.
https://octopart.com/api/v4/getting-started
I have to mail them and ask if I can get a V3 key.

what is stopping us using the V4 key? (sorry, dont know much about the internal working)
https://octopart.com/business/api-transition-guide

or if I dont want to use my API key, will kicost work with default api or kitspace api?

the debug command is not clear for me, please give some time to test.

set-soft, 2021-05-18:
Hi @sujitth !

> Still the GUI could't start due to the error "Cannot set locale to language en_IN"

This is a configuration issue in your system, most probably missing locale data. Looks like this is very specific to your Python setup. In fact the message you sent says it can't set `en_IN` but your environment says:

```
LC_ALL=en_US.utf-8
```

Where is `en_IN` forced?

Your wxWidgets lib says there is a missmatch between Windows and C/C++, which seems to be true (en_IN vs en_US.utf-8). But KiCost doesn't set the locale. The code is only asking for the current locale, `kicost_gui.py` line 914:

```
loc = locale.getdefaultlocale()[0]
```

You can try replacing it by `loc = None`, but I doubt it can change anything.

> The my API page did't say anything about API version, but their current documentation is about V4.
> https://octopart.com/api/v4/getting-started
> I have to mail them and ask if I can get a V3 key. [...]

The code was created using API v3. I don't even have an API v3 key. So I don't even know if it works for v3. I'm not sure if trying to migrate to v4 is a good idea when we don't know if the code needs adjusts for it original purpose.

I added code that does the minimal stuff described above. You can try it defining the `KICOST_OCTOPART_KEY_V4` environment variable (using your key). The only change I did is using the URL for v4. Following the recommendation:

```
Switching from APIv3 to APIv4 REST is as simple as replacing "v3" in your query URLs with "v4/rest" and using your new APIv4 token. 
```

Lets see if this is true ;-)
 
> or if I dont want to use my API key, will kicost work with default api or kitspace api?

Yes, it will use KitSpace API.

> the debug command is not clear for me, please give some time to test.

Try to investigate why the locale setting in your environment doesn't match with the one reported by wxWidgets. You can try changing `LC_ALL`  to `en_IN` this will solve the missmatch, but won't solve the lack of `en_IN` information. Is your windows localized for India? 

On Linux systems you have to choose which locales are created, in the past all locales were included, but with the time the size became huge and now you select which ones are installed. Perhaps your Python setup can also generate `en_IN`.

sujitth, 2021-05-19:
Hi @set-soft ,

Yes, the PC location was st to India, and changed that to US
also in python changed:
[image]

latest git vrsion is installed and st the KICOST_OCTOPART_KEY_V4 env variable also

[image]

and I have emailed the debug report.

Is this tool working in your linux PC perfectly?

set-soft, 2021-05-19:
Hi @sujitth !
> 
> Yes, the PC location was st to India, and changed that to US
> also in python changed:
[snip]

Good!

> latest git vrsion is installed and st the KICOST_OCTOPART_KEY_V4 env variable also
[snip]
> 
> and I have emailed the debug report.

Thanks the error is clear, I committed a fix. Please try now.

> Is this tool working in your linux PC perfectly?

Yes. But as I said before: I don't have an Octopart key, so I can't test the octopart API. It was coded long time ago, when @kasbah kindly setup a proxy at KitSpace. Now the proxy is gone and the code is outdated. Octopart also seems to be parsing the requests with small differences, like this one: "references must be a string". The code used numbers, and it worked in the past. Now should be fixed. I also removed empty queries.

sujitth, 2021-05-19:
Hi @set-soft ,

Finally it worked (CLI)! :+1: 
The GUI mode reports the same error (I forgot to update in the previous post)

The debug file helped to find the issue, API calls was not working due to my API key basic plan(attributes limit? ), 
when changed the key to pro plan started working.
[image]

I guess its better to have 2 env variables for basic and pro key( pro plan can get lead time and life cycle data which is nice to have on the xlsx file).

The API calls counter was 106 for the bom I processed which is around 122 parts (69 when grouped)
Not sure how the counter is 106, the limit works explained here: https://octopart.com/api/v4/getting-started

will explore the eeschema kicost setup options.

and how can I install the current working code in future instead of pip3 install -U +git?

set-soft, 2021-05-19:
Hi @sujitth !

> Finally it worked (CLI)! 

Nice!

> The GUI mode reports the same error (I forgot to update in the previous post)

What's the current error? The "Cannot set locale to language en_IN" should be a warning (from wxWidgets).

> The debug file(emailed) helped to find the issue, API calls was not working due to my API key basic plan(dataset limit? ),
> when changed the key to pro plan started working.

Ok, I can try disabling `&include[]=specs` and `&include[]=datasheets`.

> I guess its better to have 2 env variables for basic and pro key( pro plan can get lead time and life cycle data which is nice to have on the xlsx file).

Ok.

> The API calls counter was 106 for the bom I processed which is around 122 parts (69 when grouped)
> Not sure how the counter is 106, the limit works explained here: https://octopart.com/api/v4/getting-started

Is this for the debug file you sent me?

> will explore the eeschema kicost setup options.
> 
> and how can I install the current working code in future instead of pip3 install -U +git?

You'll need to wait until a new release is ready. Then you'll be able to use `pip install kicost`

BTW: to keep the code working we need regression tests. And one missing is a test for Octopart API. To create it I need to capture some queries and their replies. Can I use the queries you sent me? or Can you capture some queries from the tests?

sujitth, 2021-05-20:
> Is this for the debug file you sent me?

Yes.

> Can I use the queries you sent me?

Yes please, private use only.

> Can you capture some queries from the tests?

Not clear what to here, can you please explain?

set-soft, 2021-05-20:
Hi @sujitth !

> > Can you capture some queries from the tests?
> 
> Not clear what to do here, can you please explain?

Just generate the XLSX using one of the tests. The `tests/acquire-PWM.xml` has manf# and distributor codes, so is a good example. Run kicost using `--debug 10` and redirect the stderr to a file like you did with your XML. Send me the output by mail, so I can extract the queries and remove the API key.

I pushed a new patch. This patch adds a couple of command line options. You can call KiCost using `--octopart_key YOUR_KEY --octopart_level 4p`

If you want to use the environment variables you'll need to define `KICOST_OCTOPART_EXTENDED=1` in order to get datasheet and life cycle information. The default is to exclude it, so people without a Pro plan can get it working by default.

sujitth, 2021-05-21:
Hi @set-soft ,

I have created a test project with minimal parts to limit the API calls within 500 while testing.
after the new patch .xlsx file is not generating. (I have sent both debug files, before patch and after patch)

The API key CLI option is not at all user friendly, setting the KICOST_OCTOPART_KEY_V4 env variable seems the best option.
[image]

and if the KICOST_OCTOPART_KEY_V4 env variable is not set (what to put the value?, 0?), will kicost work using the kitspace API by default?

closed 2021-05-21 by set-soft

set-soft, 2021-05-21:
Hi @sujitth !

> I have created a test project with minimal parts to limit the API calls within 500 while testing.

Ok, thanks!

> after the new patch .xlsx file is not generating. (I have sent both debug files, before patch and after patch)

My fault, while allowing the use of the basic Octopart plan I forgot a detail.
If you tried the using `KICOST_OCTOPART_EXTENDED=1` this shouldn't happend.

> The API key CLI option is not at all user friendly, setting the KICOST_OCTOPART_KEY_V4 env variable seems the best option.

Both are available! I agree with you that this is best controlled by environment variables, but @sebo83910 complained when I added the environment variables mechanism. Look at #433 

Some people find too complex defining an environment variable.

> and if the KICOST_OCTOPART_KEY_V4 env variable is not set (what to put the value?, 0?), 

You have to remove it from the environment, not just assing `0`.

>will kicost work using the kitspace API by default?

Yes, KiCost currently **always** uses KitSpace. If you enable Octopart it will also use Octopart (asking Octopart first). I'll try to add some mechanism to disable KitSpace. I think we should disable KitSpace when Octopart is enabled, but I have to do some checks before.

Thanks  for the debug files you sent me, I was able to extract the queries from there and create a regression test.

I'll suggest opening another issue for the GUI problems with the locale.

commit 7ec3edc5 (2021-05-21): Added two regression tests for the Octopart API.

- One for the Pro plan and the other for the basic one.
- Added Octopart mechanism to the dummy web server.
- Closes #492 and #433