hildogjr/KiCost #493: KiCost GUI mode not working - Win7x64

sujitth, 2021-05-22:
- Use a simple and clear title.
- Do not mark as "ISSUE", "FEATURE" on title, this will be tagged by the development group.

#### Issue / Problem report
1. Add the KiCost version. Type ``kicost --version`` at terminal.
    KiCost 1.1.5
2. Add a terminal error report / txt, use ``--debug`` option on KiCost call;
    kicost -w -i %YOUR_FILE% --debug 8
3. Write the command used to call KiCost (or the graphical interface configuration);
    kicost
4. One BoM to reproduce the error (with the EDA version).

#### Feature / Enhancement request
- Explain how it will help the other users.
Hi @set-soft ,

The KiCost GUI mode not able to start, please refer the attached error report.
[gui_error.txt](https://github.com/xesscorp/KiCost/files/6526352/gui_error.txt)

The team thanks your contribution.

set-soft, 2021-05-22:
Hi @sujitth!

Thanks for adding this report. Could you add the output from `kicost --info`.

I suspect this could be related to a wxWidgets problem found in version 4.1: https://github.com/DeepLabCut/DeepLabCut/issues/682

sujitth, 2021-05-22:
Hi @set-soft ,

Please find the info:

[image]

set-soft, 2021-05-22:
Ok, can you try to downgrade wx to 4.0.7 as suggested in the issue I mentioned?

I think something like this should work:

```
pip3 install -U --upgrade wxPython==4.0.7
```

sujitth, 2021-05-22:
Hi @set-soft ,

wx 4.0.7 is installed.
[image]

[image]

still gui couldn't start, please find the error report.
[gui_error2.txt](https://github.com/xesscorp/KiCost/files/6526851/gui_error2.txt)

set-soft, 2021-05-22:
This is similar to what I have on Linux:

```
KiCost v.1.1.5 at Python 3.9.2 on Linux-5.10.0-6-amd64-x86_64-with-glibc2.31(64bit). Graphical library: 4.0.7 gtk3 (phoenix) wxWidgets 3.0.5.
```

So I don't think this is a problem of KiCost setting some locale. But just in case: Can you try commenting the `import locale` and replacing `loc = locale.getdefaultlocale()[0]` by `loc = None`?

sujitth, 2021-05-24:
Hi @set-soft ,

Please find attached the error report after made the changes as stated above.
[Gui_error.txt](https://github.com/xesscorp/KiCost/files/6532137/Gui_error.txt)

set-soft, 2021-05-24:
Hi @sujitth !

I added wxWidgets locale initialization. This also logs what you got initialized for wxWidgets and what Python thinks is the system locale. In my case:

```
$ kicost --debug 1 --gui tests/test2.xml
DEBUG:wxWidgets locale Spanish (Argentina) (es_AR) system: ('es_AR', 'UTF-8') (kicost - kicost_gui.py:1171)
DEBUG:Loaded KiCost v.1.1.5 (kicost - kicost_gui.py:633)
```

Please try it and tell me what you get.

sujitth, 2021-05-25:
Hi @set-soft ,

Please find attached the error report after installing latest kicost.
Gui_test3.txt- is after making the change loc = None.

[Gui_test2.txt](https://github.com/xesscorp/KiCost/files/6536930/Gui_test2.txt)
[Gui_test3.txt](https://github.com/xesscorp/KiCost/files/6536933/Gui_test3.txt)

set-soft, 2021-05-25:
> Hi @set-soft ,
> 
> Please find attached the error report after installing latest kicost.

 I need the output you get using: `kicost --debug 1 --gui tests/test2.xml` like I shown above. Do not apply the `loc = None` change, we need access to the `locale` module, and we already know that requesting `locale` isn't the problem.

IMHO this is a wxPython bug, but we should find a workaround.

sujitth, 2021-05-25:
Hi @set-soft ,

Yes, the file Gui_test2.txt was the test output before making the loc code changes using the command provided.

I have reverted the changes made and done the test again, same output.
[Gui_test4.txt](https://github.com/xesscorp/KiCost/files/6541438/Gui_test4.txt)

set-soft, 2021-05-25:
Hi @sujitth !

This is not the last git code. What do you get from `kicost --info`?
The last code shows information about the locale. Try reinstalling with `pip3 install -U git+https://github.com/xesscorp/KiCost.git`
`kicost --info` should inform the git hash you installed.

sujitth, 2021-05-26:
Hi @set-soft ,

Please find the kicost --info attached below:

[image]

set-soft, 2021-05-26:
Hi @sujitth!

And this version doesn't print any debug information about the locale?
I mean: `kicost --debug 1 --gui tests/test2.xml` doesn't print something like this:

```
DEBUG:wxWidgets locale Spanish (Argentina) (es_AR) system: ('es_AR', 'UTF-8') (kicost - kicost_gui.py:1171)
```

sujitth, 2021-05-26:
Hi @set-soft ,

No.

set-soft, 2021-05-26:
Please check that line 1171 of `kicost_gui.py` is:

```
logger.debug('wxWidgets locale {} ({}) system: {}'.format(loc.GetLocale(), loc.GetName(), locale.getlocale()))
```

If this is the case:

1. Try redirecting `stderr` to a file
2. Try replacing the call to `logger.debug` by a simple `print`

set-soft, 2021-05-26:
This log shows the source of the problem. wxWidgets is failing to set your locale.
Python is getting it right: english for India. But the value looks from Win32 API.
wxWidgets should be reporting something like: locale English (India) (en_IN)

About the changes you made for #492: What's the current value for the environment variable `LC_ALL`? Is this `en_IN`?

About point 2: never mind, it was a debug suggestion assuming you didn't get the message that you are getting.

Things will be solved when we get wxWidgets locale in sync with your system.

sujitth, 2021-05-26:
Hi @set-soft ,

LC_ALL env var:
[image]

and currently in Python:
[image]

The GUI mode was working on KiCost - 1.0.4, the issue started after updating to current version.

set-soft, 2021-05-26:
Your LC_ALL is wrong, doesn't match your system. Please try to define it as `en_IN`.
Python is working well because is asking to Windows. According to one of the links you sent wxWidgets is most probably asking wrongly. The proposed solution is to replace the whole locale mechanism ... an overkill.

About KiCost 1.0.4: Which wxPython and wich Python did you use with it? 

Again: I think this is purely a wxWidgets problem, nothing related to KiCost. The fact that `wx.Locale(wx.LANGUAGE_DEFAULT)` doesn't fail, but then `loc.GetLocale()` and `loc.GetName()` returns empty strings is a probe of this.

BTW: Your setup doesn't seem to be supported by wxWidgets: https://wxpython.org/Phoenix/docs/html/wx.Language.enumeration.html#wx-language  (no wx.LANGUAGE_ENGLISH_INDIA)

Things to try:
1. Properly define LC_ALL
2. Try replacing `wx.LANGUAGE_DEFAULT` by `wx.LANGUAGE_ENGLISH_US`

sujitth, 2021-05-26:
Hi @set-soft ,
Manually set LC_ALL to `en_IN` but no effect.

> Which wxPython and wich Python did you use with it?

Python 3.6 or 3.7(anaconda), wxPython- I did't checked that time.

> 2. Try replacing wx.LANGUAGE_DEFAULT by wx.LANGUAGE_ENGLISH_US

Where to do this?

and what is this thing "locale setting" is related to kicost functionality? currency setting related?

set-soft, 2021-05-26:
Hi @sujitth !

> Manually set LC_ALL to `en_IN` but no effect.

:-(
POSIX code uses it.

> > Which wxPython and wich Python did you use with it?
> 
> Python 3.6 or 3.7(anaconda), wxPython- I did't checked that time.
> 
> > 1. Try replacing wx.LANGUAGE_DEFAULT by wx.LANGUAGE_ENGLISH_US
> 
> This worked, gui mode is working now.

Nice!
 
> and what is this thing "locale setting" is related to kicost functionality? currency setting related?

Currently: nothing at all. kiCost is currently pure "en_US". 
Your problem is most probably related to the fact that wxWidgets doesn't understand your locale, that's all. KiCost doesn't set the locale, and doesn't really need it. The getdefaultlocale() value is just for future support, this currently discarded.

Please try with the last patch.

About the fail to generate test2.xlsx: The screen capture is incomplete, the scroll bar says more text is there.

closed 2021-05-26 by set-soft

sujitth, 2021-05-27:
Hi @set-soft ,

> The screen capture is incomplete, the scroll bar says more text is there.

No it seems, in windows the scroll bar in command prompt and content has no connection in general.
and I have attached the GUI screenshot later in the last command.

Updated to the latest git version and GUI could't start:
[image]

Test- kicost --debug 1 --gui tests/test2.xml
[image]

set-soft, 2021-05-27:
Hi @sujitth !

> > The screen capture is incomplete, the scroll bar says more text is there.
> 
> No it seems, in windows the scroll bar in command prompt and content has no connection in general.
> and I have attached the GUI screenshot later in the last command.

Then: are you sure you didn't wait enough? If KitSpace is saturated it will just time-out, see #417 
The time-out is long.

> Updated to the latest git version and GUI could't start:

Then wxWidgets is even more broken than what I thought ;-) (You can't change the locale twice using the API!)
I can't force "en_US" because this locale doesn't have to be installed on Linux boxes (I don't have it in fact). 
I added a command line option `--force_en_us` please tell me if using it solves the problem.

sujitth, 2021-05-27:
Hi @set-soft ,

> If KitSpace is saturated it will just time-out, see #417 ,The time-out is long.

The kitspace API limit was reached and @kasbah has reset it.
This debug info is from my another test file:
[image]

> I added a command line option --force_en_us please tell me if using it solves the problem.

This works now:
[image]

[image]

I have selected only 2 distributors only, what is the Local generated on the XL sheet?
[image]

set-soft, 2021-05-27:
Hi @sujitth !

> > If KitSpace is saturated it will just time-out, see #417 ,The time-out is long.
> 
> The kitspace API limit was reached and @kasbah has reset it.
> This debug info is from my another test file:

Sorry, I don't follow you. This looks like KitSpace failing. Do you mean that you have a BoM that can generate it even when other BoMs are working? If this is the case please send me it by mail so I can try to find where is the problem.

> > I added a command line option --force_en_us please tell me if using it solves the problem.
> 
> This works now:

Ok. So now we have a workaround for systems where wxWidgets locale misserably fails. Thanks for testing it.

> I have selected only 2 distributors only, what is the Local generated on the XL sheet?

This is the mechanism that allows defining your own prices. The SW1 component defines:

```xml
        <field name="manf#">RS-282G05A3-SM RT</field>
        <field name="kicost:pricing">1:0.10;10:0.03</field>
        <field name="kicost:link">http://www.xess.com</link>
```

So you get a Unit$ of $0,03 buying 10 parts. And the URL for the part is http://www.xess.com

sujitth, 2021-05-30:
Hi @set-soft ,

Thank you very much for resolving the issue.

commit 3b598118 (2021-05-26): Added fallback to en_US locale for wxWidgets.

- Fixes #493