hildogjr/KiCost #495: [RFC] Bizarre version format

set-soft, 2021-05-24:
The current version format is somehow confusing.

According to the SEMVER checker is valid, but it can't say much about what it means.

Try: https://jubianchi.github.io/semver-check/#/version/v.1.1.5
And then try: https://jubianchi.github.io/semver-check/#/version/v1.1.5

And according to this "strict tester" it isn't valid: https://regexr.com/39s32

@hildogjr @xesscorp : Should we change it to v1.1.5?

hildogjr, 2021-05-24:
Change "v.1.1.5" to "v1.1.5"? This is just the difference on the `kicost --version` (and `kicost --info`) command response, the real string control is just `__version__ = '1.1.5'` at `version.py` file.

But every change in sense of standardization is welcome. Could we include https://github.com/xesscorp/KiCost/issues/232 in this plan? (create some official freeze stable-banch and the pipeline for release at PyPI in it merge)

hildogjr, 2021-05-24:
If I remember correct, I created the `--version` and `--info` command to help users to report issues. I decided by "v.x.x.x" just because appeared right at that time.

closed 2021-05-25 by set-soft

set-soft, 2021-05-25:
> But every change in sense of standardization is welcome. Could we include #232 in this plan? (create some official freeze stable-banch and the pipeline for release at PyPI in it merge)

I reply there

commit afec1c0c (2021-05-25): Changed "v.VERSION" to "vVERSION"

- Closes #495