hildogjr/KiCost #496: Time data format leads to error in KiCost GUI

Light411, 2021-05-27:
Hi everyone, 

Thanks for this amazing project ! 
I'm experiencing an issue using the GUI. 

When Kicosting my Bom file, it returns me the following : 
Starting cost processing ...
ERROR:Internal error: time data 'Wed May 26 18:00:22 2021' does not match format '%a %b %d %H:%M:%S %Y' (kicost - kicost_gui.py:149)
 
I'm not placing any extra commands. 
I'm using the last version from the Git repo : KiCost v1.1.5

Do you have an idea of what could lead to this kind of issue ? 

Kind regards, 
Jean

set-soft, 2021-05-27:
Hi @Light411 !

Please try running KiCost like this: `kicost --debug 10 --gui YOUR_XML_FILE`

This will tell us the point in the code where the problem was detected.

set-soft, 2021-05-27:
I think I found the error. Please check that 538c9f4 fixes the problem

closed 2021-05-27 by Light411

set-soft, 2021-05-27:
I think KiBoM only sorts like this:
1) The type of component (C* before R*)
2) The component value (22 pF before 1 nF)
Then inside the same row, references are sorted in "natural order". But you can't sort the rows by reference.