hildogjr/KiCost #499: Unable to launch GUI on Windows

Light411, 2021-06-01:
Hi everyone, 

I'm experiencing an issue when trying to launch the GUI on windows. 
I've installed kicost from github to get the freshest version. 

I got the following error message when lauching kicost. 

``` C:\Users\amel-\AppData\Local\Programs\Python\Python39\Scripts>kicost
Traceback (most recent call last):
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\Scripts\kicost-script.py", line 33, in <module>
    sys.exit(load_entry_point('kicost==1.1.5', 'console_scripts', 'kicost')())
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\Scripts\kicost-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\lib\importlib\metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\lib\importlib__init.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 855, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\lib\site-packages\kicost__main.py", line 43, in <module>
    from .kicost import kicost, output_filename, kicost_gui_notdependences  # kicost core functions. # noqa: E402
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\lib\site-packages\kicost\kicost.py", line 66, in <module>
    from .spreadsheet import create_spreadsheet, Spreadsheet
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\lib\site-packages\kicost\spreadsheet.py", line 41, in <module>
    from validators import url as validate_url  # URL validator.
ModuleNotFoundError: No module named 'validators' ```

I cannot run kicost -version either. 

Any idea of how to solve this issue ? 

Kind regards, 
Jean

set-soft, 2021-06-01:
Hi @Light411 !

How did you install it? Using `pip install -U .`?

Note that validators is one of the dependencies named in `requirements.txt` so `pip` should install it. 

BTW: You are using Python 3, I saw other people using KiCost with Python 3 on Windows, but: if you are using KiCad 5 be careful because KiCad 5 uses Python 2.7. I suggest using the same Python used by KiCad. If you don't use KiCad or you use KiCad 5.99 this isn't an issue.

Light411, 2021-06-02:
Sorry, this was a dumb mistake, I typed 
```
pip install git+https://github.com/xesscorp/KiCost 

```
Without the -U it didn't installed validators.. 
I reinstalled everything with 

```
pip install -U git+https://github.com/xesscorp/KiCost 

```

Now I get the locale issue with wxPython... 

```kicost --info ```

returns 
```
KiCost v1.1.5 (5b6688a-2021-05-30) at Python 3.9.5 on Windows-10-10.0.19041-SP0(64bit). Graphical library: 4.1.1 msw
(phoenix) wxWidgets 3.1.5.
```
Then if I try to launch the gui I get the unknown locale error..

```
 C:\Users\amel-\OneDrive\Documents\stage\1-documentation\1leg\KiCAD_files>kicost --debug 1 --gui Shield_Nucleo.csv
Traceback (most recent call last):
  File "C:\Users\amel-\AppData\Local\Programs\Python\Python39\Scripts\kicost-script.py", line 33, in <module>
    sys.exit(load_entry_point('kicost==1.1.5', 'console_scripts', 'kicost')())
  File "c:\users\amel-\appdata\local\programs\python\python39\lib\site-packages\kicost\__main__.py", line 417, in main
    main_real()
  File "c:\users\amel-\appdata\local\programs\python\python39\lib\site-packages\kicost\__main__.py", line 342, in main_real
    kicost_gui(args.force_en_us, [os.path.abspath(fileName) for fileName in args.gui])
  File "c:\users\amel-\appdata\local\programs\python\python39\lib\site-packages\kicost\kicost_gui.py", line 1176, in kicost_gui
    logger.debug('wxWidgets locale {} ({}) system: {}'.format(loc.GetLocale(), loc.GetName(), locale.getlocale()))
  File "c:\users\amel-\appdata\local\programs\python\python39\lib\locale.py", line 593, in getlocale
    return _parse_localename(localename)
  File "c:\users\amel-\appdata\local\programs\python\python39\lib\locale.py", line 501, in _parse_localename
    raise ValueError('unknown locale: %s' % localename)
ValueError: unknown locale: fr-FR
```
I've tried to understand the workaround you explained in similar threads but failed to understand them. 
I tried downgrading wxpython without much sucess. 
What should I do ?

set-soft, 2021-06-02:
The locale settings for your system seems to be highly broken, in your case Python itself is failing, not wxWidgets (known to be buggy).

1. Try using the `--force_en_us` command line option.
2. I commited a patch to catch this error, inform the problem and try to continue. Please check if this solves your problem.
3. Do you have KiCad? In this case I **strongly** recommend using the Python and the wxWidgets shipped with KiCad. This seems to be the real solution.

Light411, 2021-06-02:
1. I've tried the ```--force_en_us``` workaround but it didn't help.
2. I'll do it tomorrow and send the results 
3. Can you explain me how do I activate the python environment shipped with KiCad so that I can use the pip command from KiCad python ? I'm not sure how to procede. 

The fr-FR variable bothered me as well, I was also expecting fr_FR. But I haven't tweaked any of the default variables I'm too noob to do that :D. Is there a specific file I should edit ?

set-soft, 2021-06-02:
> 1. I've tried the `--force_en_us` workaround but it didn't help.

What do you get when trying to use `en_US`? (other than urticaria ;-)

> 2. I'll do it tomorrow and send the results

Ok.

> 3. Can you explain me how do I activate the python environment shipped with KiCad so that I can use the pip command from KiCad python ? I'm not sure how to procede.

Don't really know about Windows. I know KiCad includes the wxWidgets module and other stuff, but never tried it, I don't really use Windows.

> The fr-FR variable bothered me as well, I was also expecting fr_FR. But I haven't tweaked any of the default variables I'm too noob to do that :D. Is there a specific file I should edit ?

Try looking at your environment, for PowerShell the documentation is: https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.core/about/about_environment_variables?view=powershell-7.1

If you find a strange definition you'll need to find where is defined. But first take a look at the variables.

set-soft, 2021-06-15:
I tried to install KiCad on Windows 7 and I took a look at what KiCad includes. Lamentably it has wx v3.0.2.0 and KiCost was created for >= 4.0

set-soft, 2021-06-15:
Now KiCost supports wx v3.0.2.0.

I installed KiCad 5.1.10 on Windows 7 using the default options. Python is installed in: `C:\Program Files\KiCad\bin\python.exe`. A really bad idea, I explain it latter.

Running it you get Python 2.7.18.

On a **Power Shell** window you can write:

```
$env:Path += ";C:\Program Files\KiCad\bin"
```

And then just running `python` will work. Once inside the Python interpreter you can check for wxWidgets like this:

```
>>> import wx
>>> print wx.__version__
```

I get `3.0.2.0`

Assuming you have Git (http://git-scm.com/download/win). You can install KiCost using this Python just running:

```
pip install -U git+https://github.com/hildogjr/KiCost.git
```

This will create a **shortcut** for KiCost in "C:\Program Files\KiCad\bin\kicost.exe". But pip for Python 2.7 on Windows has a huge bug: paths with spaces aren't supported. If you run KiCost you'll get:

```
failed to create process (C:\Program "Files\KiCad\bin\python.exe" "C:\Program Files\KiCad\bin\kicost.exe")
```

The bug is obvious and is supposed to be fixed in the last `pip` ([look here](https://stackoverflow.com/questions/24627525/fatal-error-in-launcher-unable-to-create-process-using-c-program-files-x86/)). I tried upgrading pip 20.0.2 (from KiCad) to pip 20.3.4. Got the same error.

Solutions? Not sure, here are some:

1. Move KiCad to a path without spaces
2. Explicitly run the script: `python "C:\Program Files\KiCad\bin\kicost-script.py"`

@hildogjr do you have access to a Windows box to check this?

set-soft, 2021-06-16:
A couple of notes:

- On Windows 7 using Python 2.7 from KiCad (and wxWidgets 3.0.2.0 also from KiCad) I get: `wxWidgets locale Spanish (Argentina) (es_AR) system: ('Spanish_Argentina', '1252')` quite similar to what I get on Linux: `wxWidgets locale Spanish (Argentina) (es_AR) system: ('es_AR', 'UTF-8')`
- The locale problem is a know issue in wxPython 4: [Possible Locale Mismatch on Windows](https://wxpython.org/Phoenix/docs/html/MigrationGuide.html#possible-locale-mismatch-on-windows) The `-` is mentioned. If @Light411 or someone else can reproduce it we can try what the article says.

hildogjr, 2021-07-08:
i tried to install and got:

```bash
 pip install git+https://github.com/hildogjr/KiCost
Collecting git+https://github.com/hildogjr/KiCost
  Cloning https://github.com/hildogjr/KiCost to c:\users\test\appdata\local\temp\pip-req-build-tkavbot4
  Running command git clone -q https://github.com/hildogjr/KiCost 'C:\Users\test\AppData\Local\Temp\pip-req-build-tkavbot4'
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x00000000043D9AF0>: Failed to establish a new connection: [Errno 11004] getaddrinfo failed')': /simple/beautifulsoup4/
```

Appear that `https://pypi.org/` is offline, at least I could not  reach it. I will try again later.

hildogjr, 2021-07-08:
I just tested now got the right results, also the messages appear correctly redirect to the text box.

closed 2021-07-08 by hildogjr