hildogjr/KiCost #505: kitspace service down?

JordiXTR, 2021-06-22:
- Use a simple and clear title.
- Do not mark as "ISSUE", "FEATURE" on title, this will be tagged by the development group.

#### Issue / Problem report
1. Version: KiCost v1.1.5
2. Call
```
kicost.exe --currency EUR --variant "P" --eda kicad -i D:\Esp32-Vending.xml --overwrite --debug 8

Command error. Return code 5
Error messages:
DEBUG:Octopart API configured to enabled False key None level 4 extended False (kicost.distributors - log.py:114)
DEBUG:Started KiCost v1.1.5 (c881fa4-%as) at Python 3.6.7 on Windows-10-10.0.19041-SP0(64bit). Graphical library: 4.1.1 msw (phoenix) wxWidgets 3.1.5. (kicost - log.py:114)
DEBUG:# Getting from XML 'Esp32-Vending.xml' KiCad BoM... (kicost.edas - log.py:114)
DEBUG:Getting authorship data... (kicost.edas - log.py:114)
DEBUG:Getting parts library... (kicost.edas - log.py:114)
DEBUG:Getting components... (kicost.edas - log.py:114)
DEBUG:# Removing do not populate parts... (kicost - log.py:114)
DEBUG:Splitting subparts in the manufacture / distributors codes... (kicost.edas - log.py:114)
DEBUG:R7 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R6 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R45 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R44 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R4 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R5 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R3 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R2 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R57 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R56 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R55 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R54 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R41 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R40 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R61 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R60 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
DEBUG:R38 >> ['manf#'] (kicost.edas - log.py:114)
DEBUG:part/qty>> MCWR06X000 PTL		part>>MCWR06X000 PTL	qty>>1 (kicost.edas - log.py:114)
DEBUG:OrderedDict([('value', '0R'), ('libpart', 'Device:R_US'), ('footprint', 'Resistor_SMD:R_0603_1608Metric'), ('datasheet', '~'), ('mouser price aprox (3000)', '0.0035'), ('manf#', 'MCWR06X000 PTL'), ('variant', 'P, E'), ('manf#_qty', '1')]) (kicost.edas - log.py:114)
[... 627 lines cut ...]
<head><title>504 Gateway Time-out</title></head>

<body bgcolor="white">

<center><h1>504 Gateway Time-out</h1></center>

<hr><center>nginx/1.14.0 (Ubuntu)</center>

</body>

</html>

 (kicost.distributors - log.py:114)

Progress:   0%|          | 0/88 [01:00<?, ?part/s]
                                                  
DEBUG:Status Code: <504> (kicost.distributors - log.py:114)

Progress:   0%|          | 0/88 [01:00<?, ?part/s]ERROR:One of the internal Kitspace services may experiencing problems. Contact the Kitspace support. (kicost - __main__.py:419)

Progress:   0%|          | 0/88 [01:00<?, ?part/s]
```
Seems like the request its giving timeout. Any feedback about this?

Thanks

set-soft, 2021-06-22:
Hi @JordiXTR !

> Seems like the request its giving timeout. Any feedback about this?

Yes, KitSpace server gives time-out when the maximum number of requests is reached. I already asked to Kaspar to fix it. But isn't prioritary for him. The server code is JavaScript and I don't know enough JS to solve it.

Lamentably KitSpace has a very small number of requests available. I have some solutions in mind, but will take some time before they get implemented.

JordiXTR, 2021-06-22:
Oh I see the --no_price option

Sorry to not read the docs :P

With that for the moment is enough.

Is the max request limited by my IP address?

set-soft, 2021-06-22:
I think KitSpace has a general limit, for all KiCost users.

set-soft, 2022-06-29:
See #518

closed 2022-06-29 by set-soft