hildogjr/KiCost #506: Status of pypi.org

romain145, 2021-07-15:
https://pypi.org/project/kicost/

The latest release is 1.1.4 (2020-03-24).
Images don't show, and the documentation link are broken: https://xesscorp.github.io/KiCost
The links on readthedocs are even older: https://kicost.readthedocs.io/en/latest/

set-soft, 2021-07-15:
Hi @romain145 !

Thanks for reporting. Lamentably PyPi can't be modified, you must add a new release.

The next release fixes the images:  https://test.pypi.org/project/kicost/

The documentation issue is new, I think this got broken when @devbisme transferred the project to @hildogjr 

We should ask Hildo where are the docs uploaded.

hildogjr, 2021-07-15:
I fix some links, I will let this opened until complete check.

Still need to add the doc generation (or at least check) to `.github\workflows`.

set-soft, 2021-10-12:
@hildogjr I removed `ALLOW_TAG_PREFIX` try again. 
We are using `v` as prefix so we should be ok with:
```
- in case your tag has a 'v' prefix, you can safely remove 'ALLOW_TAG_PREFIX' env.var
```

hildogjr, 2021-10-12:
I made some cleanup od the old tags and re-create `v1.1.6`, updating `version.py`. Now I got:

```
Run docker://antonyurchenko/git-release:latest
/usr/bin/docker run --name antonyurchenkogitreleaselatest_861b50 --label e1cc51 --workdir /github/workspace --rm -e GITHUB_TOKEN -e DRAFT_RELEASE -e PRE_RELEASE -e CHANGELOG_FILE -e INPUT_ARGS -e HOME -e GITHUB_JOB -e GITHUB_REF -e GITHUB_SHA -e GITHUB_REPOSITORY -e GITHUB_REPOSITORY_OWNER -e GITHUB_RUN_ID -e GITHUB_RUN_NUMBER -e GITHUB_RETENTION_DAYS -e GITHUB_RUN_ATTEMPT -e GITHUB_ACTOR -e GITHUB_WORKFLOW -e GITHUB_HEAD_REF -e GITHUB_BASE_REF -e GITHUB_EVENT_NAME -e GITHUB_SERVER_URL -e GITHUB_API_URL -e GITHUB_GRAPHQL_URL -e GITHUB_WORKSPACE -e GITHUB_ACTION -e GITHUB_EVENT_PATH -e GITHUB_ACTION_REPOSITORY -e GITHUB_ACTION_REF -e GITHUB_PATH -e GITHUB_ENV -e RUNNER_OS -e RUNNER_NAME -e RUNNER_TOOL_CACHE -e RUNNER_TEMP -e RUNNER_WORKSPACE -e ACTIONS_RUNTIME_URL -e ACTIONS_RUNTIME_TOKEN -e ACTIONS_CACHE_URL -e GITHUB_ACTIONS=true -e CI=true -v "/var/run/docker.sock":"/var/run/docker.sock" -v "/home/runner/work/_temp/_github_home":"/github/home" -v "/home/runner/work/_temp/_github_workflow":"/github/workflow" -v "/home/runner/work/_temp/_runner_file_commands":"/github/file_commands" -v "/home/runner/work/KiCost/KiCost":"/github/workspace" antonyurchenko/git-release:latest package/kicost_1.1.5-5_all.deb python3-validators_0.14.2-2_all.deb
package/python3-validators_0.14.2-2_all.deb
DEBUG git-release v4.1.1                           
FATAL error reading changelog: changelog file does not contain version 1.1.6 
```

commit 851b9b2a (2021-07-15): Fix #506 - doc links