hildogjr/KiCost #507: gui start failed on windows

imyaFamiliev, 2021-09-14:
KiCost v.1.1.4

try to launch gui version on windows 10  
install  wxPython before using 
pip3 install -U wxPython

```
kicost --debug
Traceback (most recent call last):
  File "C:\Users\yakimov\AppData\Local\Programs\Python\Python39\Scripts\kicost-script.py", line 33, in <module>
    sys.exit(load_entry_point('kicost==1.1.4', 'console_scripts', 'kicost')())
  File "c:\users\yakimov\appdata\local\programs\python\python39\lib\site-packages\kicost\__main__.py", line 253, in main
    kicost_gui() # Use the user gui if no input is given.
  File "c:\users\yakimov\appdata\local\programs\python\python39\lib\site-packages\kicost\kicost_gui.py", line 1065, in kicost_gui
    frame = formKiCost(None)
  File "c:\users\yakimov\appdata\local\programs\python\python39\lib\site-packages\kicost\kicost_gui.py", line 367, in __init__
    fgSizer1.Add(self.m_gauge_process, 1, wx.ALL|wx.EXPAND|wx.ALIGN_CENTER_HORIZONTAL|wx.ALIGN_CENTER_VERTICAL, 5)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL)) || !(flags & (wxALIGN_RIGHT | wxALIGN_CENTRE_HORIZONTAL))" failed at ..\..\src\common\sizer.cpp(1523) in wxGridSizer::DoInsert(): wxEXPAND flag will be overridden by alignment flags

```

set-soft, 2021-09-15:
Try the git code, the code on PyPi won't work.

imyaFamiliev, 2021-09-15:
it work, thanks!

closed 2021-09-18 by imyaFamiliev