hildogjr/KiCost #509: [RFC] API plug-ins

set-soft, 2021-09-29:
We all know that KiCost is saturating KitSpace gateway to Octopart. In order to reduce its load I would like to introduce alternative mechanisms. The most important distributors provides some kind of API to do stock queries. I personally use Digi-Key so the first API I'm trying is this.

I created a branch called `api_ext_plugin` (https://github.com/hildogjr/KiCost/tree/api_ext_plugin). In this branch I'm starting to play with the concept.

The first plug-in is here: [Digi-Key plug-in](https://github.com/set-soft/kicost-digikey-api-v3) 

The setup is a little bit complex, but doing it you get 1000 queries a day for free (slow queries by the way).

Any comments? Anybody can help to test it?

hildogjr, 2021-10-14:
Agree with your observations.

I made a quick look at the code. Apparently fails on `from kicost_digikey_api_v3 import foo` is this related with https://pypi.org/project/digikey-api/? (I think is missing some file)

I would rename `api_partinfo_digikey_pi.py` to simple `api_digikey.py`.

set-soft, 2021-10-18:
> I made a quick look at the code. Apparently fails on `from kicost_digikey_api_v3 import foo` is this related with https://pypi.org/project/digikey-api/? (I think is missing some file)

It works here:

```
$ pytest-3 --log-cli-level debug -k test_acquire_PWM_dk
==================================================================================================================================================== test session starts =====================================================================================================================================================
platform linux -- Python 3.9.2, pytest-6.0.2, py-1.10.0, pluggy-0.13.0
rootdir: /.../KiCost
plugins: xdist-2.2.0, forked-1.3.0
collected 63 items / 62 deselected / 1 selected                                                                                                                                                                                                                                                                              

tests/test_kicost.py::test_acquire_PWM_dk 
------------------------------------------------------------------------------------------------------------------------------------------------------- live log call --------------------------------------------------------------------------------------------------------------------------------------------------------
DEBUG    root:test_kicost.py:340 Test name: acquire_PWM_dk
DEBUG    root:test_kicost.py:69 Testing using: src/kicost --debug 10 --disable_api KitSpace --enable_api Digi-Key -o /.../KiCost/tests/acquire_PWM_dk.xlsx -wi /.../KiCost/tests/acquire-PWM
DEBUG    root:test_kicost.py:246 Converting to CSV
DEBUG    root:test_kicost.py:69 Comparing using: diff -u /.../KiCost/tests/expected_test/acquire_PWM_dk.csv /.../KiCost/tests/result_test/acquire_PWM_dk.csv
DEBUG    root:test_kicost.py:74 Converting to TXT
DEBUG    root:test_kicost.py:69 Comparing using: diff -u /.../KiCost/tests/expected_test/acquire_PWM_dk.xlsx.txt /.../KiCost/tests/result_test/acquire_PWM_dk.xlsx.txt
INFO     root:test_kicost.py:336 acquire_PWM_dk OK
PASSED                                                                                                                                                                                                                                                                                                                 [100%]

============================================================================================================================================== 1 passed, 62 deselected in 2.19s ==============================================================================================================================================
```

I installed the plug-in using `pip install -e .` from its repo, I can see the *link*:

```
$ ls ~/.local/lib/python3.9/site-packages/kicost-digikey-api-v3.egg-link 
/home/salvador/.local/lib/python3.9/site-packages/kicost-digikey-api-v3.egg-link
```

And the module can be imported without problems:

```
$ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import kicost_digikey_api_v3
>>> from kicost_digikey_api_v3 import by_digikey_pn, by_manf_pn, by_keyword, configure  # noqa: E402
>>> 
```

set-soft, 2021-11-02:
More news: now you can ask for international prices. I added an example to ask for prices in Spain.

set-soft, 2022-03-10:
After a long time:
- Farnell/Newark/CPC API available

@hildogjr : do you think is time to merge it and debug?

hildogjr, 2022-03-12:
Sorry be the delay. I am attempting a new job position, I will be quite busy next months.

The branch have passed the automatic tests. If you can confirm, I think will be fine to merge and release (mark the Gittag "vX.X").

set-soft, 2022-03-18:
> Sorry be the delay. I am attempting a new job position, I will be quite busy next months.
> 
> The branch have passed the automatic tests. If you can confirm, I think will be fine to merge and release (mark the Gittag "vX.X").

Ok, I added the TME API and will revisit the Digi-Key API before the merge.

set-soft, 2022-03-21:
I merged, but didn't release yet.

I would like to see reports from people using it. The code passes all tests for all Python versions. 

The Digi-Key plug-in now is available from PyPi and is a regular dependency. I also tested that KiCost runs without the plugin installed.

But again, I would like to see success reports before releasing it.

set-soft, 2022-06-29:
These APIs are in use since 1.1.8 and looks like they are in use because we got 3 reports for problems using Mouser API for people using languages other than english.
I'm closing this.

closed 2022-06-29 by set-soft

mik1234mc, 2022-07-09:
Confirming that Digikey, Mouser, Farnell and TME APIs are working in our case. Thanks for your effort!