hildogjr/KiCost #511: Error with Kitspace

michael-vacek, 2021-12-23:
Hi all,

I tried to use you KiCost app v1.1.6 but without success. The execution always fails with "ERROR:One of the internal Kitspace services may experiencing problems. Contact the Kitspace support. (kicost - __main__.py:419)". I there any workaround to this? I saw that there is also support for using direct Mouser/Digikey account IDs. I think, this is a more realistic usage to enter user's account because the final goal is to buy the parts.

I there anything I miss?

Thank you,
Michael

I attach a shorten execution log below:

```Run command:
kicost -w -i /home/me/ad9643_pico.xml --debug 8

Command error. Return code 5
Error messages:
DEBUG:Octopart API configured to enabled False key None level 4 extended False (kicost.distributors - api_octopart.py:98)
DEBUG:Started KiCost v1.1.6 (release) at Python 3.9.2 on Linux-5.10.0-9-amd64-x86_64-with-glibc2.31(64bit). Graphical library: 4.0.7 gtk3 (phoenix) wxWidgets 3.0.5. (kicost - __main__.py:405)
DEBUG:# Getting from XML 'ad9643_pico.xml' KiCad BoM... (kicost.edas - eda_kicad.py:72)
DEBUG:Getting authorship data... (kicost.edas - eda_kicad.py:79)
DEBUG:Getting parts library... (kicost.edas - eda_kicad.py:89)
DEBUG:Getting components... (kicost.edas - eda_kicad.py:111)
DEBUG:# Removing do not populate parts... (kicost - eda.py:248)
DEBUG:Splitting subparts in the manufacture / distributors codes... (kicost.edas - tools.py:388)
DEBUG:U2 >> ['manf#'] (kicost.edas - tools.py:443)
DEBUG:part/qty>> TPS7A8101DRB		part>>TPS7A8101DRB	qty>>1 (kicost.edas - tools.py:623)
DEBUG:OrderedDict([('value', 'TPS7A8101'), ('libpart', 'eltlib:TPS7A8101'), ('footprint', 'fpHPTFL:DRB(S-PVSON-N8)-8-1EP_3x3mm_Pitch0.65mm'), ('manf#', 'TPS7A8101DRB'), ('manf#_qty', '1')]) (kicost.edas - tools.py:482)
DEBUG:C30 >> ['manf#'] (kicost.edas - tools.py:443)
DEBUG:part/qty>> CL10B105KO8NNNC		part>>CL10B105KO8NNNC	qty>>1 (kicost.edas - tools.py:623)
DEBUG:OrderedDict([('value', '1u'), ('libpart', 'Device:C_Small'), ('footprint', 'Capacitors_SMD:C_0603'), ('manf#', 'CL10B105KO8NNNC'), ('manf#_qty', '1')]) (kicost.edas - tools.py:482)
DEBUG:C28 >> ['manf#'] (kicost.edas - tools.py:443)
....
DEBUG:# Grouping parts... (kicost.edas - tools.py:113)
DEBUG:Getting groups of identical components... (kicost.edas - tools.py:138)
DEBUG:Checking the seemingly identical parts group... (kicost.edas - tools.py:199)
DEBUG:Merging field asked in the identical components groups... (kicost.edas - tools.py:254)
DEBUG:Propagating field values to identical components... (kicost.edas - tools.py:278)
DEBUG:Distributors: ['digikey', 'farnell', 'mouser', 'newark', 'rs', 'arrow', 'tme', 'lcsc'] (kicost - kicost.py:223)
DEBUG:Scrape API list ['Local', 'Octopart(Disabled)', 'KitSpace'] (kicost - kicost.py:76)
DEBUG:# Getting part data from KitSpace... (kicost.distributors - api_partinfo_kitspace.py:265)

Progress:   0%|          | 0/23 [00:00<?, ?part/s]
                                                  
DEBUG:URL https://dev-partinfo.kitspace.org/graphql query: (kicost.distributors - distributor.py:132)

Progress:   0%|          | 0/23 [00:00<?, ?part/s]
                                                  
DEBUG:query=query+%28%24input%3A+%5BMpnOrSku%5D%21%29%7B+match%28parts%3A+%24input%29+%7Bmpn%7Bmanufacturer%2Cpart%7D%2Cdatasheet%2Cdescription%2Cspecs%7Bkey%2Cvalue%7D%2Coffers%28from%3A%5B%22Arrow+Electronics%22%2C%22Digikey%22%2C%22Farnell%22%2C%22LCSC%22%2C%22Mouser%22%2C%22Newark%22%2C%22RS%22%2C%22TME%22%5D%29%7Bproduct_url%2Csku%7Bvendor%2Cpart%7D%2Cdescription%2Cmoq%2Cin_stock_quantity%2Cprices%7BGBP%2CEUR%2CUSD%7D%7D%7D+%7D&variables=%7B%22input%22%3A%5B%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22TPS7A8101DRB%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22CL10B105KO8NNNC%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22CL21A226MPQNNNE%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22CBC3225T100KR%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%2261082-101400LF%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22142-0701-851%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22C0402C222K3GECAUTO%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22GRM1555C1H102JA01D%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22ETC1-1-13TR%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22AT0402DRD0715RL%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22RR0510R-49R9-D%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22GJM1555C1H3R9CB01D%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22ADCLK905BCPZ-R7%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22RR0510P-101-D%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22RR0510P-241-D%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22RR0510R-49R9-D%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22GRM1555C1H102JA01D%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22MABACT0059%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22MA4E1339B1-1146T%22%7D%7D%2C%7B%22mpn%22%3A%7B%22manufacturer%22%3A%22%22%2C%22part%22%3A%22AD9643BCPZ-250%22%7D%7D%5D%7D (kicost.distributors - distributor.py:133)

Progress:   0%|          | 0/23 [00:00<?, ?part/s]
                                                  
DEBUG:<html>
<head><title>504 Gateway Time-out</title></head>
<body bgcolor="white">
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx/1.14.0 (Ubuntu)</center>
</body>
</html>
 (kicost.distributors - distributor.py:140)

Progress:   0%|          | 0/23 [01:00<?, ?part/s]
                                                  
DEBUG:Status Code: <504> (kicost.distributors - distributor.py:141)

Progress:   0%|          | 0/23 [01:00<?, ?part/s]ERROR:One of the internal Kitspace services may experiencing problems. Contact the Kitspace support. (kicost - __main__.py:419)

Progress:   0%|          | 0/23 [01:00<?, ?part/s]```

AlmAck, 2021-12-23:
Same problem here, but that 504 error seems related to kitspace servers, but I'm not sure

nitrousnrg, 2021-12-30:
Same problem here

mik1234mc, 2021-12-30:
Now, I use kicost api_ext_plugin branch with https://github.com/set-soft/kicost-digikey-api-v3 and Digikey&Mouser access works fine; however, you need to use distributor account keys.

set-soft, 2021-12-30:
> Now, I use kicost api_ext_plugin branch with https://github.com/set-soft/kicost-digikey-api-v3 and Digikey&Mouser access works fine; however, you need to use distributor account keys.

Good to know somebody can use it! I have Newark/Farnell almost finished, but KiCad 6 release and some KiBot additions are delaying it.

The idea is to off-load KitSpace using the vendor APIs directly.

mik1234mc, 2021-12-30:
Great to hear that development is ongoing. Farnell was missing distributor API for me as well. Thanks for your work.

set-soft, 2022-06-29:
KiCost supports Digi-Key, Farnell, Mouser, Newark and TME native APIs since v1.1.8.
So now you have options when KitSpace is saturated.
I'm closing this issue.