hildogjr/KiCost #513: Internal error: NoneType object has no attribute strip

sujitth, 2022-01-17:
- Use a simple and clear title.
- Do not mark as "ISSUE", "FEATURE" on title, this will be tagged by the development group.

#### Issue / Problem report
1. Add the KiCost version. Type ``kicost --version`` at terminal.
    V1.1.6
I am using kicost with Kicad 5.1.12 on a Win10 PC.

[image]

[image]

2. Add a terminal error report / txt, use ``--debug`` option on KiCost call;
    kicost -w -i %YOUR_FILE% --debug 8
3. Write the command used to call KiCost (or the graphical interface configuration);
4. One BoM to reproduce the error (with the EDA version).

#### Feature / Enhancement request
- Explain how it will help the other users.

The team thanks your contribution.

sujitth, 2022-01-17:
I have run the below command:
kicost -w -i %File% --debug 10 >error , but the error file is of 0bytes.

[image]

set-soft, 2022-01-17:
You must capture all the messages, try using `2>` instead of `>` (to capture stderr, not stdout)

sujitth, 2022-01-18:
@set-soft Thanks, with 2> I could capture all the messages and I have emailed it.

set-soft, 2022-10-28:
I can't reproduce it and the user wasn't able to capture the whole traceback.