hildogjr/KiCost #520: Release inconsistencies

sephalon, 2022-04-09:
#### Issue / Problem report
While [packaging KiCost for NixOS](https://github.com/NixOS/nixpkgs/pull/165666) I stumbled upon some release inconsistencies:
- The latest release tag is v1.1.7, but the latest release on PyPI as well as GitHub is v1.1.6
- The v1.1.6 source distribution release on PyPI is broken because it does not contain the `requirements.txt` file; this seems to be fixed on master already
- The KiCost Debian package in the v1.1.6 GitHub release claims to be v1.1.5-5

A new release containing support for configuration files would be greatly appreciated as well (with Kitspace Partinfo notoriously reaching its request limit, setting up alternative services is almost a necessity). Thanks!

set-soft, 2022-04-09:
> * The latest release tag is v1.1.7, but the latest release on PyPI as well as GitHub is v1.1.6

The last release on GitHub is 1.1.6 ([Releases](https://github.com/hildogjr/KiCost/releases)). The 1.1.7 wasn't released. And most probably it won't be released and 1.1.8 will be released.

> * The v1.1.6 source distribution release on PyPI is broken because it does not contain the `requirements.txt` file; this seems to be fixed on master already

I tried `pip3 install kicost` and got 1.1.6 installed without problems.
I remember fixing some issue during 1.1.7 test, but the one at PyPi seems to be working.

> * The KiCost Debian package in the v1.1.6 GitHub release claims to be v1.1.5-5

Lamentably @hildogjr did the release and didn't know about the Debian mechanism. This is true, the last 1.1.5 package is 1.1.6.

hildogjr, 2022-04-09:
I tested create manually the distribution package `python setup.py bdist` and all the requirements were included.

But look at https://pypi.org/project/kicost/#files, appear to be missing the `reqquirements.txt` on the source distribution (*.tar.gz). The *.whl file is file, with the requirements listed.

Is this error caused depending of *.tar.gz or *.whl installation source use? Or is already fix?
Could this be fix by adding the file as dependence to package generation (MANIFEST.in and setup.py file line 145)?

set-soft, 2022-04-09:
Hi @hildogjr !
I don't use `requirements.txt` on other projects, I just fill the `install_requires` list in `setup.py`

hildogjr, 2022-04-09:
I two, I changed to read `requirements.txt` on `setup.py` because iti was creating missupdate on commits.
I will add it as dependency.

sephalon, 2022-04-13:
> The last release on GitHub is 1.1.6 ([Releases](https://github.com/hildogjr/KiCost/releases)). The 1.1.7 wasn't released. And most probably it won't be released and 1.1.8 will be released.

OK, thanks for clarification.

> I tried `pip3 install kicost` and got 1.1.6 installed without problems. I remember fixing some issue during 1.1.7 test, but the one at PyPi seems to be working.

The issue is only triggered if there is no compatible binary wheel available for the system (as is the case on NixOS) or if you explicitly [tell pip not to use binary packages](https://pip.pypa.io/en/stable/cli/pip_install/#cmdoption-no-binary).

https://github.com/hildogjr/KiCost/commit/06d2fa015548aadcc34a173828f7cbac65c04d37 should solve the issue though. Thanks, closed.

closed 2022-04-13 by sephalon