hildogjr/KiCost #521: Non stock for mouser

diegosarli, 2022-05-26:
Hi, Mouser is returning NonStock for all the parts

version 1.1.8

best regards
Diego

set-soft, 2022-06-13:
Hi @diegosarli !
Can you post an example? Are you sure they have stock for the parts you selected? Are you using the Mouser P/N or just trying to search by value?

set-soft, 2022-06-29:
Hi @hkleen !
This explains why it fails for some people and not for others.
In english you can get "NNNN In Stock" and "None", but you could also get "NNNN On Order", which isn't really in stock.

The following patch should solve your problem: 7fe35522d22f7099ba5bd85cfd1ad56b0488e2c2

It should also solve it for @Bnjamn and @diegosarli but I'm not sure if it will solve the problem for everyone. Using `--debug 10` now you get things like:

```
DEBUG:Availability: 6809 In Stock (kicost.distributors - api_mouser.py:348)
...
DEBUG:- Search for stock: <re.Match object; span=(0, 13), match='6809 In Stock'> (kicost.distributors - api_mouser.py:352)
...
DEBUG:- Detected stock: 6809 (kicost.distributors - api_mouser.py:355)
```

So we will be able to adjust the detection. Can you confirm that this works? (at least for german ;-)

hkleen, 2022-06-29:
Hi @set-soft !
Sure thing! Just tried out the new version and everything seems to work just fine, at least for german.
Also the "NNNN On Order" case, which is "NNNN Auf Bestellung" in german, is handled correctly and results in No Stock reported. At least for me this fixed the problem.  Thanks for the quick fix!

set-soft, 2022-06-29:
Note for @Bnjamn and @diegosarli: if you find it doesn't work for your language please open a new issue including the debug output explained above. In this way I'll be able to fix a particular language.

closed 2022-06-29 by set-soft

Bnjamn, 2022-06-29:
Hello set-soft,   
it work's better right now.   
Thanks for the quick fix.   
Never the less, from my observation, it doesn't show the effective stock of Mouser, but the maximum quantity Mouser can ship.   

For example, AD835ARZ here :
     -----> https://www.mouser.fr/ProductDetail/Analog-Devices/AD835ARZ?qs=%2FtpEQrCGXCx2l1%252Byta%2FM0g%3D%3D   
Mouser have a stock of : 4 486pcs   
But we can buy only 196pcs max.   

I'm also interesting of getting the "On Order" column, is there a possibility to add columns in the Kicost results ?   

Best,
Benjamin.

set-soft, 2022-06-29:
Hi @Bnjamn !
Currently KiCost has just one column for the stock. Adding extra columns for one distributor is complex.

If you take a look at @hkleen data (from a direct query using curl) you'll see Mouser only informs a fuzzy field called `Availability`. If they have stock the field says "XXX In stock", if they have pending orders, but no stock, they indicate "XXX On order". But never both values.

set-soft, 2022-10-06:
Hi @Bnjamn !
The above patch will help in the case that Mouser reports the part is "On Order". As I mentioned before we get only the fuzzy *Availability*, but now parts that aren't in stock, but on order will be known by looking at the cell comment.