hildogjr/KiCost #525: Where is KiCost 1.1.11?

craigarno, 2022-07-26:
After having success with KiCost 1.1.11 I decided to install KiCost on another system using the same config file.  The first thing I noticed is the "git Clone" process says it is version 1.1.10.  When I try to run 1.1.10, I get an error.

#### Issue / Problem report
1. Add the KiCost version. Type ``kicost --version`` at terminal.
kicost --version
KiCost v1.1.10

2. Add a terminal error report / txt, use ``--debug`` option on KiCost call;```

    kicost --show_dist_list --debug 8
DEBUG:API options with defaults {'Local': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Local'}, 'Digi-Key': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Digi-Key'}, 'Mouser': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Mouser'}, 'Element14': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Element14'}, 'TME': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\TME'}, 'Octopart': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Octopart'}, 'KitSpace': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\KitSpace'}} (kicost - log.py:115)
DEBUG:Final API options {'Local': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Local'}, 'Digi-Key': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Digi-Key'}, 'Mouser': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Mouser'}, 'Element14': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Element14'}, 'TME': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\TME'}, 'Octopart': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Octopart'}, 'KitSpace': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\KitSpace'}} (kicost - log.py:115)
DEBUG:Local API configured to enabled True (kicost.distributors - log.py:115)
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\__main__.py", line 487, in main
    main_real()
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\__main__.py", line 360, in main_real
    configure_kicost_apis(args.config, args.config is None, command_line_api_options, args)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\__main__.py", line 175, in configure_kicost_apis
    configure_apis(api_options)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\__init__.py", line 90, in configure_apis
    distributor_class.configure_apis(options)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\distributor.py", line 222, in configure_apis
    api.configure(ops)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\api_digikey.py", line 113, in configure
    format(api_digikey.enabled, DK_API.id, DK_API.secret, cache_path))
AttributeError: type object 'DK_API' has no attribute 'id'
```

3. Write the command used to call KiCost
 kicost --show_dist_list --debug 8

4. One BoM to reproduce the error (with the EDA version).
N/A

set-soft, 2022-07-26:
Hi @craigarno !

> The first thing I noticed is the "git Clone" process says it is version 1.1.10

The `git clone` command can't say which version are you cloning. You clone the whole repo, not just a tag. How did you run `git clone` (the whole command).

v1.1.10 is the last stable, is what you get using `pip install kicost`. The v1.1.11 is just the current development code. Did you install the code you cloned? (I.e change to the dir you cloned and run `pip install .`)

craigarno, 2022-07-26:
> > The first thing I noticed is the "git Clone" process says it is version 1.1.10
> 
> The `git clone` command can't say which version are you cloning. You clone the whole repo, not just a tag. How did you run `git clone` (the whole command).

It is the next day for me, and difficult to remember exactly what I did.  I suspect I meant, "After executing Git Clone, then executing `KiCost --version` ...".  I'm not sure why I'd do this, and suspect the results I saw were from just executing "pip install KiCost", then noticing my Config.YAML file wasn't working.

I'm pretty sure I followed the instructions at https://hildogjr.github.io/KiCost/docs/_build/singlehtml/index.html#windows-with-kicad without the last step
 prompt> pip install git+https://github.com/hildogjr/KiCost.git

After performing the last step (executing the command) and rerunning KiCost I get
  kicost --version 
  KiCost v1.1.11

And the more helpful 1.1.11 output:
```
kicost --show_dist_list
WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - log.py:107)
Distributor list: arrow digikey farnell lcsc mouser newark rs tme
```
Though I am unsure why KiCost can't find Vendor API keys, because I am using the same Config.YAML as the working system, and I'm looking at the Config.YAML now and the "missing" information is in Config.YAML, enabled, and not commented out.

It might be helpful if KiCost would show a path and filename for the Config.YAML file, because this message only makes sense to me if KiCost is ignoring Config.YAML and using defaults.  --debug 8 doesn't display this Config PATH/FILENAME information:
```
C:\Users\craig\Documents\KiCad\6.0>kicost --show_dist_list --debug 8
DEBUG:API options with defaults {'Local': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Local'}, 'Digi-Key': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Digi-Key'}, 'Mouser': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Mouser'}, 'Element14': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Element14'}, 'TME': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\TME'}, 'Octopart': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Octopart'}, 'KitSpace': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\KitSpace'}} (kicost - log.py:115)
DEBUG:Final API options {'Local': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Local'}, 'Digi-Key': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Digi-Key'}, 'Mouser': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Mouser'}, 'Element14': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Element14'}, 'TME': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\TME'}, 'Octopart': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\Octopart'}, 'KitSpace': {'cache_ttl': 7, 'cache_path': 'C:\\Users\\craig/.cache/kicost\\KitSpace'}} (kicost - log.py:115)
DEBUG:Local API configured to enabled True (kicost.distributors - log.py:115)
DEBUG:Digi-Key API not available (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)
DEBUG:Mouser API configured to enabled False key None path C:\Users\craig/.cache/kicost\Mouser (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - log.py:107)
DEBUG:Element14 API configured to enabled False key None path C:\Users\craig/.cache/kicost\Element14 (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - log.py:107)
DEBUG:TME API configured to enabled False token None app_secret None path C:\Users\craig/.cache/kicost\TME (kicost.distributors - log.py:115)
DEBUG:Octopart API configured to enabled False key None level 4 extended False (kicost.distributors - log.py:115)
DEBUG:KitSpace API configured to enabled True (kicost.distributors - log.py:115)
Distributor list: arrow digikey farnell lcsc mouser newark rs tme
```

From the KiCad 6.0 Command Prompt, this command shows me the working Config.yaml copied and installed from the other Windows 10 Pro system:
` C:\Users\craig\Documents\KiCad\6.0>notepad c:\Users\craig\.config\config.yaml`

The only entry `enable: false` is OctoPart.  All other entries are enabled and have valid keys.

Now that I have 1.1.11 installed, I'll keep scratching my head to see if I can figure this out.  I will be more comfortable with project productivity risk if I have 2 or 3 working KiCost seats (I use multiple machines during development).

set-soft, 2022-07-27:
Hi @craigarno !

> It might be helpful if KiCost would show a path and filename for the Config.YAML file, because this message only makes sense to me if KiCost is ignoring Config.YAML and using defaults. --debug 8 doesn't display this Config PATH/FILENAME information:

Ok, I added debug info about it. Note that is just at debug level, currently the config file isn't mandatory and is a new stuff.

> C:\Users\craig\Documents\KiCad\6.0>notepad c:\Users\craig\.config\config.YAML

The file should be c:\Users\craig\\.config\\**kicost**\config.yaml
And try to avoid using uppercase letters, I know Windows ignores them, but the name is `config.yaml`, not `Config.YAML`, this could be a problem if you use Linux in the future.

craigarno, 2022-07-27:
Excellent!
```
c:\Users\craig\.config>kicost --show_dist_list --debug 8
DEBUG:No config file found (C:\Users\craig\.config\kicost\config.yaml) (kicost - log.py:115)
```
```
c:\Users\craig\.config>kicost --show_dist_list --debug 8
DEBUG:Loading config from: C:\Users\craig\.config\kicost\config.yaml (kicost - log.py:115)
```
And as expected, after the issue is corrected [config.yaml placed in the kicost config sub-directory], kicost 1.1.11 works properly! (on both systems), yea!

And yes, I have Linux experience and generally follow your "lowercase only" rule for the reasons you suggest.  I have a spell checker which had too much **RED** showing because it recognized YAML as an acronym for Yet Another Markup Language.  To keep from overlooking poor grammar and misspelling, to which I am prone, I let is capitalize all the YAML's.

Sorry to put you through a "worst customer" experience, but someone has to be the dumb tester!  ;)  The mistake of overlooking the "kicost" subdirectory was real.  [like you, I'm over qualified for the "dumb tester" task]

closed 2022-07-27 by craigarno