hildogjr/KiCost #526: DigiKey API: add option to exclude marketplace items

salfter, 2022-07-27:
#### Feature / Enhancement request

DigiKey provides product pricing not just for itself, but for a number of other sellers as well through its "marketplace" feature.  In my admittedly limited experience, these additional sellers frequently have higher MOQs than I can handle, or they might indicate zero inventory at a lower price when DigiKey itself has the same MPN in stock, even if it's at a higher price.

When doing manual searches, I've taken to making sure the "exclude marketplace products" checkbox is checked.  It'd be nice if KiCost could implement a similar capability, assuming the DigiKey API allows it.  (I've only had it up and running for maybe an hour since getting the Gentoo ebuild for KiCost written and tested, so I've not yet had a chance to give the program a full workout.)

set-soft, 2022-10-06:
Hi @salfter !
Do you have any example to test it?

set-soft, 2022-10-06:
The above patch needs the plugin v0.1.2.
It seems to work (different results when excluding marketplace), but I didn't check it deeply

commit 7e971207 (2022-10-06): Added option to exclude Digi-Key marketplace products

Closes #526
closed 2022-10-06 by set-soft