hildogjr/KiCost #527: TME Cat# URLS in file:\\ format

craigarno, 2022-07-28:
#### Issue / Problem report
TME Hyperlinks are in the format:
\\\\www.tme.eu\en\details\2n7002-7-f\smd-n-channel-transistors\diodes-incorporated\

Converting to 
https://www.tme.eu/en/details/2n7002-7-f/smd-n-channel-transistors/diodes-incorporated/

works, but sends me to the TME EU site.  I'm in the US.  (not a big deal, I get asked if I'd like to switch to the US site.)
https://www.tme.com/us/en-us/details/2n7002-7-f/smd-n-channel-transistors/diodes-incorporated/

When I see
[image]
I start looking at supplier site pages and data sheets for supporting information, like what part is recommended for replacing the part which is now "not recommended for new design".

These new additions to KiCost 1.1.11 are useful.

1. Add the KiCost version. 
KiCost 1.1.11 was just installed this morning after installing KiCAD 6.0.7 which removed KiCost and KiPart from my installation.

2. Write the command used to call KiCost (or the graphical interface configuration);
[image]

5. One BoM to reproduce the error (with the EDA version).

[USB-C PWR.xlsx](https://github.com/hildogjr/KiCost/files/9213239/USB-C.PWR.xlsx)

set-soft, 2022-10-05:
Hi @craigarno !
Sorry for the big delay.
Is now fixed.
About the european site:
- The US counterpart (api.tme.us) resolves to a valid IP, but isn't usable
- Even when we ask for US as default country the URLs are for the european domain

closed 2022-10-05 by set-soft

craigarno, 2022-10-06:
@set-soft , perfect!

commit ad86e699 (2022-10-05): [TME] Fixed URLs for components

Closes #527