hildogjr/KiCost #531: Add KiKost to the Addon Manager

RolandWa, 2022-08-30:
Hi All,
Add KiKost to the Addon Manager, follow instruction on webside :

https://dev-docs.kicad.org/en/addons/

set-soft, 2022-10-05:
Hi @RolandWa !

This is a good idea ... but ... I don't see a mechanism to declare Python dependencies. I think KiCad plug-ins packaged for the Addon Manager are limited to plug-ins without extra Python dependencies.

Also: no hooks are provided, like `pre-install`, `post-install`, `pre-rm` and `post-rm`.

We could implement a complex mechanism that installs a KiCost installer, in the first run it could install the dependencies and the real KiCost script. But I'm not sure about this.

set-soft, 2022-10-05:
Note that this problem was discussed in the KiCad Forum 1 year ago, without solutions https://forum.kicad.info/t/plugin-and-content-manager-how-to-handle-plugins-with-dependencies/31884