hildogjr/KiCost #535: Warnings about APIs keys

Kralg, 2022-10-09:
Hello,
I am using the command: kicost -i boards.xml
I am unable to get it working, I am getting warnings:

WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - log.py:107)

Or if I run it over GUI:
Starting cost processing ...
- KitSpace [api] (https://kitspace.org/)
ERROR:One of the internal Kitspace services may experiencing problems. Most probably is saturated, please consider using other API. (kicost - __init__.py:1463)

Can you please explain do I have to register and get the tokens from each distributor and set it somehow or is this something else?

mik1234mc, 2022-10-09:
Yes, exactly, you need to have account with all distributors you want to
see in the Kicost result Excel sheet.

The distributors provide the tokens.

Kralg, 2022-10-09:
Thank you for the fast respone. Yeah it would be really handy to get some information in the warning in what way we can pass the tokens and what names to use. 
I believe this is system environment variables for windows?
For Mouser:
MOUSER_PART_API_KEY

For TME:
TME_TOKEN - place here "Anonymous key" from TME API. ?
TME_APP_SECRET - place here "Application secret:" -> "Token" ?

set-soft, 2022-10-11:
Hi @Kralg !
Please read the docs: [Configuration file](http://hildogjr.github.io/KiCost/docs/_build/singlehtml/index.html#document-configuration)
You don't need environment variables, you can store all in a configuration file. Note that the comments in the template config will help you to understand what is each field.
If you have example files that produce errors when using the current git code please create an issue posting the example file.

closed 2022-10-11 by set-soft