hildogjr/KiCost #536: Nexar API error - KeyError: 'manufacturer'

craigarno, 2022-11-02:
- Use a simple and clear title.
- Do not mark as “ISSUE”, “FEATURE” on title, this will be tagged by the development group.

#### Issue / Problem report

Many component libraries have things sometimes undesirable defined.  In this case there was a predefined “MANUFACTURER” field which needs to be ignored.  I define my own with “mfg” and “mfg#”.

1. Add the KiCost version. Type “kicost --version`“at terminal.
>kicost --version
KiCost v1.1.13
2. 
```
- Nexar [api] (https://nexar.com/api)
Progress: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 30/30 [00:03<00:00,  8.72part/s]
Progress:   0%|                                                                                                                                                  | 0/1 [00:00<?, ?part/s]Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\__main__.py", line 471, in main
    main_real()
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\__main__.py", line 461, in main_real
    kicost(in_file=args.input, eda_name=args.eda,
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\kicost.py", line 251, in kicost
    query_part_info(parts, dist_list, currency)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\kicost.py", line 78, in query_part_info
    get_dist_parts_info(parts, dist_list, currency)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\__init__.py", line 50, in get_dist_parts_info
    distributor_class.get_dist_parts_info(parts, dist_list, currency)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\distributor.py", line 146, in get_dist_parts_info
    solved = api.query_part_info(parts, list(remaining), currency)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\api_nexar.py", line 631, in query_part_info
    unsolved, n_unsolved = api_nexar.solve_queries(unsolved, n_unsolved, currency)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\api_nexar.py", line 557, in solve_queries
    api_nexar.get_part_info(unsolved[slc], currency, to_retry)
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\api_nexar.py", line 315, in get_part_info
    q.remove_manfucturer()
  File "C:\Users\craig\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kicost\distributors\api_nexar.py", line 155, in remove_manfucturer
    del self.query['manufacturer']
KeyError: 'manufacturer'
Progress:   0%|
```

3. Write the command used to call KiCost (or the graphical interface configuration);
`kicost -w -i "USB-C PWR.xml" --eda kicad` 

#### Feature / Enhancement request
- Explain how it will help the other users.
Reduce the amount of "cleanup" required from UltraLibrarian, SnapEDA, and other component library vendors.
The other API's [digikey, mouser, etc.] seem to ignore this "problem"

set-soft, 2022-11-07:
Hi @craigarno !
Thanks for reporting.
The problem seems to be like this: You have a part without manufacturer and we got 0 hits from Nexar. When we get 0 hits we try again without specifying a manufacturer (in case the name didn't match because Nexar calls it in a different way or is just wrongly typed). But here we didn't have a manufacturer, so we failed to remove the manufacturer from the query. The patch dfa55ab86c2c1259ac44b1e1106e6065f09cd804 should avoid this situation.

closed 2022-11-07 by set-soft

craigarno, 2023-01-12:
Thanks, my email is broken, so I have to remember to login and check...

I downloaded the latest KiCost 1.1.15 and still get a very similar error:
```
Starting cost processing ...
WARNING:(WC010) Warning: in Q1 overwriting footprint=usbc-cypd3177-24lqxq:SOIC127P600X175-8N with PACKAGE=SOIC-8 (kicost.edas - log.py:107)
- Digi-Key [api] (https://developer.digikey.com/)
- Mouser [api] (https://api.mouser.com/)
- Element14 [api] (https://partner.element14.com/)
WARNING:(WC015) Got 9 hits for mpn C0603C105K4RACTU (kicost.distributors - log.py:107)
WARNING:(WC015) Got 5 hits for mpn 0402ZC104KAT2A (kicost.distributors - log.py:107)
WARNING:(WC015) Got 3 hits for mpn C1210C335K5RACAUTO (kicost.distributors - log.py:107)
WARNING:(WC007) No information found at farnell for part/s '['C8']' (kicost.distributors - log.py:107)
WARNING:(WC015) Got 3 hits for mpn C1005X5R1V105K050BC (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn SPHV24-01ETG-C (kicost.distributors - log.py:107)
WARNING:(WC015) Got 4 hits for mpn LTST-C190EKT (kicost.distributors - log.py:107)
WARNING:(WC015) Got 5 hits for mpn LTST-C190GKT (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn NZH10C (kicost.distributors - log.py:107)
WARNING:(WC015) Got 6403 hits for mpn 115 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn 632723300011 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 5 hits for mpn SFC-110-T2-L-D-A (kicost.distributors - log.py:107)
WARNING:(WC007) No information found at farnell for part/s '['J3', 'J4']' (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn IRF9358TRPBF (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn BSL308PEH6327XTSA1 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 8 hits for mpn NTR4503NT1G (kicost.distributors - log.py:107)
WARNING:(WC015) Got 7 hits for mpn RC0603JR-071KL (kicost.distributors - log.py:107)
WARNING:(WC015) Got 17 hits for mpn RC0603FR-0710KL (kicost.distributors - log.py:107)
WARNING:(WC015) Got 7 hits for mpn ERJ-3EKF1002V (kicost.distributors - log.py:107)
WARNING:(WC007) No information found at farnell for part/s '['R17']' (kicost.distributors - log.py:107)
WARNING:(WC015) Got 6 hits for mpn ERJ-2RKF4992X (kicost.distributors - log.py:107)
WARNING:(WC015) Got 4 hits for mpn RC0402JR-0710KL (kicost.distributors - log.py:107)
WARNING:(WC007) No information found at farnell for part/s '['R22']' (kicost.distributors - log.py:107)
WARNING:(WC015) Got 3 hits for mpn RC0402JR-072KL (kicost.distributors - log.py:107)
WARNING:(WC015) Got 133 hits for mpn 5001 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 137 hits for mpn 5003 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 434 hits for mpn 5000 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 8 hits for mpn 0402ZC104KAT2A (kicost.distributors - log.py:107)
WARNING:(WC015) Got 6 hits for mpn C1210C335K5RACAUTO (kicost.distributors - log.py:107)
WARNING:(WC007) No information found at newark for part/s '['C8']' (kicost.distributors - log.py:107)
WARNING:(WC015) Got 5 hits for mpn C1005X5R1V105K050BC (kicost.distributors - log.py:107)
WARNING:(WC015) Got 6 hits for mpn LTST-C190GKT (kicost.distributors - log.py:107)
WARNING:(WC015) Got 15783 hits for mpn 115 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 7 hits for mpn NTR4503NT1G (kicost.distributors - log.py:107)
WARNING:(WC015) Got 10 hits for mpn RC0603FR-0710KL (kicost.distributors - log.py:107)
WARNING:(WC015) Got 8 hits for mpn ERJ-3EKF1002V (kicost.distributors - log.py:107)
WARNING:(WC015) Got 5 hits for mpn RC0402FR-070RL (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn ERJ-P6WJ222V (kicost.distributors - log.py:107)
WARNING:(WC015) Got 3 hits for mpn ERJ-2RKF4992X (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn ERJ-P6WJ104V (kicost.distributors - log.py:107)
WARNING:(WC015) Got 6 hits for mpn RC0402JR-0710KL (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn CRCW04021K00FKEDC (kicost.distributors - log.py:107)
WARNING:(WC015) Got 9 hits for mpn RC0402JR-072KL (kicost.distributors - log.py:107)
WARNING:(WC015) Got 260 hits for mpn 5001 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 130 hits for mpn 5003 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 1836 hits for mpn 5000 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn CYPD3177-24LQXQ (kicost.distributors - log.py:107)
- Nexar [api] (https://nexar.com/api)
ERROR:Internal error: list index out of range (kicost - __init__.py:1475)
```
On the command line I get this error:
```
WARNING:(WC015) Got 1836 hits for mpn 5000 (kicost.distributors - log.py:107)
WARNING:(WC015) Got 2 hits for mpn CYPD3177-24LQXQ (kicost.distributors - log.py:107)
Progress: 100%|███████████████████████████████████████████████████████████████████████████████████████████████| 33/33 [00:00<00:00, 60.54part/s]
- Nexar [api] (https://nexar.com/api)
Progress: 100%|███████████████████████████████████████████████████████████████████████████████████████████████| 33/33 [00:03<00:00,  9.32part/s]
Progress: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  1.27part/s]
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
[... 6 lines cut ...]
[... cut ...]
  File "C:\Program Files\KiCad\6.0\bin\Lib\site-packages\kicost\distributors\api_nexar.py", line 395, in select_best_part
    best_match = difflib.get_close_matches(manf, manufacturers.keys())[0]
IndexError: list index out of range

C:\Users\craig\Desktop\Catamaran\usbc-cypd3177-24lqxq\kicad>kicost --version
KiCost v1.1.15
```

I'm adding my XML input file from KiCAD 6.0.10 to assist in finding this error.
[usbc-cypd3177-24lqxq.zip](https://github.com/hildogjr/KiCost/files/10398503/usbc-cypd3177-24lqxq.zip)

Thank you.