hildogjr/KiCost #543: Startup breaks with circular import error

baettigp, 2023-02-16:
Hi, I just installed Kicost under Windows10 (python 3.11). When I try to start "kicost" it breaks with

`$ kicost
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\Python311\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python311\Lib\site-packages\kicost\__main__.py", line 476, in main
    main_real()
  File "C:\Program Files\Python311\Lib\site-packages\kicost\__main__.py", line 189, in main_real
    version=kicost_version_info(),
            ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\kicost\__main__.py", line 92, in kicost_version_info
    import wx
  File "C:\Program Files\Python311\Lib\site-packages\wx\__init__.py", line 12, in <module>
    __version__ = wx.__version__.VERSION_STRING
                  ^^^^^^^^^^^^^^
AttributeError: partially initialized module 'wx' has no attribute '__version__' (most likely due to a circular import)
`
I am now not entirely sure as to how I can get it running. Any help would be appreciated.

Pio

set-soft, 2023-03-19:
If you don't use the Python included with KiCad you'll most probably find a lot of problems.

closed 2023-03-19 by set-soft

baettigp, 2023-03-19:
Thank you very much for the hint. Where could I read on how to get it to use kicads python? Or is there a command window inside kicad with the correct version that I overlooked? I will admit to pebkac as I am working under Linux at home and there it works no problem, thank you very much for the great tool!

set-soft, 2023-03-19:
KiCad has a "KiCad Command Prompt", you can see how to access it in PcbDraw docs:

https://github.com/yaqwsx/PcbDraw/blob/master/doc/installation.md

This shortcut creates an environment to run KiCad's embedded python